opendatacam/opendatacam #134: Persist counter settings

b-g, 2019-10-07:
@tdurand 

Currently playing with the video file processing mode #88 ... one thing which drives me a bit nuts from an usability point of view, is that every time my counter settings are gone on restart. Is there any way to load the last counter settings on start up?

(Or save these settings aside the data results ... so that one can re-load them?)

tdurand, 2019-10-07:
Right now none of these things are built in, but it's not that hard to add this.. everything is in place to do so..  (you could with the API , there is a get areas and register areas endpoint : https://opendatacam.github.io/opendatacam/apidoc/#api-Counter ) 

What default would be nice ? Latest counter settings ? 

I may dedicate some time to opendatacam at the end of the week, or early next week.

b-g, 2019-10-07:
Great to hear! :)

> What default would be nice ? Latest counter settings ?

Yes, having the "last counter settings" load on default ... would already be ok and from the usability a big improvement.

tdurand, 2019-10-14:
Ok, should work now on v2.1 which I'm releasing later today

closed 2019-10-14 by tdurand