opendatacam/opendatacam #135: OpenDataCam v2.1 beta for workshop?

b-g, 2019-10-11:
Hi @tdurand,

We have a workshop coming up next week at HS Mannheim around Prof. @tillnagel .... Hence @rraapphhaaeell and myself are wondering which version we should ask the students to use? I I would vote to use `v2.1.0-beta.1` but wanted to hear your opinion on it.

@rraapphhaaeell you can give it a try with:
```
wget -N https://raw.githubusercontent.com/opendatacam/opendatacam/v2.1.0-beta.1/docker/install-opendatacam.sh
chmod 777 install-opendatacam.sh
./install-opendatacam.sh --platform nvidiadocker_cuda_archbin_6_1
```
It you want to use the webcam instead of a video file or change the YOLO .weight file etc. edit the `config.json` which is created on installation.

You have to have docker [installed](https://github.com/opendatacam/opendatacam/blob/master/documentation/nvidia-docker/INSTALL_NVIDIADOCKER.md) and running. Good news is that docker is pre-installed with the recent Ubuntu for the Jetson family.

tdurand, 2019-10-11:
Hi, yes `v2.1.0-beta.1` is the latest but I didn't build it yet for jetsons as I was trying to fix some memory leaks problems but infortunately didn't succeed to improves things in a reliable way.

What machine will student use for workshop ? When it is ? 

I had planned to do the build for the final v2.1 for all platform early next week, I could do it on monday 

Best, 

Thibault

b-g, 2019-10-11:
OK + great. I think the workshop will start mid of next week. I believe it is planned to use all of our platforms (Jetson TX2, Nano and Desktop). Have a nice weekend!

tdurand, 2019-10-11:
Ok great, so I schedule to do this on monday !

tillnagel, 2019-10-11:
Currently, we have a Nano and a TX2. 

Some of my students volunteered to try out the install even before the workshop (which is Tuesday early morning CEST).

Besides stability/last issues in the release, are there major differences in the installation routine? If not, it might be good to proceed installing the latest release and then update when your beta is ready.

tdurand, 2019-10-11:
Hi @tillnagel 

No it is the same installation process: https://github.com/opendatacam/opendatacam#2-install-and-start-opendatacam- . Just replacing v2.0.0 by the version number.

If they want to try out before, they should install the v2.0.0 as the docker builds for v2.1-alpha / beta were only generated for Ubuntu desktop.

The 2.1 release add better support to run on files and a few other fixes, besides that, it is pretty much the same.

I'll get all the 2.1 builds for docker Monday, "upgrading" process only consist in trashing previous version and installing the new one 😉

Let me know if you reach into any issues. Happy to help

rraapphhaaeell, 2019-10-14:
Heyhey, I tried this morning, to update the jetsons, we have at the office. Had I only read this thread before, had I known my impossibility to succeed 🙈️... I'll do this in the next few days as soon as you are ready, @tdurand. 

@tillnagel I'll have to send you a fully updated version by mail then, since I didnt pack it up today, due to the outdated version we have running on the ones we have in the office.

tdurand, 2019-10-14:
@rraapphhaaeell I am working on the build new builds today. Should be ready by tomorrow for you

tdurand, 2019-10-14:
OK ! just released the builds for v2.1.0 for all platform.

In order to upgrade , you need to trash the old stuff https://github.com/opendatacam/opendatacam#2-bis-optional-upgrade-opendatacam-from-v2x-to-another-v2x-version before , and then follow the normal install procedure again.

Let me know if you have any issue, the release note are: https://github.com/opendatacam/opendatacam/releases/tag/v2.1.0 , as I said, the only big change is improving run from file mode, which is documented here: https://github.com/opendatacam/opendatacam/blob/master/documentation/CONFIG.md#run-opendatacam-on-a-video-file

b-g, 2019-10-15:
@tdurand Yay! Thanks!

rraapphhaaeell, 2019-10-16:
amazing. Thanks. Just updated our xavier and tx2.

closed 2019-11-06 by tdurand