opendatacam/opendatacam #143: cannot run with camera stream

MeTaXaS4, 2019-11-06:
[with docker]
I am trying to run the latest version with `usbcam` or `usbcam_no_gstreamer` but I am getting on console always `Video stream stopped` how can I debug more?

tdurand, 2019-11-08:
Hello, 

Are you working with a jetson ? 

Can you try this command:
```
ls /dev/video*
```

Output should be: /dev/video0 , if it not the case it means the webcam has been mounted on another /dev , in that case you should edit the `config.json` file to reflect this.. For example if it is /dev/video1 you should change to

```
"usbcam": "v4l2src device=/dev/video1 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink",
"usbcam_no_gstreamer": "-c 1",
```

MeTaXaS4, 2019-11-08:
Hello,
my bad I didn't said that I am not working with jetson, but an ubuntu machine

I already tried that and without camera I see nothing with command, with the camera I see `video0` and `video1`, tried both on config but nothing

MeTaXaS4, 2019-11-08:
miss click closing, reopening

tdurand, 2019-11-08:
Ok this is normal is you are not working with a jetson, there is an undocumented thing to do:

You need to edit the `run-nvidiadocker.sh` script that is downloaded when installing,  and add the following to allow docker to access the webcam:

Remplace the last line: https://github.com/opendatacam/opendatacam/blob/master/docker/run-nvidia-docker/run-nvidiadocker.sh#L7  

`docker run --runtime=nvidia -p 8080:8080 -p 8090:8090 -p 8070:8070 $DOCKER_VOLUMES -v /data/db:/data/db $@`

with 

`docker run --runtime=nvidia --device=/dev/video0:/dev/video0 -p 8080:8080 -p 8090:8090 -p 8070:8070 $DOCKER_VOLUMES -v /data/db:/data/db $@`

tdurand, 2019-11-08:
And then try to `run-opendatacam.sh` again

MeTaXaS4, 2019-11-08:
how did I miss that?? https://github.com/opendatacam/opendatacam/pull/104

tdurand, 2019-11-08:
Did it works ? 

Yes we had to revert #104 because if you have no webcam connected it makes the script fail..

MeTaXaS4, 2019-11-08:
it is working like that yes...

how did I forgot that I need to do something like that? we checked it in the past.. :D

closed 2019-11-08 by MeTaXaS4