opendatacam/opendatacam #144: Evaluate upcoming Jetson Xavier NX for OpenDataCam

b-g, 2019-11-07:
$400
size of a credit card 
10 Watt

https://venturebeat.com/2019/11/06/nvidia-unveils-jetson-xavier-nx-for-edge-ai/

via @florianporada

tdurand, 2019-11-08:
Sounds great ! Getting smaller and smaller and cheaper.

prat96, 2019-11-27:
It's possible to get started with dev already, Nvidia has released a patch that emulates a Xavier NX on a full Xavier system :)

tdurand, 2020-04-27:
I think the final Jetpack 4.4 is set to be released alongside with the developer kit for Jetson Xavier NX which was planned to be released this month: https://twitter.com/raffaello86/status/1254740965203554305

tdurand, 2020-05-14:
Jetson Xavier NX developer kit came out today, $400 USD , a bit less power than the full Jetson Xavier but only 15W power consumption ;-) 

https://www.rnext.it/review/jetson-xavier-nx/

To give you an example jetson nano with deepstream should run tiny yolo at 30 FPS, but wit the Jetson Xavier NX you can run tiny yolo at 570 FPS 😅

This also makes the Jetson TX2 obsolete.. which is priced the same but is 10x less powerful... Actually we might want not to provide out of the box docker builds for jetson tx2 in the future to ease a bit the maintenance...

Regarding OpenDataCam .. From what I understood the build for Jetson Xavier should work for Jetson Xavier NX.. but would be nice to confirm once someone buys one 😎

Denizzje, 2020-06-23:
Unless the postal service fails me, I shall have one by tomorrow. 

Curious to see if it runs Yolov4.

Denizzje, 2020-06-24:
Sample Yolov4 demo video from docker on Xavier NX:

15W 6core: 12fps
15W 2core: 12fps
10W 2core: 7-8fps

Seems usuable for the cars video atleast on the 15W mode.

tdurand, 2020-06-24:
great ! thanks for the benchmark.. I think I get 18 FPS with yolov4 with Xavier (normal)

Will close this ;-)

closed 2020-06-24 by tdurand