opendatacam/opendatacam #145: VIDEOIO ERROR:Unable to query number of channels

shrthprsd, 2019-11-11:
Hello, 
Thanks for open Sourcing this code. I am trying to run the code on a Jetson TX2 with the on-board camera as well as trying it with a USB camera. 
```
v4l2-ctl --list-devices
```
gives the following output:
```
vi-output, ov5693 2-0036 (platform:15700000.vi:2):
	/dev/video0

EyeToy USB camera Namtai (usb-3530000.xhci-2.2):
	/dev/video1
```
The following works when I execute in the terminal:
```
gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM),width=3820, height=2464, framerate=21/1, format=NV12' ! nvvidconv flip-method=0 ! 'video/x-raw,width=960, height=616' ! nvvidconv ! nvegltransform ! nveglglessink -e
```

However I am running into this error 
When I tried with nvarguscamerasrc in the config.json file, I get the following error.

```
(darknet:80): GStreamer-CRITICAL **: 22:29:48.583: gst_element_get_state: assertion 'GST_IS_ELEMENT (element)' failed

VIDEOIO ERROR: V4L: device nvarguscamerasrc ! 'video/x-raw(memory:NVMM),width=3820, height=2464, framerate=21/1, format=NV12' ! nvvidconv flip-method=0 ! 'video/x-raw,width=960, height=616' ! nvvidconv ! nvegltransform ! nveglglessink -e: Unable to query number of channels
```

When I tried with v4l2src in the config.json file I get the following error:

```
GStreamer-CRITICAL **: 16:12:31.784: gst_element_get_state: assertion ‘GST_IS_ELEMENT (element)’ failed

VIDEOIO ERROR: V4L: device v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink: Unable to query number of channels
```
Any help is greatly appreciated.

tdurand, 2019-11-12:
Hi ! 

Some pointers that might help you:

- From the docker image, you can't use `nvarguscamerasrc` , it is not available  , you will be able to use it if you install [opendatacam without docker](https://github.com/opendatacam/opendatacam/blob/master/documentation/USE_WITHOUT_DOCKER.md)

- We tested only with usbcamera, so I think if you try with /dev/video1 it should work ..

shrthprsd, 2019-11-12:
Thanks @tdurand . I want to run this within a container, so I will not nvarguscamerasrc as you mentioned it is not available.
I tried with /dev/video1 and it did not work. I am using a PS2 eye toy camera. The output from  
```
$ v4l2-ctl -d /dev/video1 --list-formats-ext
```
 is as follows: 
```
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'JPEG' (compressed)
	Name        : JFIF JPEG
		Size: Discrete 320x240
		Size: Discrete 640x480
```
I tried the following command too without any luck:
```
v4l2src device=/dev/video1 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink
```

Any suggestions to get it to work with the PS2 eye camera ? Or do I need to buy another camera ?
Thanks.

shrthprsd, 2019-11-13:
Hello @tdurand, 

Quick Question: Are you using `cv2.VideoCapture` ? If so, could you please point us to where it is in your code ?
The reason I ask is this [post](https://devtalk.nvidia.com/default/topic/1057470/jetson-tx2/to-use-onboard-camera-/post/5377509/#5377509) says: "adding a second argument cv2.CAP_GSTREAMER after your pipeline string in cv2.videoCapture" - will solve the issue I am facing.

Thanks

tdurand, 2019-11-14:
Hi, 

- We successfully ran Opendatacam with those cameras:  Logitech C222, C270, C310, C920
- Yes it does use videoCapture API of opencv under the hood, but this is handled by https://github.com/alexeyab/darknet , the Neural network framework we use, you can't change this easily.. also Gstreamer pipeline works as we have OpenCV compile with Gstreamer support, but from docker `nvarguscamerasrc` won't work
- Also since 2.1 , we support inputting the camera as `-c 1` or `-c 0` (which is the default for darknet command) that doesn't use Gstreamer. You maybe can try to set `usbcam_no_gstreamer` for VIDEO_INPUT.

If this doesn't work, I think the way forward would be to try a logitech cam.

shrthprsd, 2019-11-16:
Thanks @tdurand . It works with the Logitech C920 camera. Appreciate your help.

closed 2019-11-16 by shrthprsd