opendatacam/opendatacam #151: gstreamer pipeline for rtsp x265 IP Cam

jonwilliams84, 2019-11-26:
Hi,

Great work so far, kudos!

I am trying to use a custom pipeline for a Hikvision x265 rtsp IP CAM in a bid to get better performance from a Jetson Nano.

With the standard 'remote_cam' VIDEO_INPUT I get approx 10fps for this feed. However when I run the following pipeline natively (and with deepstream) the nano gets 20fps (which is the same rate as the camera FTR). I have also run this and another feed concurrently using the 'tiled' source sample from deepstream and I get the 20fps and roughly 7-15% CPU usage.

`gst-launch-1.0 rtspsrc location=rtsp://user:pass@x.x.x.x:554/ch1/main/av_stream ! rtph265depay ! queue ! h265parse ! nvv4l2decoder ! nvvideoconvert ! "video/x-raw(memory:NVMM),format=RGBA" ! nvegltransform ! nveglglessink sync=False window-x=0 window-y=0 window-width=1080 window-height=720`

So, I have modified my config.json, adding a new VIDEO_INPUT_PARAM called "rtsp_x265" and set the pipeline to:

`rtspsrc location=rtsp://admin:123456789abc@192.168.254.7:554/ch1/main/av_stream ! rtph265depay ! queue ! h265parse ! nvv4l2decoder ! nvvideoconvert ! "video/x-raw(memory:NVMM),format=RGBA" ! nvegltransform ! appsink`

However when I attempt run with the input set to this new input it complains with error:

Loading weights from yolov3-tiny.weights...Done!
nvbuf_utils: Could not get EGL display connection
nvbufsurftransform: Could not get EGL display connection
nvbuf_utils: Could not get EGL display connection
nvbufsurftransform: Could not get EGL display connection
nvbuf_utils: Could not get EGL display connection
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device

(darknet:75): GStreamer-CRITICAL **: 22:10:00.226: gst_element_get_state: assertion 'GST_IS_ELEMENT (element)' failed
VIDEOIO ERROR: V4L: device rtspsrc location=rtsp://user:pass@x.x.x.x:554/ch1/main/av_stream ! rtph265depay ! queue ! h265parse ! nvv4l2decoder ! nvvideoconvert ! appsink: Unable to query number of channels
2019-11-26T22:10:15.862+0000 I COMMAND  [ftdc] serverStatus was very slow: { after basic: 499, after asserts: 1791, after backgroundFlushing: 2268, after connections: 2541, after dur: 2541, after extra_info: 2541, after freeMonitoring: 3020, after globalLock: 3199, after locks: 3522, after logicalSessionRecordCache: 3522, after network: 3600, after opLatencies: 3630, after opReadConcernCounters: 3707, after opcounters: 3707, after opcountersRepl: 3707, after repl: 3849, after security: 3859, after storageEngine: 4130, after tcmalloc: 4247, after transactions: 4509, after transportSecurity: 4509, after wiredTiger: 4850, at end: 4907 }
Restart YOLO

I have tried various iterations of this pipeline but no joy.

Any ideas?

Thanks in Advance!

tdurand, 2019-11-27:
Hello, 

Thanks ! From our test Jetson nano should be able to have around 15 FPS with Opendatacam and tiny YOLO.

I'm not an expert in Gstreamer pipeline so I won't be able to help you on the pipeline itself unfortunately, but you method is correct. 

Something you can do to debug it faster would be to [Install opendatacam without docker](https://github.com/opendatacam/opendatacam/blob/master/documentation/USE_WITHOUT_DOCKER.md) , and try to launch darknet directly with your pipelines.. so it is faster to debug, for example: `./darknet detector demo cfg/voc.data cfg/yolo-voc.cfg yolo-voc.weights "v4l2src ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink" -ext_output -dont_show`

And if this doesn't work also while using directly darknet, try the nvidia forum, there are some Gstreamer experts there that might be able to help you.

Thibault

tdurand, 2020-01-28:
closed due to inactivity

closed 2020-01-28 by tdurand