opendatacam/opendatacam #152: reverse proxy

jonwilliams84, 2019-11-26:
Hi
I am trying to get this running behind a proxy (traefik + nginx) for oauth forward auth...however in doing so I am getting a 5xx error.

My nginx config is as follows:

```
upstream app-upstream {
    server jetson-nano:8080;
}

server {
    listen 8085;
    server_name ~.;
   
    location ^~ / {
        proxy_pass http://app-upstream;
        proxy_set_header       Authorization "";
        proxy_set_header        Host $http_host;
    }

}
```

Any ideas?

Ta

Jon

tdurand, 2019-11-27:
It might be related to #126  , I did try to fix it but I didn't get feedback . 

Would be great if you can install opendatacam without docker and try this PR: #128

tdurand, 2020-01-28:
closed due to inactivity

closed 2020-01-28 by tdurand