opendatacam/opendatacam #154: Evaluate stand-alone-camera + streaming

florianporada, 2019-11-27:
I stumbled upon this repo https://github.com/dataplayer12/homesecurity/ which uses the jetson for CCTV system.
Idea: use a raspberry pi zero with a pi-camera-module could be a simple solution for uses cases like: Hardware inside and camera module outside.
- more secure since the pi only costs a fraction of the jetson
- no cable connection between board and camera
- option to connect more than one camera (if processing power can handle it)

The setup of the pi is rather easy since it only has to connect to the jetson board and provide a web-stream.
e.g. https://github.com/dataplayer12/homesecurity/blob/master/common/stream_video.py

I could do a POC and share the insights. As far as I know we're already capable of piping a video stream into the odc server right?

cc @tdurand @b-g

tdurand, 2019-11-27:
Yes, if you change the remote_cam in config.json with the URL of the video stream it should work: https://github.com/opendatacam/opendatacam/blob/master/config.json#L12 

Regarding the setup, I think @b-g had a cheap DIY IP camera setup that didn't even require a raspberry pi.

b-g, 2019-11-27:
Yes! 

The **Xiaomi Xiaofang 1S** (only old Version with T20 = $20) ... in combination then with crazy cool custom OS: https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks

florianporada, 2019-11-27:
Wow then this scenario is basically already covered. 🎉
Then I think it's only a documentation task, right?. 🙂

shams3049, 2019-11-27:
How do you get one in Germany? the website looks to only offer to US.

florianporada, 2019-11-27:
@shams3049 you mean the Xiaomii Cam? 
This ebay-shop seems to have it. 
https://www.ebay.de/itm/Xiaomi-Xiaofang-1S-Haushalts-Smart-IP-Kamera-1080P-Ir-Cut-Nachtsicht-Cn-/283635908891

shams3049, 2019-11-28:
I just have natural aversion towards ebay, so never checked there.  thanks for the link

b-g, 2019-11-29:
Hi all! I've bought mine at [aliexpress](https://www.aliexpress.com/store/1980403?spm=a2g0s.9042311.0.0.3a5c4c4d7oxCKX) ... shipping was 2-3 weeks. Make sure to carefully check whether the Xiaomi Xiaofang 1S works with https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks (e.g. only old Version with T20). 

Also I've never tested it with OpenDataCam ... but the Xiaomi-Dafang-Hacks turns it to proper IP cam. It should probably work.

(I can give it a try as I currently don't have access to the cam. Sorry!)

shams3049, 2019-11-30:
I bought two different models too, but they arrive third week of December. So would be quite late to try them out this year. How about your? When do you get to try them out?

b-g, 2019-11-30:
Hi @shams3049, I'm based in the US until Jan 2020. Won't happen on my end this year, either :) Sorry!

CLFPosthumus, 2020-07-14:
I have purchased below camera, flashed it with the custom RTSP firmware (bought the Wyze variant, they provided RTSP firmware). Although the camera is perfect for my use case (handy magnetic base, RTSP enabled) it only streams 1080p to RTSP.
It only allows for changes in bitrate, not actual resolution. 

This causes major issues with opendatacam (1080p video input not possible on jetson nano). 
TP-link Tapo C200 streams in both 1080p and 360p, which would work (not tested, but the docs point out that 360p should be fine). But it's not as small and convenient to mount as the Wyze cam. 

[image]

florianporada, 2020-07-17:
Hi @CLFPosthumus , 
you could try to capture the stream of the camera via gstreamer first, convert it to the desired resolution and then pipe it to the odc.
I'm not 100% sure if this works but you could try it.

Quick google search brought me this:
https://stackoverflow.com/questions/25840509/how-to-save-a-rtsp-video-stream-to-mp4-file-via-gstreamer

tdurand, 2021-05-22:
closing this.. I guess remote cam via RTSP etc etc.. is documented already..

closed 2021-05-22 by tdurand