opendatacam/opendatacam #156: Run opendatacam without docker-Reg

aswinkumar2019, 2019-11-29:
Hello guys,I think the instructions to install opendatacam without docker image is not updated.
Please add it soon.
I would like to make some modifications to the code for use in our project.
Atleast, please provide the instructions to modify the docker image for our own project.

shams3049, 2019-11-29:
Here is a script that I modified from V1 for opendatacam,, I am going to upload the final version to the project sometime next week, hopefully it helps in the mean time: 

#!/bin/sh
################################################################################
# Script for installing opendatacam on Ubuntu 18.04
################################################################################

set -e

echo "-----------------------------------------------------------"
echo "----             start basic package setup             ----"
echo "-----------------------------------------------------------"

#--------------------------------------------------
# Update ubuntu packages
#--------------------------------------------------
echo "---- Update Server ----"
cd ~
sudo apt-get update
sudo apt-get upgrade -y

#--------------------------------------------------
# Install basic packages
#--------------------------------------------------
echo "---- Install cURL, git-core, nano, build-essential, hostapd, dnsmasq ----"
sudo apt-get install curl git-core nano build-essential -y
#echo "---- Experimental packages (hostapd dnsmasq) for hotspot config  ----"
#sudo apt-get install hostapd dnsmasq -y

#--------------------------------------------------
# Add sources for ffmpeg v3 and nodejs v12
#--------------------------------------------------
echo "---- Add sources for ffmpeg v3 and nodejs v12 ----"
curl -sL https://deb.nodesource.com/setup_12.x | sudo -E bash -
sudo add-apt-repository ppa:djcj/hybrid
sudo apt update
#--------------------------------------------------
# Install nodejs v8
#--------------------------------------------------
echo "---- Install nodejs v12 ----"
sudo apt-get install -y nodejs

#--------------------------------------------------
# Install nodejs ffmpeg v3
#--------------------------------------------------
echo "---- Install ffmpeg v3 ----"
sudo apt-get install -y ffmpeg

echo "-----------------------------------------------------------"
echo "----           finished basic package setup            ----"
echo "-----------------------------------------------------------"
echo "-----------------------------------------------------------"
echo "----         start darknet-net installation            ----"
echo "-----------------------------------------------------------"

#--------------------------------------------------
# Install libwsclient
#--------------------------------------------------
echo "---- Install libwsclient ----"
cd ~
git clone https://github.com/PTS93/libwsclient
cd libwsclient
./autogen.sh
./configure && make && sudo make install

#--------------------------------------------------
# Install liblo
#--------------------------------------------------
echo "---- Install liblo ----"
cd ~
wget https://github.com/radarsat1/liblo/releases/download/0.29/liblo-0.29.tar.gz --no-check-certificate
tar xvfz liblo-0.29.tar.gz
cd liblo-0.29
./configure && make && sudo make install

#--------------------------------------------------
# Install json-c
#--------------------------------------------------
echo "---- Install json-c ----"
cd ~
git clone https://github.com/json-c/json-c.git
cd json-c
sh autogen.sh
./configure && make && make check && sudo make install

#--------------------------------------------------
# Install darknet-net
#--------------------------------------------------
echo "---- Install darknet-net ----"
cd ~
git clone https://github.com/opendatacam/darknet.git
cd darknet
echo "---- Download weight files ----"
wget https://pjreddie.com/media/files/yolo-voc.weights --no-check-certificate
make

echo "-----------------------------------------------------------"
echo "----        finished darknet-net installation          ----"
echo "-----------------------------------------------------------"

echo "-----------------------------------------------------------"
echo "----       start open data cam installation            ----"
echo "-----------------------------------------------------------"

#--------------------------------------------------
# Install open-data-cam
#--------------------------------------------------
echo "---- Install opendatacam ----"
cd ~
echo "---- Install node modules: pm2, next ----"
sudo npm i -g pm2
sudo npm i -g next
git clone https://github.com/opendatacam/opendatacam.git
cd opendatacam
echo "---- write config.json file ----"
echo '{"PATH_TO_YOLO_DARKNET":"/home/shams3049/darknet"}' > config.json
echo "---- run install and build script ----"
npm install
npm run build
echo "---- configure open-data-cam to start on boot ----"
sudo pm2 startup  
sudo pm2 start npm --name "opendatacam" -- start
sudo pm2 save

echo "-----------------------------------------------------------"
echo "----      finished open data cam installation          ----"
echo "-----------------------------------------------------------"

aswinkumar2019, 2019-11-29:
Thank you very much for your help with fast response.
it will be very useful for me.

closed 2019-11-29 by aswinkumar2019

tdurand, 2019-11-29:
Here is the documentation: https://github.com/opendatacam/opendatacam/blob/master/documentation/USE_WITHOUT_DOCKER.md

aswinkumar2019, 2019-11-29:
Thank you guys thanks a lot

tdurand, 2019-11-29:
And to create your docker image: https://github.com/opendatacam/opendatacam/blob/master/documentation/jetson/CREATE_DOCKER_IMAGE.md