opendatacam/opendatacam #157: Pull data and use with ros

aswinkumar2019, 2019-11-30:
hello guys,
I am new to nodejs,i need to pull the dashboard data like number of bikes,cars,total number of vehicles and send to an ros(robot operating system) node or just pull the data for use in another python code.
Is there any way for doing that,please help
Thanks in advance

tdurand, 2019-11-30:
Hello

You can have a look at the API: https://opendatacam.github.io/opendatacam/apidoc

aswinkumar2019, 2019-11-30:
Is there any documentation for using this api?
Thanks in advance

tdurand, 2020-01-28:
The only documentation using this API would be to look into the source of the actual opendatacam code, which uses this same API.

We don't have the ressources to build more tutorial right now, sorry

closed 2020-01-28 by tdurand