opendatacam/opendatacam #160: Processing multiple files or stream at the same time

rantgithub, 2019-12-06:
Hello

I am testing open data cam for processing some files and live stream from ip cameras.

Based on the default configuration, only 1 file can be processed or 1 stream at the time on one instance of open data cam, based on my understanding.

I would like to know if is possible to have open data cam set to process under the same instance 2 videos, or 1 video and 1 stream or multiple streams.

Thank you

tdurand, 2019-12-06:
Hello, 

Thanks for your message.. But sorry, no this is not possible, only 1 video or 1 stream at a time. 

we might be able to support this once we switch to Deepstream https://github.com/opendatacam/opendatacam/issues/112

rantgithub, 2019-12-07:
Hi

Question, 

As possible workaround using virtualization/containers or similar.

I can run different instances of  ODC the same machine, but to access different browser i need 2 different webport server, to process 2 files.

For example

I can have ODC#1, running on machine 1, on webserver 127.0.0.1:8080

and

ODC#2, running on machine 2, on webserver 127.0.0.1:9000

How I can change the port on the webserver ?, if i want for this purpose or for security to access a different port ?

Thank you

tdurand, 2019-12-09:
Are you running on a jetson or on a desktop computer ? 

I think you can change the port of the webserver that docker is using by changing this file (that is pulled in the directory you ran the install command): https://github.com/opendatacam/opendatacam/blob/master/docker/run-jetson/run-docker.sh#L42 

For example in order to map 8080 to port 9000 it should work like this: 

`docker run -p 8080:9000 -p 8090:8090 -p 8070:8070 --privileged $DOCKER_VOLUMES -v /data/db:/data/db $@`

rantgithub, 2019-12-09:
1. I am running ODC on a desktop and without docker.

To follow up on your command, if I want to change all the ports, to 9000, 9010 and 9020 will be something like this ?

docker run -p 8080:9000 -p 8090:9010 -p 8070:9020 --privileged $DOCKER_VOLUMES -v /data/db:/data/db $@

2. if the above command is right for docker, how I can change the ports on the desktop version without docker ?

Thank you

tdurand, 2019-12-09:
1. Yes this should work

2. Without docker it would be a bit harder as you would need to change directly in the code as it is not setup as configurable variable

You should modify those lines ( I may forget some):

- https://github.com/opendatacam/opendatacam/blob/db86d2f63a1c6baff4c2cc435e100f8216985ac7/server/processes/YOLO.js#L34

-  https://github.com/opendatacam/opendatacam/blob/master/server/Opendatacam.js#L519

- https://github.com/opendatacam/opendatacam/blob/master/server.js#L146

- https://github.com/opendatacam/opendatacam/blob/db86d2f63a1c6baff4c2cc435e100f8216985ac7/package.json#L7

rantgithub, 2019-12-09:
Ok

1. I know what needs to be changed, I am doing some tests for that and I will follow up.

2. Following the same thinking  process, how i can change the default port for the mongo db ?

tdurand, 2019-12-09:
For mongodb, this is configurable directly in the config.json: https://github.com/opendatacam/opendatacam/blob/master/config.json#L72

rantgithub, 2019-12-09:
Hi

I reviewed several files and was able to change the port for webserver

but for mongodb, as a test,  i changed from 27017 to 37017 and i am getting these errors

{ MongoNetworkError: failed to connect to server [127.0.0.1:37017] on first connect [MongoNetworkError: connect ECONNREFUSED 127.0.0.1:37017]
    at Pool.<anonymous> (/home/ana1/opendatacam/node_modules/mongodb/lib/core/topologies/server.js:431:11)
    at Pool.emit (events.js:198:13)
    at createConnection (/home/ana1/opendatacam/node_modules/mongodb/lib/core/connection/pool.js:559:14)
    at connect (/home/ana1/opendatacam/node_modules/mongodb/lib/core/connection/pool.js:973:11)
    at makeConnection (/home/ana1/opendatacam/node_modules/mongodb/lib/core/connection/connect.js:39:11)
    at callback (/home/ana1/opendatacam/node_modules/mongodb/lib/core/connection/connect.js:261:5)
    at Socket.err (/home/ana1/opendatacam/node_modules/mongodb/lib/core/connection/connect.js:286:7)
    at Object.onceWrapper (events.js:286:20)
    at Socket.emit (events.js:198:13)
    at emitErrorNT (internal/streams/destroy.js:91:8)
  name: 'MongoNetworkError',
  errorLabels: [ 'TransientTransactionError' ],
  [Symbol(mongoErrorContextSymbol)]: {} }
(node:27331) [DEP0001] DeprecationWarning: OutgoingMessage.flush is deprecated. Use flushHeaders instead

i think, this on maybe this more changes

any suggestions?

thank you

tdurand, 2019-12-09:
Ah yes, this just change which port Opendatacam is looking for mongodb but you need to start mongodb on that port, maybe this will help: https://medium.com/mongoaudit/how-to-change-mongodb-default-listening-port-27017-92e35f65670e

rantgithub, 2019-12-09:
Ahh ok

I will work on this, because I can have different instances with 1
database or multiple databases

Tx

tdurand, 2019-12-09:
Yes, also you could change for one instance the name of the database here: https://github.com/opendatacam/opendatacam/blob/master/server/db/DBManager.js#L23

rantgithub, 2019-12-10:
When I changed the name of the database, ODC starts, but nothing shows up when you try to see the mongodb
page is blank.

something else to change ?

tdurand, 2019-12-10:
Did you try to record something ?

rantgithub, 2019-12-12:
yes

as i mentioned all works from main screen, but on the database portion the screen is just white, you dont see anything there

tdurand, 2019-12-12:
Hum, I just tried, and if I change 

`let db = client.db('opendatacam')`

to 

`let db = client.db('opendatacamalternative')`

I restart opendatacam, I then have empty data but if I record something it works.

Not sure what would be your issue

rantgithub, 2019-12-12:
I am redoing a new instance to test, I will follow up.

Working in all these, the workaround is an option at the expense of much more usage of cpu and GPu.

I was comparing with a deepstream feed i have, and the difference is notorious.

Based on the initial comments you made, is any part of ODC you have tested or done with deepstream ? 

happy to test or help on that :)

just

shams3049, 2019-12-12:
Hi, 

To my knowledge deepstream is a work in progress, I am working on making a pipeline with deepstream too, so far nothing positive yet.

tdurand, 2019-12-12:
Regarding deepstream, we definitely thing this is the way forward, you can follow progress here: https://github.com/opendatacam/opendatacam/issues/112  .

rantgithub, 2019-12-12:
Thank you for the feedback.

:)

closed 2020-01-28 by tdurand

gcervantes2005, 2023-06-05:
> Hello
>
> I am testing open data cam for processing some files and live stream from ip cameras.
>
> Based on the default configuration, only 1 file can be processed or 1 stream at the time on one instance [...]

We are trying to merge 4 streams using ffmpeg and forward it to process on one instance. We process live video feed on rtsp.