opendatacam/opendatacam #161: RTSP stream not processed

froststrm, 2019-12-06:
I'm trying to open a RTSP stream by modifying the config.json file:

"VIDEO_INPUT": "remote_cam",
...
"remote_cam": "rtsp://10.0.0.44:554/user=admin_password=_channel=6_stream=0.sdp ! video/x-raw, framerate=30/1, width=1280, heigth=720 ! videoconvert ! appsink"

My platform ist jetson nano. I can see the video stream on port 8090, but when I try to open port 8080 it just gives me "Error with message send by YOLO, not valid JSON" in the log.

USB cam and the MP4-file demo are running fine.

Any ideas? Thanks in advance!

tdurand, 2019-12-06:
Hi, 

For RTSP stream it should work directly with:

`"remote_cam": "rtsp://10.0.0.44:554/user=admin_password=_channel=6_stream=0.sdp"`

froststrm, 2019-12-07:
Today I tried it again and waited a little longer for the RTSP stream getting processed. Now it works.
One problem is now, that the stream resolution is 1280x720, which is too high for proper processing. It there any option to resize the stream before input to neural network?

tdurand, 2019-12-09:
You should change it in the input gstreamer pipeline directly, but I'm not an expert on those commands so I'm not sure how it should work for the RSTP stream

For the USB cam I know this works: `"v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink"` , where you specify a different width / height resolution that is supported by the camera

closed 2019-12-11 by froststrm