opendatacam/opendatacam #162: Add specific cuda/cudnn tag version to Dockerfile

b-g, 2019-12-09:
I'm wondering in terms of long term support whether it would not be better to add the cuda/cdnn version tag to our dockerfiles?

e.g. https://github.com/opendatacam/opendatacam/blob/master/docker/run-nvidia-docker/Dockerfile#L1

`FROM nvidia/cuda`

becomes

`FROM nvidia/cuda:10.2-cudnn7-runtime-ubuntu18.04`

as currently it is simply `latest` ...

shams3049, 2019-12-09:
latest is always with cool new features, but will definitely struggles with stability.  So the question is in long term are you going for stability or features. I personally prefer the latest tag.

tdurand, 2019-12-09:
Yes, I mean at some point latest might break, and it would prevent it by setting a fixed version... Not sure if we get any perf improvement of having always the latest CUDA.

tdurand, 2020-04-28:
This will be done with this : https://github.com/opendatacam/opendatacam/issues/178

closed 2020-04-28 by tdurand