opendatacam/opendatacam #163: Use opendatacam at night-Reg

aswinkumar2019, 2019-12-10:
Hello guys, 
I would like to use opendatacam in national highways.
Will this work with live data which may be taken during night with light there or, maybe taken with night vision camera.
Will opendatacam work in these sceneries???
Has anybody tested it???
Regards,
Aswin Kumar

aswinkumar2019, 2019-12-11:
Hello guys,
I tried opendatacam with video of national highways at night(https://drive.google.com/open?id=1Cchjs7TaoD64qf6EdZ1sQ174Dqwst9h3) and got results as shown in the video attached in this link.I find that opendatacam did not perform well on night videos.

Is the model not trained for night time deployment or,it may be issue of clarity of video or,position at which video is taken?

Please provide your views on this.
Regards,
Aswin Kumar

tdurand, 2019-12-11:
Hi, 

We didn't try much with night time footage so I can't say if this is the result you should expect.

Maybe you should investigate to see if some YOLO weights are trained with specific night time footage.  We provide a way to use third party yolo weights: https://github.com/opendatacam/opendatacam/blob/master/documentation/CONFIG.md#use-custom-neural-network-weights 

Good luck and don't hesitate to share your finding with night time footage.

rantgithub, 2019-12-11:
Hi 

The default yolo2 or 3, wont perform well at night.

For specific situations, you need to train your own model.

I have tried at night, and depends of a lot of factors including angle and resolution of camera.

What I suggest is perform analysis of your own situation and create a dataset for this and train a new model accordingly.

shams3049, 2019-12-12:
Hi,

I am using a low light sensitive camera from e-con, which is giving me quite similar results as the day time. Maybe you can look into e-consystem tech.

aswinkumar2019, 2019-12-12:
@shams3049 can you please specify the model of e-con camera you have used?

shams3049, 2019-12-13:
@aswinkumar2019 I am using multiple camera from e-consystems, do not have any comparison on which one works better. 

one Model I am using is https://www.e-consystems.com/3MP-Jetson-TX1-Camera-board.asp 

be aware e-consystems is really slow in there software production. For most of the sensor its a nightmare to get them running on Jetpack 4.2 as officially they are not yet support. So some manual work is expected before being able to run them.

closed 2019-12-27 by aswinkumar2019