opendatacam/opendatacam #165: Display synchronization issue between video and yolo detection boxes

rantgithub, 2019-12-17:
Hi

Performing several testing with some videos, when the connection is slow between the local webserver and the local client, there is a big delay between the detection boxes and the video itself.

the yolo detection boxes get in front (most of the time) than for instance the object detected.
meaning a white empty box is in front of a car(without detection box)

This causes a synchronization issue between the detection and the video itself, that doesn't allow you track properly objects because gets very confusing (you dont know if the box x1 is from the car y1 or the car y2, due to the gap is big)

Even the counting still works, under this situation there is no way to validate it.

just to check if you have seen this issue before and something you can think can address this situation

Thanks

tdurand, 2019-12-17:
Yes, this is a know issue with the MJPEG stream.. In my experience lowering the resolution of the webcam feed fixes this issue. 

We tested only this tool with a local connection (wifi) to the place where OpenDataCam is deployed.. In order to have better performance other the network, I think the solution would be to encode the webcam stream in something else like HLS but this brings lots more complexity.

rantgithub, 2019-12-17:
Just to clarify

You mean to encode before comes to ODC or actually inside of ODC chnging the code ??

because there are some converters/servers for HLS, and the stream can be processed prior sent it to ODC into HLS, if this resolves the issue.

tdurand, 2019-12-17:
- For lowering the resolution I mean outside ODC , we have good results using 640x360 over network

- I mean inside of ODC, to change MJPEG you need somehow to convert the MJPEG stream provided by darknet into something else, an also change how the front-end consumes it. And this is also just an idea, not even sure this will work. Also this is something that might change with Deepstream.

rantgithub, 2019-12-17:
sorry, still some confused

Lowering the resolution, you mean the output video from ODC? or the resolution of the incoming file you setup under the config.json file ?

and if is the output file ,  are there some parameters inside of ODC to lower resolution ?

tdurand, 2019-12-17:
no problem. I mean the resolution of the incoming stream you setup in config.json .

closed 2020-01-28 by tdurand