opendatacam/opendatacam #166: Counting lines position issue

rantgithub, 2019-12-18:
Hi

Performing more testing there is an issue with the counting lines.

If have a video or live stream and i draw a line for counting on x1,y1 to x2,y2  the line remains in position fine, but when I start the recording process for the counting, the lines moves to x3,y3 to x4,y4, even if the video hasnt move. 

the counter works still fine, for what appears the original location of the line and not the current visual location, but this get very confused, because if the objects are closer is very hard to validate if the object was counted or not, because the detection or because if the line is out of position 

this is very notorious, specially if you see the the video, in another device, even when the image and monitor has the same resolution.

Doing some many test, I think the issue is related to the resolution of the screen of the PC/laptop relative to the video.

My question here is if there is a resolution setting that needs to be adjusted when the video is seen in another device, or this done automatically, in order for the lines to be always at the same position regardless if is doing or not the counting.

Thank you

tdurand, 2019-12-18:
Hi, 

I remember having fixed a bug with this same problem. What happen is when you draw the line and then start recording it then gets drawed on another canvas.. maybe there is still an issue.

Can you reproduce it all the time or just some times ? Did you try on other browsers ?

rantgithub, 2019-12-18:
the issue is reproducible very easy and all the time

the lines "displaced" a little bit, to either up/down and/or left/right when you start/stop the counting process and happens in all browsers.....

i think the problem is the positioning of the lines, and for what you mentioned is like the other canvas is not getting the same parameters before/after the count starts

tdurand, 2019-12-18:
Could you record a video of the bug ? I just add a quick test and wasn't able to reproduce

rantgithub, 2019-12-18:
sure, let me work on that

i have a question, i remember looking into the code and found  "resolution" in 1-2 files  and I think the default was 1280x720.....

can you tell me is there is a fix resolution parameter that i have to check/adjust

tdurand, 2019-12-18:
This is the default resolution for the Simulation mode, a mode we use for development.

But what happen when you run in normal mode is that we get the resolution used by YOLO here: https://github.com/opendatacam/opendatacam/blob/master/server.js#L69

So, no parameters to check.

What would be helpfull along with the a video of the bug would be if you can give me:

- The video resolution used by Opendatacam : https://opendatacam.github.io/opendatacam/apidoc/#api-Webcam-Resolution

- The resolution of your browser window , you can get this information by opening dev tools and type `window.innerHeight `, `window.innerWidth` in the console

rantgithub, 2019-12-19:
I think i have find how to duplicate the issue and an explanation

if you add a line and after you try to delete, the canvas for delete shows the original position that shows  exactly the offset on the screen

the resolution i am testing is with 3840x2160

if you draw the line on 1920x1080 screen and move it to the 3840x2160 or viceversa, you can see clearly how the lines are not keeping proportion

if you refresh the screen and or press the recycle icon to try to delete, you see the 2 set of lines and the one of the recycle is the "good one" and the one using for counting.

the same happens when you draw the lines and they move. 

There is a difference between the relative position of the original lines and the visible lines on the screen

see the file attached

[lines.pdf](https://github.com/opendatacam/opendatacam/files/3981863/lines.pdf)

rantgithub, 2019-12-19:
hi

to add info, i think the problem happens only  after you start and record a video.

i cant upload videos here, but as test I start the counting process and line moved to a different position, i press the STOP and see how the line moves back to original position

rantgithub, 2019-12-19:
Hi

just to add a resolution from all testing i did, until find a way to fix this

once you behavior is active, the "fix" is to delete all the lines, refresh (f5) the browser, and add new lines, and press the recording button.

there is clearly some kind of adjustment that remains active until all is deleted, coming from the recording modules.

tdurand, 2019-12-19:
Ok, yes I was able to reproduce, will try to fix soon

rantgithub, 2019-12-19:
good

as additional comment, if you refresh, the first time the screen, before add any lines, always 
 works fine, meaning the lines dont move.

but still remains the issue on changing resolution to a different device like other pc, or phone, meaning it is not auto adjusting the lines according new resolution size.

tdurand, 2019-12-19:
Ok, just fixed it on master: https://github.com/opendatacam/opendatacam/commit/92bf34e4870c4080b763520200e9301f501c5047 

That said, generating a new release (generating new docker images etc etc..) is something that takes lots of manual work, so you can fix it only by installing opendatacam without docker with latest code for now

tdurand, 2019-12-19:
feel free to reopen the issue if it is still not fixed after you try this

rantgithub, 2019-12-19:
the scale is working fine now,

but still, the lines are offset.

i open the video, draw the lines and they moved to the left. (a diagonal counter)

i erased the lines, refreshed the browser/video with F5, and now the lines are fine, 

not sure if is something with the browser, or refresh is "part of updating" the resolution and display properly

tdurand, 2019-12-19:
Thanks for testing, I tried again but didn't find a way to make it draw badly now.

Could you list the exact order of action and maybe associate some screenshot on each step in order for me to be able to reproduce ?

rantgithub, 2019-12-20:
Hi

I am not sure it was a browser glitch, but I reset all, including the pcs and after that, the issue appears got resolved, because I cant duplicated anymore.

But doing all tests, I realized that now the objects are not changing the color of the detection box when crosses the counting line as before.

The counting is correct as well the lines, but all is either white or black, depending the mode you are set.

it is helpful to see what line the objects crosses.

rantgithub, 2019-12-20:
Hi

I saw very unseen behavior, on latest  test, colors not changing, FPS at 10000, nd recording time completely showing 300:00:00. It looks like something didnt  load right

In reset all again and all is working

i will keep testing and let you know

tx

rantgithub, 2019-12-20:
i have done now lot of testing and what appears to happen is that depending how powerful is the GPU, the time after crossing the line for an object goes from almost instantaneous to 1-2 seconds. 
is this by design or there is an specific time/frames to display the color of the object after crossing the line ?

rantgithub, 2019-12-20:
on a related but different topic/question,

when there is a busy intersection and a line is set, depending the angle of the camera sometimes cars and people crosses the line and increases the counter, regardless of object class and direction.

if you have a 4 lane intersection, people can go east-west or west east as well the cars. with 1 counter, there is no way to know what is what as total counters.

will be very useful  to have a line that only counts  people or cars or any of classes, meaning to draw the line and associate/ask the class to count and direction.
 
is this the same module for counting to put some kind of logic for this ?

thank you

tdurand, 2019-12-20:
- No, there is no delay, what might happen is indeed that the browser gets busy rendering and delay render... what could help is to open a smaller browser window...

- There is an issue for this: https://github.com/opendatacam/opendatacam/issues/122

I'll close this one, feel free to reopen if you have further issue

rantgithub, 2019-12-21:
Hi
I am not seeing how to reopen this, looks like i dont have that capability.

But doing additional testing, one more small issue

If you draw a line on a mobile unit, phone, ipad, Samsung tablet, etc, all the line disappears from the screen, and you need to refresh the screen for the old lines and the current one to be displayed.

The scale and counting works fine, but anytime a line is added, the refresh is needed otherwise the screen doenst include any lines.

When you draw a line on a regular monitor, all works fine, meaning the old lines remain and the new line is added.

In both cases counting and scale are working fine.

This is an issue only affecting the mobile devices.

rantgithub, 2020-01-07:
Hi,

hope all is well

Doing several tests, I have found that sometimes ODC hangs or freezes the machine/jet son completed and I have to press CTRL_C or killall to free the resources or sometimes physical restart is the only option.

I was looking the code and I saw a couple of functions to clean the ODC processes but I tested and it is not working for me. (like process exit)

The question here, is if I want to add a button just to fully kill and restart ODC in case this happens, can you please tell me what function/method or similar should I call to close all process and start  application again.

Thank you

tdurand, 2020-01-07:
Hi, 

When this happen, it is the darknet process (YOLO) that freeze. I'm not sure if this will work, but you can try to kill it with: https://github.com/opendatacam/opendatacam/blob/master/server/processes/YOLO.js#L111 (we use https://www.npmjs.com/package/forever-monitor to monitor the darknet process)

Also I'm thinking if you are running on Linux and manage the node.js with Docker or PM2 to auto-restart you might be able also to directly kill the whole node.js app like this `process.kill(process.pid, 'SIGINT')` (not tested)

Hopefully once we integrate with deepstream we won't have anymore those issues #112

rantgithub, 2020-01-07:
Hi

I actually saw that and I tested calling YOLO.stop(), opendatacam.clean() and process.on(exit ..  and doesn't stop all the time.

I think the solution is the kill the whole node.app and start it again.

I tried the process.kill and didnt stop it either, my question the process.pid will be the 'darknet" process or the actual pid of process.

suggestions?

tdurand, 2020-01-07:
the pid is given by the Operating system I guess, this seems a related issue: https://stackoverflow.com/questions/27827983/forever-monitor-kill-child-with-specific-uid

I'm not very familiar with killing stuff with node.js, so I'm afraid I can't help lot more, it should be great if you can share back your solution once implemented

rantgithub, 2020-01-07:
ok sure

i will provide feedback.  The idea is to have it the app doing it, instead of manually.

rantgithub, 2020-01-07:
on the line, issue,

did you have a time to see the "refresh" issue when the lines are drawing on the phone/tablet?

tdurand, 2020-01-07:
For the line issue no, I didn't have time to investigate yet.. very little time to dedicate to it right now.

rantgithub, 2020-01-09:
hi

just to provide feedback for the kill process

one way this can be done is to use the process.kill(process.pid, 'SIGINT')

i added a routine to kill this and the .child process from jolo, in this way yolo can be restarted again, even manually or can be part of the foreever monitor used before

once is forcebly stopped the forever start it again.

rantgithub, 2020-01-09:
doing some analysis, 

when the objects is detected, we have the coordinates X,Y plus the W, H that defines the boxes and the objects are tracked based on the center of this box, X + W and Y + H and just calculated the middle with /2.

depending where is the object and how big, when comes closer to the camera, the center gets really high if the object is big, is comparison a small object.

If we can track and follow with the base of the box, meaning x+w/2 and just Y, the big object can fit better on the counting line and not other or not even been counted at all.

I modified the math to remove Y+H and just leaving Y, but I think this is been calculated in other places because i didnt see any changes on how the lines are plot for the pathfinder.

Do you know what else i  need to change  in order to track the objects by the center of the lower base of the square and not the center of the square to address the issue of scale with the big objects.

thank you

tdurand, 2020-01-09:
I think you should modify this line: https://github.com/opendatacam/opendatacam/blob/master/server/Opendatacam.js#L259

`let deltaY = - trackedItem.y - countingAreaProps.a * trackedItem.x - countingAreaProps.b;`

Maybe like this (not tested)

`let deltaY = - (trackedItem.y + trackedItem.h / 2) - countingAreaProps.a * trackedItem.x - countingAreaProps.b;`

rantgithub, 2020-01-09:
something to mentioned as feedback

doing the test to restart yolo, the console shows this line on the opendatacam.js file, that is a syntax error because the object doesnt have .id

const biggestTrackedItemIdThisFrame = trackerDataForThisFrame[trackerDataForThisFrame.length - 1].id; (line 244)

rantgithub, 2020-01-09:
i tested 

let deltaY = - (trackedItem.y + trackedItem.h / 2) - countingAreaProps.a * trackedItem.x - countingAreaProps.b; 

and yes it changes the counting behavior and initially appears worst (testing)

but the lines for the path on the screen are still been drawing by the center (x-w)/2 and (y-h)/2

i look the pathengine.js at

drawPaths (context, currentFrameData,  canvasResolution, originalResolution) {
    this.lastFrameData = currentFrameData.map((objectTracked) => {
      let trackedItemScaled = scaleDetection(
        objectTracked,
        canvasResolution,
        originalResolution

but still no success on drawing the path by the base and no center

suggestions?

tdurand, 2020-01-09:
The suggestion I made was to count things when the base is crossing the counting line and not when the center cross.

If you want to draw the path from the base it is indeed here: https://github.com/opendatacam/opendatacam/blob/master/components/canvas/engines/PathViewEngine.js#L33

Maybe something like this:

```javascript
this.drawLine(context, {
          pointA: {
            x: lastFrameTrackedItem.x,
            y: lastFrameTrackedItem.y + lastFrameTrackedItem.h / 2
          },
          pointB: {
            x: trackedItemScaled.x,
            y: trackedItemScaled.y + lastFrameTrackedItem.h / 2
          }
 }, color)
```

rantgithub, 2020-01-09:
thank you for the clarification, i was not sure if that was coming from the tracker alg

for this particular view, and after doing some testing, the count using  base wont work, because the objects are too big, but the "left" of the box will work (i think)

(i already draw it on the tracker at looks better)

but
because i dont see deltaX on the opendatacam.js, can you please tell me if  to count based on the left of the box, this will be the line

 let deltaY = - trackedItem.y - countingAreaProps.a * (trackedItem.x-trackedItem.w / 2) - countingAreaProps.b;

or

let deltaY = - (trackedItem.x - trackedItem.w / 2) - countingAreaProps.a * trackedItem.x - countingAreaProps.b;

thank you

tdurand, 2020-01-09:
I guess it will be the first one, but I mean I wrote this long ago I can't really remember all the logic without spending time to rethink everything...

rantgithub, 2020-01-10:
thank you. I was able to see behavior of count and tracker.

I have this issue, that i am not sure if is that the database got too big on mongo.

I have a database XX1 and when i start the record process, if you press the dataview button, you can see is recording. But when I  press stop, nothing shows on the view window, it is empty.

looking at logs, this is the error is been displayed.

(node:7651) UnhandledPromiseRejectionWarning: MongoError: Executor error during find command :: caused by :: Sort operation used more than the maximum 33554432 bytes of RAM. Add an index, or specify a smaller limit.
    at Connection.<anonymous> (/home/ubuntu/odc/node_modules/mongodb/lib/core/connection/pool.js:454:61)
    at Connection.emit (events.js:198:13)
    at processMessage (/home/ubuntu/odc/node_modules/mongodb/lib/core/connection/connection.js:368:10)
    at Socket.<anonymous> (/home/ubuntu/odc/node_modules/mongodb/lib/core/connection/connection.js:537:15)
    at Socket.emit (events.js:198:13)
    at addChunk (_stream_readable.js:287:12)
    at readableAddChunk (_stream_readable.js:268:11)
    at Socket.Readable.push (_stream_readable.js:223:10)
    at TCP.onStreamRead [as onread] (internal/stream_base_commons.js:94:17)

now, if I go to the DBmanajer.js and on line 23, I change the client.db name from XX1 to XX2, (to create a new db), now all works fine again. When you press the dataview, either during recording or liveview,  the data is display properly.

Do you have a parameter to limit size on each db that can be modified  or any suggestion how to connect back to XX1 ?

Thank you.

tdurand, 2020-01-10:
I think there is a mistaken on the index for the mongodb, please try to replace https://github.com/opendatacam/opendatacam/blob/master/server/db/DBManager.js#L29 with

```
// Create the index
recordingCollection.createIndex({ dateStart: -1 })
```

Because then we sort by dateStart and not dateEnd : https://github.com/opendatacam/opendatacam/blob/master/server/db/DBManager.js#L182

It might solve the problem, if it does please tell us so we fix this

rantgithub, 2020-01-10:
Indeed this fixed it.

Now makes sense, based on your explanation of the index.

I have the same problem in 2 more databases, and changing to dateStart is the solution. 

This just make me realize, a function/procedure to backup the database is needed.

Is there any process already for  backup ? or do you think  based on the current code an easy way to make a call to mongo to backup the database ?

After you a good number of counts, keeping a copy of the database is very important

tdurand, 2020-01-10:
Great, I fixed the code on master.

No there is not. I have no recommandation regarding this, you should be able to find plenty of ways to backup a mongodb database from node.js 😉

rantgithub, 2020-01-10:
ok.....thank you...

i will implement something and I will share 

Thanks

rantgithub, 2020-01-10:
one more question

on a live view and when for the drawing of the path, when detecting boxes around the objects, sometimes these boxes move a lot, specially on big vehicles, like trucks or buses, when the image is getting closer to camera, making the center also to move a lot, causing the path also to move a lot.

I have seen some alg to make the detection box more stable. What part calculates this boundary box, to see if we can make this more stable

Thank you

tdurand, 2020-01-12:
we get the bbox from the output of YOLO (darknet), that we input into our tracker: https://github.com/opendatacam/opendatacam/blob/master/server/Opendatacam.js#L236 https://github.com/opendatacam/node-moving-things-tracker to assign them an unique ID . 

We never change them

rantgithub, 2020-01-14:
Hi

Looking at the boxes I have observed behavior when the counting is performed.

For small objects, the counting works very well.

But for the big objects, the counting zone trigger, is not working as initially thought.

As example, I have 2 lines and you see a big truck approaching  . (Truck 1 and Truck 2 picture) that are on the left lane.

Even the center is passing the counting line, what appears to happen is that until the full box doesn't pass completely the line, is not detected/counted.

What this causes, is that all big objects on the left, are been counted on the right. 
(Please see Truck 3)

This is reproducible on different views.

Can you please clarify how the trigger increase the counter, because it was was my impression that once the center of the box, or any reference point crosses the line, the counter is increased and based on this, a possible solution for this issue.

Thank you
[image]
[image]
[image]

tdurand, 2020-01-14:
Hi, 

Yes the behaviour is when the center of the box crosses the line, it is counted. 

What seems to happen in your case is that the center of the truck box seems to be indeed slightly on the other lane, so it gets counted on the blue line instead of the green one...  But I guess it is counted when the center pass the blue line

The solution would be what we were discussing earlier, to count with the bottom center of the bbox instead of the center,  haven't you been able to change to this behaviour ? 

BTW we did choose the center because if the trajectory could be any direction.. but this use case might tell use that is should be configurable (one feature more 😄)

rantgithub, 2020-01-14:
I actually changed to the bottom on the box and tested as well and results were similar.

If you see the truck path, it clearly passes the counting line properly, but the counter doesn't counts on the correct zone.

Now, if you see on the second picture, i separated the lines to see if on the left line the truck will be counted, but still it was counted on the right. 

On this picture, any of the box actually touches the blue line when you seeing crossing,  but still only counts after the full boxes passes and still on the right side.

It is my understanding, that where ever the path crosses the line, should match with the counting. But again, that is not the case.

Reason I am asking to see if something else needs to be changed.

When only trucks are selected as classes, the count is very unreliable, because on this behavior.

tdurand, 2020-01-14:
Could you send me a 1 min footage (or less) of those trucks so I can try to reproduce ?

tdurand, 2020-01-14:
Also if you try to draw the line with a different angle, does it change ? Like this:

[image]

rantgithub, 2020-01-14:
yes...drawing the red line caused same behavior

what i observed, is that for these big objects, until the full box crosses line, the counter is not increased and because the current position is different that the "real crossing position", it counts the object on the different line.

[test.zip](https://github.com/opendatacam/opendatacam/files/4060566/test.zip)

Here is a 2 min file.....and you can reproduce this

tdurand, 2020-01-14:
I just did a test and it behave like I though it would, please see this video

[counting-test.zip](https://github.com/opendatacam/opendatacam/files/4060730/counting-test.zip)

So yes it is true that for big box we need to set the line a bit to the right, as you can see at the end of the video two paths are on the other lane (the one of the trucks), but it seems to count correctly when the center of the box crosses the line.

Not sure what is going wrong on your case, did you revert your changes ? Let me know

rantgithub, 2020-01-14:
Hmmmm...

It looks like i have different code.....because i put the lines exactly like you have on your video, and mine counts on the right.

i know opendatacam.js is the file that counts. anything else i have to check ?

tdurand, 2020-01-14:
yes in opendatacam.js where I told you is the code for counting

rantgithub, 2020-01-14:
i am checking what is the difference on the code, with another version i had

rantgithub, 2020-01-14:
i think the problem, is this

let deltaY = - (trackedItem.y - trackedItem.h / 2) - countingAreaProps.a * trackedItem.x - countingAreaProps.b;

      //let deltaY = - (trackedItem.y - trackedItem.h / 2) - countingAreaProps.a * trackedItem.x - countingAreaProps.b;

original   is
          let deltaY = - trackedItem.y - countingAreaProps.a * trackedItem.x - countingAreaProps.b;
I am reducing the same amount as the tracker, but if you do this, counts incorrect

can you try to see if yours counts properly lowering the center

tdurand, 2020-01-14:
If you say that this counts once the box passes completely, maybe you need to reverse and do this:

```
let deltaY = - (trackedItem.y + trackedItem.h / 2) - countingAreaProps.a * trackedItem.x - countingAreaProps.b;
```

rantgithub, 2020-01-15:
Hi
 
The last change fixed the issue with the box.

Thank you.

And adjusting the center, help to address the issue of big boxes.

Doing the adjustment for the paths and working with the drawing line based on : 
           x: lastFrameTrackedItem.x,
            y: lastFrameTrackedItem.y + lastFrameTrackedItem.h / 2

makes the path to have  some gaps, and the lines is not continuous like when the calculation is done with the center.

Initially I thought that wont make any difference, but please see attachment picture. Sometimes the gap is much bigger/wider.  I have played with different numbers but the gaps remains.

any suggestions on this ? and not sure if this is also affecting counting because could be a scenario when a gap is at the counting line and can be ignored. ?

Thank you

[image]

tdurand, 2020-01-15:
did you changed it for pointB also ?

```javascript
pointA: {
            x: lastFrameTrackedItem.x,
            y: lastFrameTrackedItem.y + lastFrameTrackedItem.h / 2
          },
          pointB: {
            x: trackedItemScaled.x,
            y: trackedItemScaled.y + trackedItemScaled.h / 2          
}
```

rantgithub, 2020-01-16:
Hii
Actually I changed B, but I didn't keep the Itemscaled. I saw the typo. This fixed the issue.
Thank you

rantgithub, 2020-01-16:
Doing all this tests, I realized something related to my question a couple weeks ago about the difference/delay  between the boxes and the video and you mentioned is a known issue.

When I have a live feed, the image can only be processed as fast as you are getting it. Meaning for FPS(a) are coming from the feed.
When the feed is recorded as a video, usually is processed faste FPS(b)r, depending of what Hardware you use. Meaning the FPS(b) are greater than FPS(a)

If FPS(a) and FPS(b) are the same, there is no delay between the identification boxes and the video.

Based on this, make me thing and on a video case, if the video is slow down to the original FPS(a) the delay can be eliminated.

Based on this, when a video is loaded for analysis, is any parameter than can be adjusted to set a specific FPS and therefore eliminate the boxes/video difference as a workaround.

Thank you

tdurand, 2020-01-16:
HI,  

No there are not, only thing you could do is to slow down the video file beforehand... It is a technical limitation that can't really workaround as it is.

rantgithub, 2020-01-27:
Hi

Hope all is well :)

Looking for options of how to handle the limitation about the "processing frame speed" and real display, I am trying to calculate/display a delta Frame, between two zone counter lines.

To expand more,

I have C1 (counter line 1) and C2 (counter line 2), that they are on the same x,y plane of each other.

Meaning c1 is for instance, x1, y1 and C2 is X1, y2, with y2 greater y1.

Under the live view (liveview.js) I want to calculate  on what frame, the object.id=AA, crosses  C1 and on what frame crosses C2, in this way I can have a delta of frames.

something like
deltaframe = Frame.C2 - Frame.C1.

I was  looking the code and I saw under the opendatacam.js  there is the

Opendatacam.currentFrame as well the counting trigger

and under liveview.js 

context.fillText(
deltaframe (what is use to display  the object type and confidence.)

I am trying to make the call from the liveview to get this delta but not sure about how to get it from the opendatacam procedure, (getting errors)

With this, what I like to do is to display after object.id=AA crosses C2, what deltaFrame value was calculated.

Can you please provide your input on what I need to call in order to display this on screen.

Any comment on this is greatly appreciated.

Thank you

[22 more replies]

closed 2020-03-23 by tdurand