opendatacam/opendatacam #168: Opendatacam crashs after a few hours while recording

gabkol, 2019-12-18:
After some hours of recording with yolo3 network opendatacam freezes and crashs. The data is lost then. I extraced the logs from docker container.

Important log parts: 

> Something went wrong: read ECONNRESET
Too much retries, YOLO took more than 3 min to start, likely an error
6000
Process YOLO exit
...
Process YOLO stopped
==== HTTP Stream closed by darknet, reset UI ====
==== If you are running on a file, it is restarting  because you reached the end ====
==== If you are running on a camera, it might have crashed for some reason and we are trying to restart ====
Something went wrong: connect ECONNREFUSED 127.0.0.1:8070
Too much retries, YOLO took more than 3 min to start, likely an error
0
Process YOLO started
Something went wrong: socket hang up
Too much retries, YOLO took more than 3 min to start, likely an error
0
Process YOLO exit
...
Process YOLO started
Something went wrong: socket hang up
Too much retries, YOLO took more than 3 min to start, likely an error
0
Process YOLO exit
...
Process YOLO started
Something went wrong: read ECONNRESET
Too much retries, YOLO took more than 3 min to start, likely an error
0
Process YOLO exit
...
Process YOLO started
Something went wrong: socket hang up
Too much retries, YOLO took more than 3 min to start, likely an error
0
Process YOLO exit

tdurand, 2019-12-18:
Hi, 

Yes, this is a know issue: https://github.com/opendatacam/opendatacam/issues/95 

Please follow the work on : https://github.com/opendatacam/opendatacam/issues/112 that should improve a lot reliability.

I think if it crashes and you restart it you still should be able to have the recording data until it crashes.

closed 2020-01-07 by tdurand

tdurand, 2020-03-30:
Please try this: https://github.com/opendatacam/opendatacam/issues/184