opendatacam/opendatacam #170: Export images detected objects

trickititit, 2019-12-19:
How i can download or write on drive when detecting objects images?
When detecting an object, it’s easy for you to save a frame.

tdurand, 2019-12-19:
Hi ! 

You can't "by design" , some explanation for our README, we do this cause of privacy issue

> OpenDataCam never records any photo or video data. The system only saves surveyed meta-data, in particular the path an object moved or number of counted objects at a certain point. The novelty of OpenDataCam is, that everything happens on location, while no visual data is saved or sent to online cloud processing.

That said, you can fork the tool and do it yourself, you need find a way to persist the MJPEG stream coming out the neural network.

closed 2019-12-19 by tdurand

vsaw, 2020-10-17:
@tdurand Would you be open to a Pull Requests implementing this as a configurable feature?

tdurand, 2020-10-21:
Hi !! Sorry I’m on the move will process everything in 1-2 weeks .. thanks for all the activity..

Pinging @b-g on this ... I’m not very open to it as the philosophy of the tool is not to record any images.. maybe as a debug tool

vsaw, 2020-10-21:
To give you a little bit of background on why I would need it in my use case.

1. *Debugging*: We work with a custom model to detect custom objects. As the model is still under development we don't know how well the detection works in real-life scenarios yet. Pictures could help us determine and improve the accuracy.
1. *Proof of Work*: Our use case is counting of non-personal objects. Therefore no privacy concerns have been raised. However there's a high demand for documentation and "proof of work" as our use case is directly linked to quality control.

I know that in a public space privacy concerns matter, but I hope that we can find a way to set up Opendatacam in a way to ensure privacy by default, yet enabling a more detailed collection where required.

b-g, 2020-10-21:
Hi there, yes an incredible strength of the ODC always was that we only saved aggregated anonymized data and NEVER anything down to an individual person. Hence I guess we are not very keen to push any features in this direction. I get and I can see why you potentially want this feature ... but I don't think we should add them to the official ODC core.

Hence I would suggest that you create such a "detect and shot" tool with the API as an "external" piece of software. Sorry! I hope you understand.

vsaw, 2020-10-21:
@b-g Thanks for the clear position on this.