opendatacam/opendatacam #173: not working, help...

PnM72, 2020-01-09:
We have the feeling this is not a New Issue, but didn't really know where to post this so sorry for cross-posting...
Here in Lisbon we have been trying first with a Nano (which run for approx. 5 mins until it freezes) and now with a TX2 which we are only able to run for less than an hour until it freezes and therefore crashes as well.
We have tried different types of installations and in case we were missing something, we re-installed it many times (with and without docker, with different cameras...) but without success.
Through the System Monitor we are able to see that when it starts recording, the memory keeps increasing until it runs out of memory, and therefore freezes quickly. 
Reading other posts makes us wonder if this is a common situation (like described on the memory leak issues)? 
Is there any way to make it work? 
Any help is very appreciated! We would love to have it running in the city.

tdurand, 2020-01-09:
Hi,

Yes, we indeed have this long term run issue.. #95 

5 min sounds terrible though, it should at least run for 1-2h.. Not sure how to help as we weren't able to fix that memory leak #108 #131 

I guess your best move would be to wait on #112

closed 2020-01-09 by tdurand

tdurand, 2020-03-30:
Please try this version which should fix the memory leak: https://github.com/opendatacam/opendatacam/issues/184