opendatacam/opendatacam #174: Integrating to python app

SaddamBInSyed, 2020-01-23:
Hi there.

thanks for the awesome work.

I would like to ask How can I integrate this object detection code into my app?

If integrating is a BIG task for a beginner like me. I would like to get an event/signal to my application whenever any truck is identified.

highly appreciated your help. 

platform:
jetson nano with the jetpack 4.3
using SD card image
IP camera dahua brand..

tdurand, 2020-01-24:
Hi, 

You can listen to this endpoint of the API for events: https://opendatacam.github.io/opendatacam/apidoc/#api-Tracker-Data

SaddamBInSyed, 2020-01-26:
Hi @tdurand ,

Thanks for your reply.

Highly appreciated, if you share any sample code for getting the opendatacam processed frame result from my python app.

Thanks

tdurand, 2020-01-28:
You should search some tutorials / documentation on how to listen to an HTTP API from python, I'm sorry but I don't work with python so I can't help you more specifically.

closed 2020-01-28 by tdurand