opendatacam/opendatacam #175: Error trying to run opendatacam on nvidia jetson tx2

MalcolmTomisin, 2020-01-27:
After following the quick start up steps in https://github.com/opendatacam/opendatacam#-get-started-quick-setup, when I run "sudo ./run-opendatacam.sh" on my jetson tx2 I get a  _docker: Error response from daemon: error while creating mount source path '/usr/lib/aarch64-linux-gnu/libGL.so': mkdir /usr/lib/aarch64-linux-gnu/libGL.so: file exists._

tdurand, 2020-01-27:
Maybe try to clear up your docker cache and reinstall opendatacam ?  

```
sudo docker system prune -a
```

Did it work when you first installed it ?

MalcolmTomisin, 2020-01-27:
Yes it did work when I installed and ran the container 5 months ago, so I will do as you say and reinstall opendatacam.

MalcolmTomisin, 2020-01-28:
Went ahead to clear up my docker cache and reinstall opendatacam and still got the same error!

tdurand, 2020-01-28:
Hum.. not sure how to solve this, one way it might work would be to reinstall Jetpack ... 

Maybe you can try to comment out this line: https://github.com/opendatacam/opendatacam/blob/097a89b9818009bc558e6237fb819daddcf99a60/docker/run-jetson/run-docker.sh#L21 of run-docker.sh

Also we are gonna work on the next version of opendatacam soon, and the docker integration should be much better

MalcolmTomisin, 2020-01-28:
> Hum.. not sure how to solve this, one way it might work would be to reinstall Jetpack ...
>
> Maybe you can try to comment out this line: [...]

Commenting out the line and running the ./run-opendatacam.sh causes this error - _docker: Error response from daemon: OCI runtime create failed: container_linux.go:345: starting container process caused "process_linux.go:430: container init caused \"rootfs_linux.go:58: mounting \\\"/usr/lib/aarch64-linux-gnu/libgstbadvideo-1.0.so.0.1401.0\\\" to rootfs \\\"/var/lib/docker/overlay2/5b117d307ca087ecfd15a86b29bdfc4ebaae9e943c717d4839cfd4f970ae24dd/merged\\\" at \\\"/var/lib/docker/overlay2/5b117d307ca087ecfd15a86b29bdfc4ebaae9e943c717d4839cfd4f970ae24dd/merged/usr/lib/aarch64-linux-gnu/libgstbadvideo-1.0.so.0.1401.0\\\" caused \\\"not a directory\\\"\"": unknown: Are you trying to mount a directory onto a file (or vice-versa)? Check if the specified host path exists and is the expected type._

tdurand, 2020-01-28:
Well, I think for now your only option would be to re-install a clean jetpack install... Or maybe try uninstalling docker completely... Not sure.

As I said in the next version of opendatacam coming in a few month the docker integration will be "natively" supported by Jetpack and should be much more reliable than this hack of mounting things into the container

closed 2020-01-28 by tdurand