opendatacam/opendatacam #176: traffic (group-)speed recognition with opendatacam and fm sending to fm-navi in cars

qayqay, 2020-02-24:
i wrote a white paper about a traffic signaling system where a camera on top of a red light measures the traffic speed and congestion in order to estimate the probability of passing on green, and sending the "green phase" distance directed towards the traffic line including gps position and analyzing direction, pic attached. 
[image]

this concept will work with citeeecar.com (in progress), an electric city car concept that will reduce particulate matter by knowing the redlight s phase and traffic movement.

so i am looking for someone being familiar with the opendatacam project who can detect traffic speed (ie according to buildings length?) and create a system that can send out directed wireless signals (<1km) on a suited frequency (standard ultra short wave fm, so no hardware needs to be modified in many navis that already comprise of gps/fm device), so that a detecting device (navi-fm) receives the distance signal (including red light s gps position and analyzing direction), verifies the situational validity, defines its "green status" and indicates to the driver if he will pass on the actual phase or not. later stages could do more with that info, ie elon could already control his cars in order to reduce particulate matter in towns and smoothen traffic flow.

it is my understanding that opendatacam can already do more than that, so additional info such as traffic counting (per hour/day/month/year) may also be offered to the communities in order to get funding.

please tell me if you are interested in such a project, if you can deliver such system (or what would be needed to be accomplished by other parties). what high quality all-weather cam do you suggest, what fm device and how to direct standard fm frequences (does that make sens?)? we could start by an ldr glued to the green light just for demonstration purposes.

shams3049, 2020-02-25:
Hi, 

I am detecting speed opendatacam, however before I go into questions specific to your concept prototyping, My first question is what is your envisioned timescope for this project?

regards

qayqay, 2020-03-01:
hi shams3049, time scope; whatever it takes;)
i would be happy to see the project realized, and hopefully, we could contribute to particle matter reduction where it is needed most.

tdurand, 2020-03-23:
I'll close this one as unrelated to direct work on OpenDataCam, but feel free to continue the discussion between you

closed 2020-03-23 by tdurand