opendatacam/opendatacam #178: Docker setup improvements

tdurand, 2020-03-05:
Since jetpack 4.3 , nvidia provides an [official base image for jetson's](https://ngc.nvidia.com/catalog/containers/nvidia:l4t-base): . 

Our current setup to "dockerize" opendatacam on jetson is pretty hacky because we didn't have access to CUDA stuff inside docker and we needed to:

- Compile darknet and opencv outside docker and copy the builds inside the image at build time 
- When running the image, we need to [manually mount all the CUDA dependency](https://github.com/opendatacam/opendatacam/blob/master/docker/run-jetson/run-docker.sh#L17) from the base OS and rely on the fact that CUDA was properly installed on the jetsons we were launching OpenDataCam from 

This new docker base image provided from nvidia should enable us to 

- have almost the same Dockerfile for Ubuntu CUDA machine / Jetsons by using cuda to compile darknet / compile opencv directly when building the image
- have access the CSI cam like raspberry pi cam from docker , for platform like jetson nano it was a limitation

I'm working on updating our Docker process and I successfully managed to have opencv compiled inside the docker build, but I'm still struggling with some issues to compile darknet when building the image.

I've posted a repro on the nvidia forums: https://devtalk.nvidia.com/default/topic/1072782/jetson-nano/-compiling-darknet-in-l4t-base-r32-3-1-docker-image-get-error-cicc-not-found/ 

I'll pick up this this later, and for now stick to our current pipeline to provide some testable build for OpenDataCam v3, as from user point of view, this doesn't change much. 

For reference, here is the current work in progress Dockerfile for jetsons

```Dockerfile
FROM nvcr.io/nvidia/l4t-base:r32.3.1

RUN  apt-get update -y && apt-get install -y \
            libgstreamer1.0-0 \
            gstreamer1.0-plugins-base \
            gstreamer1.0-plugins-good \
            gstreamer1.0-plugins-bad \
            gstreamer1.0-plugins-ugly \
            gstreamer1.0-libav \
            gstreamer1.0-doc \
            gstreamer1.0-tools \
            libgstreamer1.0-dev \
            libgstreamer-plugins-base1.0-dev
RUN  apt-get update -y && apt-get install -y  pkg-config \
 zlib1g-dev  libwebp-dev \
 libtbb2 libtbb-dev  \
 libgtk2.0-dev pkg-config libavcodec-dev libavformat-dev libswscale-dev libv4l-dev \
 cmake
RUN apt-get install -y \
  autoconf \
  autotools-dev \
  build-essential \
  gcc \
  git
RUN apt-get update -y && apt-get install -y ffmpeg
ENV OPENCV_RELEASE_TAG 4.1.1
RUN git clone --depth 1 -b ${OPENCV_RELEASE_TAG}  https://github.com/opencv/opencv.git /var/local/git/opencv
RUN cd /var/local/git/opencv
RUN mkdir -p /var/local/git/opencv/build && \
    cd /var/local/git/opencv/build $$ && \
    cmake -D CMAKE_INSTALL_PREFIX=/usr/local CMAKE_BUILD_TYPE=Release -D WITH_GSTREAMER=ON -D WITH_GSTREAMER_0_10=OFF -D WITH_CUDA=OFF -D WITH_TBB=ON -D WITH_LIBV4L=ON WITH_FFMPEG=ON -DOPENCV_GENERATE_PKGCONFIG=ON ..
RUN  cd /var/local/git/opencv/build && \
      make install

RUN git clone --depth 1 -b uselib https://github.com/tdurand/darknet /var/local/darknet

WORKDIR /var/local/darknet
RUN sed -i -e s/GPU=0/GPU=1/ Makefile;
#RUN sed -i -e s/CUDNN=0/CUDNN=1/ Makefile;
RUN sed -i -e s/OPENCV=0/OPENCV=1/ Makefile;
RUN sed -i -e s/LIBSO=0/LIBSO=1/ Makefile;
# Uncomment line corresponding to ARCH , 42 nano, 45 tx2, 33 xavier
RUN sed -i '42 s/^#//g' Makefile;

# ERROR here,  darknet doesn’t compile
RUN make 
```

cc @b-g

shams3049, 2020-03-05:
hey, 

I remember a similar error the one you posted on Nvidia website, can you check if your CUDA path are set right for Jetson Nano. 

$ export PATH=/usr/local/cuda-9.0/bin${PATH:+:${PATH}}
$ export LD_LIBRARY_PATH=/usr/local/cuda-9.0/lib${LD_LIBRARY_PATH:+:${LD_LIBRARY_PATH}}

tdurand, 2020-03-05:
hey @shams3049 , many thanks for the pointer, I believe I tried but I will double check.

tdurand, 2020-03-06:
Some more pointers (from nvidia forum answers)

- Try with this base image which also have cuDNN, TensorRT and DeepstreamSDK pre-installed:
https://ngc.nvidia.com/catalog/containers/nvidia:deepstream-l4t

- nvidia people answered that they are trying to reproduce our issue : https://devtalk.nvidia.com/default/topic/1072782/jetson-nano/-compiling-darknet-in-l4t-base-r32-3-1-docker-image-get-error-cicc-not-found/

tdurand, 2020-03-18:
Ok, got an answer from nvidia, there seems to be an issue with there docker base image: https://forums.developer.nvidia.com/t/compiling-darknet-in-l4t-base-r32-3-1-docker-image-get-error-cicc-not-found/113154/4 , they did successfully reproduced the bug. 

So I guess we just need to wait they publish an update on their base docker image . @b-g

tdurand, 2020-03-31:
They updated the issue with this, seems they are taking it pretty seriously so we can hope a fix soon !

```
Hi,

Thanks for your patience.

This issue is still under checking.
The latest status is that we can find other binary (ex. nvcc) in the l4t-base:r32.3.1 image.
However, we cannot find the corresponding cicc binary.

Will update more information with you once we got any progress.
Thanks.
```

tdurand, 2020-04-15:
I got a new anwer on this: https://forums.developer.nvidia.com/t/compiling-darknet-in-l4t-base-r32-3-1-docker-image-get-error-cicc-not-found/113154/10?u=tibo.durand

> /usr/local/cuda is readonly
> 
> One of the limitations of the beta is that we are mounting the cuda directory from the host. 
> This was done with size in mind as a development CUDA container weighs 3GB, on Nano it’s not always possible to afford such a huge cost. We are currently working towards creating smaller CUDA containers.

So it is still not possible to build darknet inside the docker image, we will need to do as we previously did, compile it outside and copy the compiled version inside the docker image.

That said, if we have only this to workaround this still simplify greatly our docker setup... and hopefully bring us the possibility to access CSI camera (raspberry pi cam) from the docker runnable container. 

I guess this also unblocks the work to be done on this.

tdurand, 2020-04-16:
Ok, I got the confirmation this will be something that will be solved later on but no ETA, so let's  do the new docker setup for OpenDataCamv3 with still a little workaround around darknet

> Sorry that we cannot disclosure a concrete schedule here.
> But we are trying to extract the essential part of CUDA toolkit for the l4t docker image.
> Will let you know once it is public available.

tdurand, 2020-04-24:
A new Jetpack version came out: https://www.jetsonhacks.com/2020/04/23/jetpack-4-4-dp-release/ , and it seems they also upgraded their docker base image. I've asked if there has been any improvements.. 

Also I was planning to work on updating our Docker setup next week... but i'm not sure if we should go with jetpack 4.3 or 4.4 Developer preview.. will see

tdurand, 2020-04-29:
Updates from the marvelous world 🌈️🌈️🌈️🌈️🌈️🌈️ of Jetson + Jetpack + Docker + Nvidia  😋️😋️ 

I tried to update our Docker image & process to use the latest base image `nvcr.io/nvidia/l4t-base:r32.4.2` 

It works (we still need to compile darknet outside and copy it to the Dockerfile at build time), and simplify greatly all the hacks we were doing at runtime to mount CUDA etc etc...

But I still can't access to the raspberry pi cam on jetson nano, I've posted an issue on the nvidia forums https://forums.developer.nvidia.com/t/access-to-raspberry-cam-v2-fails-from-docker-container-using-l4t-32-4-2-using-nvarguscamerasrc/121512

Here is the work in progress Dockerfile

```Dockerfile
FROM nvcr.io/nvidia/l4t-base:r32.4.2

RUN  apt-get update -y && apt-get install -y \
            libgstreamer1.0-0 \
            gstreamer1.0-plugins-base \
            gstreamer1.0-plugins-good \
            gstreamer1.0-plugins-bad \
            gstreamer1.0-plugins-ugly \
            gstreamer1.0-libav \
            gstreamer1.0-doc \
            gstreamer1.0-tools \
            libgstreamer1.0-dev \
            libgstreamer-plugins-base1.0-dev
RUN  apt-get update -y && apt-get install -y  pkg-config \
 zlib1g-dev  libwebp-dev \
 libtbb2 libtbb-dev  \
 libgtk2.0-dev pkg-config libavcodec-dev libavformat-dev libswscale-dev libv4l-dev \
 cmake
RUN apt-get install -y \
  autoconf \
  autotools-dev \
  build-essential \
  gcc \
  git
RUN apt-get update -y && apt-get install -y ffmpeg
ENV OPENCV_RELEASE_TAG 4.1.1
RUN git clone --depth 1 -b ${OPENCV_RELEASE_TAG}  https://github.com/opencv/opencv.git /var/local/git/opencv
RUN cd /var/local/git/opencv
RUN mkdir -p /var/local/git/opencv/build && \
    cd /var/local/git/opencv/build $$ && \
    cmake -D CMAKE_INSTALL_PREFIX=/usr/local CMAKE_BUILD_TYPE=Release -D WITH_GSTREAMER=ON -D WITH_GSTREAMER_0_10=OFF -D WITH_CUDA=OFF -D WITH_TBB=ON -D WITH_LIBV4L=ON WITH_FFMPEG=ON -DOPENCV_GENERATE_PKGCONFIG=ON ..
RUN  cd /var/local/git/opencv/build && \
      make install

# Install pre-compiled darknet on docker
COPY darknet/ /var/local/darknet/

# Install node.js
RUN apt-get install -y libcurl4 curl
RUN curl -sL https://deb.nodesource.com/setup_10.x | bash -
RUN apt-get install -y nodejs

# Technique to rebuild the docker file from here : https://stackoverflow.com/a/49831094/1228937
# Build using date > marker && docker build .
# date > marker && sudo docker build -t opendatacam .
#COPY marker /dev/null

RUN git clone --depth 1 -b development https://github.com/opendatacam/opendatacam /var/local/opendatacam

WORKDIR /var/local/opendatacam

RUN npm install
RUN npm run build

# Install Mongodb
# NB: for some reason this needs to be at the end otherwise mongod command isn't installed
# https://github.com/dockerfile/mongodb#run-mongod-w-persistentshared-directory
ENV DEBIAN_FRONTEND noninteractive
RUN apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv 9DA31620334BD75D9DCB49F368818C72E52529D4 && \
    echo "deb [ arch=amd64,arm64 ] https://repo.mongodb.org/apt/ubuntu xenial/mongodb-org/4.0 multiverse" | tee /etc/apt/sources.list.d/mongodb-org-4.0.list
RUN apt-get update && apt-get install -y openssl libcurl3 mongodb-org
VOLUME ["/data/db"]

EXPOSE 8080 8070 8090 27017

# # Because we want to run mongodb and the node.js app
# # See https://docs.docker.com/config/containers/multi-service_container/
COPY docker-start-mongo-and-opendatacam.sh docker-start-mongo-and-opendatacam.sh
RUN chmod 777 docker-start-mongo-and-opendatacam.sh
CMD ./docker-start-mongo-and-opendatacam.sh
```

tdurand, 2020-04-30:
Ok got an [answer from the nvidia forums](https://forums.developer.nvidia.com/t/access-to-raspberry-cam-v2-fails-from-docker-container-using-l4t-32-4-2-using-nvarguscamerasrc/121512/5?u=tibo.durand) (always very responsive, not worth it to be stuck and google for 3 hours) . There was an undocumented trick to make the CSI cam available from the docker container, you need to mount some volume at run time

```
-v /tmp/argus_socket:/tmp/argus_socket
```

For reference to run the container interactively on the jetson and mount a local config.json  here is the command:

```
sudo docker run --runtime nvidia -p 8080:8080 -p 8090:8090 -p 8070:8070 -v /data/db:/data/db -v /tmp/argus_socket:/tmp/argus_socket -v /home/nvidia/Documents/run-docker/config.json:/var/local/opendatacam/config.json --rm -it opendatacam
```

And then it works !! 

But it will be too easy if that was it... Now I'm investigating yet another issue that is I get only  6 FPS on the jetson nano for tiny yolo v3 from the docker container whereas I have usual 16 FPS from outside... It might be a problem because of the Jetpack versions (4.3, 4.4) and CUDA / CUDNN versions... not sure... need to trial and error again ;-)

tdurand, 2020-04-30:
Ok more on this, I keep my monologue 💁‍♂️️💁‍♂️️

Seems the guilt is to put on the new Jetpack version 4.4 , which ships with cuDNN 8.0 whereas Jetpack 4.3 is shipping with 7.x

I'm not sure if the problems comes from Darknet that doesn't support this new cuDNN or from nvidia that may have a big with cuDNN 8.0. 

For now I've opened an issue on darknet: https://github.com/AlexeyAB/darknet/issues/5426

tdurand, 2020-04-30:
Also found out this repo that can come handy to improve a bit the docker build workflow when releasing new images.. it has some .sh scripts that might be interesting: https://github.com/dusty-nv/jetson-containers

tdurand, 2020-04-30:
Regarding the speed issue with the new Jetpack 4.4 , it is reproducible inside / outside docker, I posted it on the nvidia forums: https://forums.developer.nvidia.com/t/darknet-slower-using-jetpack-4-4-cudnn-8-0-0-cuda-10-2-than-jetpack-4-3-cudnn-7-6-3-cuda-10-0/121579 , must be an issue with cuDNN 8.0.. let's wait a patch

I will build the images for v3.0.0-beta.2 with Jetpack 4.3 for now.

tdurand, 2020-05-05:
Ok, this is more or less done... updated:

1. [Jetson Dockerfile](https://github.com/opendatacam/opendatacam/blob/development/docker/run-jetson/Dockerfile)

Now uses `nvcr.io/nvidia/l4t-base:r32.3.1` as a base (not the latest as [there are performance issues](https://forums.developer.nvidia.com/t/darknet-slower-using-jetpack-4-4-cudnn-8-0-0-cuda-10-2-than-jetpack-4-3-cudnn-7-6-3-cuda-10-0/121579/5) ) , 

This enabled:

- building OpenCV dynamicly when creating the Docker image and avoid having to pre-build OpenCV outside, I deprecated : https://github.com/opendatacam/opencv-builds
- Support for CSI / Raspberry pi cam for a Docker install of OpenDataCam on jetson nano
- Simplifies a lot the complexity to run the container, no need to mount 10 volumes to gives a hacky access to CUDA stuff

2. [Desktop CUDA Dockerfile](https://github.com/opendatacam/opendatacam/blob/development/docker/run-nvidia-docker/Dockerfile)

Uses now `nvidia/cuda:10.2-cudnn7-devel-ubuntu18.04` as a base

This enabled building darknet with the CUDNN flag and gives a 6 FPS boost for the docker build.. (I never had the time to work this out, it was also possible to do in v2.x)

I'm currently uploading the docker builds for v3.0.0-beta.2 . For final release the only remaining thing to do may be to upgrade the base for the Jetson dockerfile if nvidia fixes the bug: https://forums.developer.nvidia.com/t/darknet-slower-using-jetpack-4-4-cudnn-8-0-0-cuda-10-2-than-jetpack-4-3-cudnn-7-6-3-cuda-10-0/121579/5

tdurand, 2020-05-05:
For future improvements , here is a very nice project: https://github.com/balena-io-playground/opendatacam that does two things to improve the docker setup:

- Use a `docker-compose.yml` to have the mongoDB setup externaly from the main Dockerfile
- Extend the base image of nvidia with OpenCV pre-installed, which results on a much simpler Dockerfile: https://github.com/balena-io-playground/opendatacam/blob/master/Dockerfile

fabianmoronzirfas, 2020-05-05:
> For future improvements , here is a very nice project: [balena-io-playground/opendatacam](https://github.com/balena-io-playground/opendatacam) that does two things to improve the docker setup:
> 
>     * Use a `docker-compose.yml` to have the mongoDB setup externaly from the main Dockerfile
> 

We are also using docker-compose with opendatacam. You can see our WIP here https://github.com/technologiestiftung/otc-jetson-nano-docker (the docker-compose.yml is generated with this [tool](https://github.com/technologiestiftung/otc-toolkit/tree/master/docker-compose-builder))

We are using:

- external mongodb (started as replicaset so we can have change streams)
- logspout to collect logs
- mongoku/mongo-express to inspect the data

tdurand, 2020-05-06:
@fabianmoronzirfas Many thanks for sharing this ! I like the idea of having "out of the box" some log collector and a UI based way to explore the data in the mongodb

Great, I think I will try to do another iteration on the docker flow before v3.0.0 release , but you can already simplify a lot this part ;-) : https://github.com/technologiestiftung/otc-jetson-nano-docker/blob/master/docker-compose.test.yml#L101  .. Now the volumes to mount to run the container is much simpler: https://github.com/opendatacam/opendatacam/blob/master/docker/run-jetson/run-docker.sh

fabianmoronzirfas, 2020-05-06:
> Great, I think I will try to do another iteration on the docker flow before v3.0.0 release , but you can already simplify a lot this part ;-) : [technologiestiftung/otc-jetson-nano-docker:docker-compose.test.yml@`master`#L101](https://github.com/technologiestiftung/otc-jetson-nano-docker/blob/master/docker-compose.test.yml#L101) [](/technologiestiftung/otc-jetson-nano-docker/blob/HEAD@{2020-05-06T05:26:10Z}/docker-compose.test.yml#L101)

Cool. That will make the docker-compose.yml way easier to read

JaledMC, 2020-05-23:
Hi @tdurand 

I'm not a opendatacam user, but searching about docker found your useful info. Have you check this dockerfile? It seems very clean to me, and compiles darknet inside it (no Cuda version selection, although).

Happy coding :)

tdurand, 2020-05-23:
Hi ! Thanks, which dockerfile are you speaking about ?

JaledMC, 2020-05-23:
My fault! [Here](https://github.com/Azure/IntelligentEdgeHOL/blob/master/docker/jetson-nano-l4t-cuda-cudnn-opencv-darknet/Dockerfile) is the docker. And [this folder] have multiple bases that could be useful.

tdurand, 2020-05-23:
thanks for stoping by and sharing, will have a look next week , seems interesting indeed !

alanb128, 2020-06-01:
Hi @tdurand 

There's a new balena repo for OpenDataCam 3.0.0b2 [here]( https://github.com/balena-io-playground/opendatacam/tree/3.0.0-beta2). The base image has been streamlined a bit and is now using OpenCV 4.1.1. In addition, this version supports an unmodified config.json file which can be overridden by adding balenaCloud variables via the dashboard. (If a variable is deleted in the dashboard, that property reverts to the config.json value)
Unfortunately I'm not able to get the Raspberry Pi cam working yet using nvarguscamerasrc because I can't seem to get that plugin installed, even when following various Gstreamer instructions.

shams3049, 2020-06-01:
@alanb128 Thats some good work, Is it possible for you leave a couple of guidelines lines to be able to make a similar setup for TX2.

Regarding GStreamer with CSI-2 camera, creating a stable nvarguscamerasrc pipeline proved a challenge for me so far too, I am not using Raspberry Pi cam but a similar interface camera from e-con systems.  The workaround that seams to be working at the moment, is creating a pipeline that is streaming to an ip port and then consuming it in the application as as ip stream.

tdurand, 2020-06-01:
@alanb128 This sounds very nice, I love how simple and clean is the final dockerfile:  https://github.com/balena-io-playground/opendatacam/blob/3.0.0-beta2/Dockerfile .. looks you also optimized the size of the Docker container...  I'm gonna try this today and play around and let you know.. 

The things I need to investigate are:

- is it a significant effort to have a similar image for Jetson Xavier / TX2 , joining @shams3049 on this
- see if we can provide an easy way to mount some custom weights 
- is this mounting the opendatacam_videos folder ?

I would very much like to just put this setup as the default setup for docker on jetson.

I though that Raspberry Pi could not run YOLO at more than 1-2 FPS which is too low for OpenDataCam to work properly.. so this is not very much an issue in my opinion if it cannot run there..

tdurand, 2020-06-01:
@alanb128 

Just starting to play with it, I'm wondering how do you build / run image (with dockercompose) ?

Because right now I did this:

```bash
#cloning your repo branch ODCv3.0.0beta2 and open go to the directory

sudo docker build -t balena-opendatacam .
```

Then running with

```
sudo docker run --runtime nvidia balena-opendatacam
```

But this doesn't start the mongodb and so on.. I must admit I never worked with docker-compose so I'm gonna look in the documentation but I'm wondering how you tell dockercompose to use the GPUs ?

tdurand, 2020-06-01:
@alanb128 

Update, did that to install docker-compose via pip3

```
sudo apt-get install -y libffi-dev
sudo apt-get install -y python-openssl
sudo apt-get install libssl-dev

pip3 install docker-compose
```

And was able to build the container using

```
sudo `which docker-compose` build
```

And then run the container using

```
sudo `which docker-compose` up
```

And seems to work properly.. Seems it figures out automaticly that it needs to give access to GPU / CUDA stuff.. Also I need to work out a better way to install docker-compose because right now the command doesn't work with sudo out of the box, I need to do this weird `which docker-compose` trick.. 

Too bad docker-compose isn't available out of the box on Jetpack.. 

I'm gonna test the implications of building this for Xavier / TX2 and how this would change our install process... but to me it seems that we would go this way for v3.0.0 docker setup.. Also this would fix out issues right now with Jetpack 4.4 DP ..as with your solution everything is inside the container 

As a note to myself, we need to change the [path where darknet is installed](https://github.com/balena-io-playground/opendatacam/blob/master/Dockerfile#L10) in the container so it is consistent with our Docker image for Ubuntu desktop. Change `/darknet` to `/var/local/darknet`

good job !

alanb128, 2020-06-01:
@tdurand 

If using balenaCloud, there is no need for docker-compose. One can simply click the "deploy with balena" button on the repo, the application gets pushed to the cloud builders, and then the containers get pushed to the device. Alternatively, a user can pull the repo locally, make changes if desired, then use the balena CLI to push the code to the cloud builders and/or directly to the device. (I'd be glad to do a quick demo, can get in touch with you via e-mail to set that up if you'd like)
For a non-balenaCloud install, I run docker-compose as a container itself which is a much easier install. I haven't tried that on the Nano though, so will test and post that if it works.

tdurand, 2020-06-01:
> If using balenaCloud, there is no need for docker-compose. One can simply click the "deploy with balena" button on the repo, the application gets pushed to the cloud builders, and then the [...]

Ok ! I love the click a button to deploy ;-) . Yes I think for this we will just add a link in the README to your blog post / documentation and maybe this button directly.

> docker-compose as a container itself which is a much easier install

This is the same as what I did no ? Use the docker-compose CLI ? Because if we make this Docker setup the default setup for OpenDataCam we need a way to install without Balena with only docker.

alanb128, 2020-06-01:
I don't think the way you installed compose is quite the same. See this method: https://www.digitalocean.com/community/tutorials/how-to-install-docker-compose-on-ubuntu-18-04 (although I have not tested on a Nano yet)

tdurand, 2020-06-01:
@alanb128 Yes I installed it with Pip, but I will try this way ;-)

tdurand, 2020-06-01:
FYI docker-compose has no build for ARM devices , so the easiest way to install it on the nano is via Python PIP3: https://blog.hypriot.com/post/nvidia-jetson-nano-install-docker-compose/

I'm wondering what is you workflow when working on those Dockerfile is you do not have docker-compose installed ?

alanb128, 2020-06-02:
When I work on Dockerfiles/docker-compose files usually a device is running balenaOS, so I use "[local mode](https://www.balena.io/docs/learn/develop/local-mode/)"  which uses the Docker daemon on the device to build container images.

tdurand, 2020-06-03:
@alanb128 Seems from my early test that your `jetson-nano-cuda-cudnn-opencv:v0.2-slim` is working also properly on Jetson Xavier .. I'm gonna test and try more but looks like that it would work "universaly" across all jetsons

alanb128, 2020-06-03:
That's great news @tdurand ! We could further automate using multiple Dockerfiles by device type to push the appropriate Makefile, weights, and config.json to the device. In that case we could have the following files:
```
Dockerfile.jetson-nano
Dockerfile.jetson-tx2
Dockerfile.jetson-xavier
```
When deploying an app, the builders will automatically use the appropriate Dockerfile.

tdurand, 2020-06-04:
> That's great news @tdurand ! We could further automate using multiple Dockerfiles by device type to push the appropriate Makefile, weights, and config.json to the device. In that case we could have [...]

Yes, just need to change the Makefile ARCH= for darknet + which weight to default pull + default config.json .. 

**"Thinking" in progress:**

_Didn't test yet, just writing some notes, will try later_

This might be irrelevant for you as you use the Balena CLI, but the thing that I'm looking into is to mount the config.json / opendatacam_video folder  at runtime with `docker-compose up` .. This is what I'm doing now with the simple Dockerfile (see [run-docker.sh](https://github.com/opendatacam/opendatacam/blob/master/docker/run-jetson/run-docker.sh)):

```
sudo docker run --runtime nvidia -p 8080:8080 -p 8090:8090 -p 8070:8070 --privileged -v $(pwd)/config.json:/var/local/opendatacam/config.json -v $(pwd)/opendatacam_videos:/var/local/darknet/opendatacam_videos:ro 
```

Seems with `docker-compose up` the only way to do this is by using ENV variable ( https://stackoverflow.com/a/58900415/1228937 )

The means I will need to add changes the docker-compose.yml you bootstrapped:

```yml
version: '2'
volumes:
    home-data:
services:
  opendatacam:
    restart: always
    build: .
    privileged: true
    volumes:
      - 'home-data:/data/db'
      #  Mount argus socket for CSI cam access
      - '/tmp/argus_socket:/tmp/argus_socket' 
      #  Mount config.json
      - '${PATH_TO_INSTALL_FOLDER}/config.json:/var/local/opendatacam/config.json'
      # Mount an external video directory for people to run on files
      - '${PATH_TO_INSTALL_FOLDER}/opendatacam_videos:/var/local/darknet/opendatacam_videos:ro'
      # Mount an external video directory for custom weights
      - '${PATH_TO_EXTERNAL_WEIGHTS}:/var/local/darknet/external_weights:ro'
    ports:
      - "8080:8080"
      - "8070:8070"
      - "8090:8090"
  mongo:
    image: mongo
    restart: always
    ports:
      - "27017:27017"
    volumes:
      - 'home-data:/data/db'
```

And then run with this:

```
PATH_TO_INSTALL_FOLDER=$(pwd) PATH_TO_EXTERNAL_WEIGHTS='pathtomyweights' docker-compose up
```

One question I would have for you is what is `home-data` ? Also please let me know if you think of a better way ... and if you think that adding those ENV variable in the docker-compose would make the thing not compatible with Balena install-run ? (as the goal would be to have the same docker-compose file)

fabianmoronzirfas, 2020-06-04:
@tdurand  I think you don't need to full path when you want to mount volumes with docker-compose. I always use relative paths. The [docs](https://docs.docker.com/compose/compose-file/#short-syntax-3) say:

> You can mount a relative path on the host, which expands relative to the directory of the Compose configuration file being used. Relative paths should always begin with `.` or `..`.

Or maybe I don't get what you are aiming for.

tdurand, 2020-06-04:
@fabianmoronzirfas thanks for the ping! very helpful indeed, this will simplify things ! I think you got what I'm looking to do. So this means we can get rid of the env variables ! Just tell people to put there config / video files / custom weights in the folder they have the docker-compose file.

I didn't had the time to play with it yet, I was just reading the docker-compose documentation and thinking out loud in the issue, that's why it might look a bit messy.. But you comment might save me a few hours of tweaking so thanks !

fabianmoronzirfas, 2020-06-04:
I guess an installer like the current one could also write docker-compose.yml on the fly on the CWD directory and add the full path into the volumes section.

tdurand, 2020-06-04:
Yep, will could be a way to do it, but if relative path work I guess it is simpler to just download the docker-compose.yml and use it "out of the box" . ( I might be missing something )

fabianmoronzirfas, 2020-06-05:
One thing I would also suggest is using tags for the images. Like `mongo:4.2.6` and `opendatacam:???`

tdurand, 2020-06-05:
Thanks, will do! I'm aiming to release a new beta with this setup in the coming days

alanb128, 2020-06-05:
@tdurand the 'home-data` is a [volume](https://docs.docker.com/storage/volumes/) as opposed to the others you added which appear to be [bind mounts](https://docs.docker.com/storage/bind-mounts/). Volumes are the preferred method for persisting data in Docker containers (currently bind mounts are [not supported](https://www.balena.io/docs/learn/develop/multicontainer/#named-volumes) in balena.) So basically you define your volumes (persistent folders) at the top of the docker-compose and then you can map them to any folder in any of the containers. (That's what we are doing with `home-data`) Docker keeps track of where the volume is actually located on the host; we don't need to worry about it. 
You could store the config.json and/or the videos in the `home-data` volume or create new volumes for each or both of them.

tdurand, 2020-06-05:
@alanb128 Thanks ! 

I just read quickly the documentation , seems using named volumed require more work on the host machine with docker like `docker volume create` etc.. also from my understanding we still would need to create several volume to be able to "surgically" overwrite config.json / opendataca_videos / weights files  ... So I think it might be better to use bind volume in our case if we want the simpler setup as possible... As we don't need the power of named volume.

I guess in the end you will have a slightly different docker-compose.yml file for the Balena install of OpenDataCam .. but the idea would be to have them as similar as possible

Please let me know if I overlooked something. As I told @fabianmoronzirfas I'm working on releasing a new beta with this docker install setup "reworked" in the coming days

alanb128, 2020-06-05:
If you're using docker-compose, I don't think you need to use `docker volume create` or any additional work on the host to create a named volume. However I see your point about easily overwriting specific files with persistent ones stored on the host through bind volumes. I'll think about that a bit and see if there is a way the compose files can work for both!

fabianmoronzirfas, 2020-06-05:
I had a little test to see how one could use a path in the working directory with named volumes in docker-compose. You can see the setup here

https://gist.github.com/fabianmoronzirfas/2f0449e4c9a5cc563c81b65591f4b00a

Prerequisites:  

- The folder `./data` in the current working directory must exist
- `$PWD` in the docker-compose.yml does not exist Windows

So a user might need to set this path manually or a install script could do that.
But it works to have all changes from the host machine available in the running containers

`Dockerfile` This is just to keep a process running, so we can run `docker exec -it one /bin/bash` and `docker exec -it two /bin/bash` :

```Dockerfile
FROM ubuntu
ENTRYPOINT [ "tail" ]
CMD ["-f","/dev/null" ]
```

`docker-compose.yml` I aggregated this volumnes setup from searching the web. There might be errors in it.

```yml
version: '3.4'

services:
  one:
    container_name: one
    build:
      context: .
      dockerfile: Dockerfile
    volumes:
      - type: volume
        source: mydata
        target: /mydata
#	  - mydata:/mydata # should also work here	
  two:
    container_name: two
    build:
      context: .
      dockerfile: Dockerfile
    volumes:
      - type: volume
        source: mydata
        target: /mydata
#	  - mydata:/mydata # should also work here
volumes:
  mydata:
    driver: local
    driver_opts:
      type: none
      o: bind
      device: "${PWD}/data"
```

`.dockerignore` Ignore the data folder so we don't end up with a local copy:

```gitignore
data/
**/.dockerignore
**/.env
**/.git
**/.gitignore
**/docker-compose*
**/Dockerfile*
README.md
```

fabianmoronzirfas, 2020-06-05:
> @fabianmoronzirfas Thanks... from my understanding I see no upside going this way.. Using bind volume like below does the same and enables us to "surgically" overwrite some Docker container data.

Yes. Nothing gained this way. I just thought to get a Balena compatible docker-compose.yml this way could serve both setups.

tdurand, 2020-06-05:
yes thanks,  it don't think it is a big deal if we have a slighly difference `docker-compose.yml` file for different type of deployment as long as it is easy to maintain. For a balena deployment if I'm not mistaken there is no way to overwrite files like config.json or mount video / weight files @alanb128 ? we can just set Env variables ?

alanb128, 2020-06-06:
Yes @tdurand we can set ENV variables or device variables in the cloud dashboard to modify the config.json, etc... I'm also still looking into an approach that would allow a "surgical" file overwrite as well, will update here on that progress.

fabianmoronzirfas, 2020-06-07:
> we can just set Env variables ?

While reviewing config.json I asked myself with some of the variables in there: “hey why isn’t that a env variable?”
I’ll compile a list of the ones I think could go into env tomorrow and post it here. Maybe you agree on some.

tdurand, 2020-06-08:
@fabianmoronzirfas you can do like @alanb128 is doing, just overwrite the env variable in the config.json : https://github.com/balena-io-playground/opendatacam/blob/master/launch.sh

alanb128, 2020-06-08:
Just FYI, In the latest version, I'm using [jq](https://stedolan.github.io/jq/) which simplifies the overwriting of the config.json a lot:
https://github.com/balena-io-playground/opendatacam/blob/3.0.0-beta2/launch.sh

fabianmoronzirfas, 2020-06-09:
This feels pretty hackish. :) Since you are doing the switch to docker-compose.yml you could also leverage the [docker-compose/environment](https://docs.docker.com/compose/environment-variables/) key and use some `process.env` calls in Node.js. If you like I can make a PR for it?

tdurand, 2020-06-09:
@fabianmoronzirfas yes go ahead it is useful for you.. @alanb128 ways of doing it seems good enough for me as I think I will only document how to overwrite the whole file.. What would be nice is to move every "GET" to a config var on the node.js part to this utility: https://github.com/opendatacam/opendatacam/blob/master/server/utils/configHelper.js  and here you can add for some of them an overwrite by the ENV variables if defined.

tdurand, 2020-06-09:
FYI, I'm working on releasing the next beta with the initial support for docker-compose later today.. so you might want to wait for this before working on adding ENV variables

tdurand, 2020-06-24:
I tried something with some inputs from @alanb128 , which is to copy those two file inside the docker container:

```
COPY libgstnvvidconv.so /usr/lib/aarch64-linux-gnu/gstreamer-1.0/
COPY libgstnvarguscamerasrc.so /usr/lib/aarch64-linux-gnu/gstreamer-1.0/
```

And mount the argus_socket as a volume

```
volumes:
      - './config.json:/var/local/opendatacam/config.json'
      - '/tmp/argus_socket:/tmp/argus_socket'
```

I get a different error but this still does not work

```
(Argus) Error NotSupported: EXT_platform_device extension missing (in src/eglutils/EGLUtils.cpp, function getDefaultDisplay(), line 44)
opendatacam_1  | (Argus) Error BadParameter:  (propagating from src/eglstream/FrameConsumerImpl.cpp, function initialize(), line 89)
opendatacam_1  | (Argus) Error BadParameter:  (propagating from src/eglstream/FrameConsumerImpl.cpp, function create(), line 44)
opendatacam_1  | Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, threadInitialize:247 Failed to create FrameConsumer
opendatacam_1  | Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, threadFunction:188 (propagating)
opendatacam_1  | Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:588 (propagating)
opendatacam_1  | 
opendatacam_1  |  seen 64, trained: 32012 K-images (500 Kilo-batches_64) 
opendatacam_1  | video file: nvarguscamerasrc sensor_id=0 ! video/x-raw(memory:NVMM),width=1280, height=720, framerate=30/1, format=NV12 ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=360 ! videoconvert ! video/x-raw, format=BGR ! appsink
opendatacam_1  | nvbuf_utils: Could not get EGL display connection
opendatacam_1  | nvbuf_utils: ERROR getting proc addr of eglCreateImageKHR
opendatacam_1  | nvbuf_utils: ERROR getting proc addr of eglDestroyImageKHR
opendatacam_1  | GST_ARGUS: Creating output stream
opendatacam_1  | [ WARN:0] global /usr/src/app/opencv-4.1.1/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
opendatacam_1  | Something went wrong: connect ECONNREFUSED 172.19.0.2:8070
opendatacam_1  | Too much retries, YOLO took more than 3 min to start, likely an error

```

shams3049, 2020-07-07:
@alanb128 thanks for your good work on balena, I tried out balena with opendatacam more then an year ago, and got stuck with getting my camera installed properly. I have multiple different jetons all using [e-con system cameras](https://www.e-consystems.com/nvidia-jetson-camera.asp). Do you have any pointers on how I can approach adding custom driver image to balena, and how to debug it.

tdurand, 2021-02-02:
Todo need to test with the new Jetpack 4.5 if improvements on this side, from the release note:  _"V4L2 API 2 is extended to support interacting with CSI Camera. Implementation is compliant with V4L2 spec."._

From what i understand this might means that now the CSI cam is available on the same interface as the USB cams: /dev/video* ?

vsaw, 2021-02-03:
This would be great news! I'm really struggling to work around the CSI limitations and the bugs in `nvargus-daemon` in the current recommended JetPak version. If this is confirmed we should consider changing the recommended JetPak version for ODC too!

tdurand, 2021-02-03:
yes, need to double check this.. and yes also got a bit delay, but definitely will be working this month on testing everything on latest jetpack and updating everything.. would be great to have support for CSI cam from docker..

In the case we still need to use `nvargus` I saw that the people from balena updated their base image: github.com/balenalabs-incubator/opendatacam so maybe they fixed it, and also I can't really remind why I moved from the l4t-base official image from nvidia.. I think it was to have OpenCV / CUDA  included in docker and do not depend on the jetpack version on device..

[3 more replies]