opendatacam/opendatacam #182: Feedback v3.0.0-alpha.1

b-g, 2020-03-16:
Is is way better now! It was running fine for days ... see screenshots. Also I can confirm bug #177, definitely a problem on start up and a blocker to use OpenDataCam in the wild.

[image]
[image]
[image]
[image]
[image]

tdurand, 2020-03-18:
Great news !!

rantgithub, 2020-03-24:
if you need help with testing V3.0 alpha let me know and i will assign some time to that

tdurand, 2020-03-24:
@rantgithub thanks, will ping you later this week once I release the next version

tdurand, 2020-03-29:
New version released: https://github.com/opendatacam/opendatacam/issues/184

closed 2020-03-29 by tdurand