opendatacam/opendatacam #183: [2.1.0 and 3.0.0 Alpha 1] Issue with recording to null

Denizzje, 2020-03-20:
Hello there,

First a thank you for this great project. I wish all documentation was written like this, to be honest. I am running a Jetson Nano and I use the non docker install due to using a custom Yolov3-tiny connected over RTSP to an IP Camera over the internet. Getting about 17fps with that on a 640x480 stream.

All functionalities seem to work, counter and tracker and such. However, recordings do not seem to work for me. I get the following error:

The programme does say it is recording after a delay of about five minutes. It seems to be writing to nothing however. Indeed it turned out that the counter had 4 objects passed but the non existant record still had 0.

I did have some issuses with installing MongoDB and got it from somewhere else. It might be that... the repo link at the non docker install link for MongoDB is broken. However it does seem to run, and I get no warning from the programme that the DB is not working. For the rest I have no idea what might cause it.

Some other things:

- There is always a "person" being "detected". It does not show up in the Yolo logs, however it is on the screen for some odd reason along with a variable confidence. There is no 'person' class in my model.
- .. the stream does not start untill anything is being detected. Fortunally this non existant person comes to the rescue after a few minutes. Is it possible to change that or is that a sign of a borked install?

Thank you very much in advance!

tdurand, 2020-03-23:
Hi, 

Thanks for your message.

- The error seems to be unrelated to mongodb. I double checked the line and yes you might have spotted a bug.. If nothing is tracked and the record starts, it might cause this.. try to click start recording when something is detected on the screen as a temp workaround ?

- Yes this is a known issue: https://github.com/opendatacam/opendatacam/issues/177

- For the ghost person.. weird, no idea what could be going on

To try to isolate the source of the pb, could you try a docker install of v3.0.0-alpha.1 https://github.com/opendatacam/opendatacam/issues/131#issuecomment-595651705 and use the default weights to see if you have the same problems ??

tdurand, 2020-03-23:
FYI I was able to reproduce something similar to you... I'm working on some fixes and will publish a new version later this week for you to test

tdurand, 2020-03-23:
I was able to fix thoses bugs I think. I'm not gonna release an update until end of the week as it is a bit of manual work to do.. but if you installed opendatacam without docker you should be able to test if you like by pulling the development branch https://github.com/opendatacam/opendatacam/tree/development  of this repo, and do

```
npm install
npm run build;npm run start;
```

Denizzje, 2020-03-24:
Hello there,

Thank you for the prompt reply. I did a quick test with pulling the dev branch. I noticed:

- Ghost person is gone. No more non existent person being detected under water ;).
- With the ghost person being gone I cannot get into the UI anymore. Unfortunally it may take hours before I detect something. I leave it running nontheless. Hope a fix can be made for that. Thankfully the API comes to the rescue!

Recording:

- Started via Postman a recording this time with a POST command. It returns an "internal server error". The console again gives out the error as described above but with minor differences. And when doing a GET status command the programme seems to be doing its best to record to null. 

Console output: https://i.imgur.com/NmRvOGA.jpg  (sorry for the photo).

Next time I have time to retry I will try the docker as asked. Just wanted to test this quickly and return the feedback.

tdurand, 2020-03-24:
Thanks !

- Weird, now normaly I added something that goes straight to the video even if nothing is detected...
- Recording.. ok, seems you still have some issues.

I aim to release an alpha.2 version in 1-2 days so I guess your best move is to wait for this.. maybe the dev branch is a bit corrupted with all the changes I'm making

shams3049, 2020-03-24:
Does it matter what node.js version is used? Even if not, whats the version @tdurand  is using?!

PS: didnt know where to post this.

tdurand, 2020-03-24:
@shams3049 no, it "shouldn't matter", if you use a very old node.js version it might only not work.. are you experimenting issues related to node.js version ?

tdurand, 2020-03-24:
I'm on v10.x right now, we had set this on the documentation... but more recent version should also work

shams3049, 2020-03-24:
@tdurand was just checking to be use, I am trying a parallel set up, which isnt running, the only diffrence in the enviroment is diffrent node.Js version. However I am not sure if it really in due to node version.

shams3049, 2020-03-24:
@tdurand all sorted out, there is no problem with node version compatibility.

Denizzje, 2020-03-24:
Seems the mysterious ghost person returned during the time I kept it on. I came back to a frozen image with the null record still recording to nowhere for 326 minutes. Probably my Nano ran out of memory I do not have a swap file I think.

I shall await the new alpha in the meantime!

Denizzje, 2020-03-25:
Well this is interesting. I left the programme running still, and came to my office founding the video stream to be responsive again. It inmediately opened after I put in the IP adress, without any "initalising opendata". It was running for 1258 minutes at the moment of writing.

Instead of spitting out "invalid JSON output" in the console log, it does the "nothing detected for frame no XXXXXX" which I think is intended to be shown?

I was able to stop the 'null' recording in the webinterface and start a new one. It now has a name assigned to it! Now to wait for some actual detections ;).

The phantom objects still go on, I now have underwater bicycles instead of persons. It seems it is constantly tracking those ghost items too. 

I found this error to be generated unfortunally dont know the circumstances: https://i.imgur.com/nNSwGxd.png

Here is the tracker json of the mysterious underwater bicycle: 
[tracker.zip](https://github.com/opendatacam/opendatacam/files/4379883/tracker.zip)

tdurand, 2020-03-29:
Hi, 

Could you please test with v3.0.0-beta.1 : https://github.com/opendatacam/opendatacam/issues/184

Make sure to remove all previous opendatacam and darknet folders to be certain that you have everything on the latest versions.

Feel free to re-open a new issue if you still experiment problems, I'll close this one for now.

Thanks  !

closed 2020-03-29 by tdurand

bilx6043, 2020-04-21:
> Hello there,
>
> Thank you for the prompt reply. I did a quick test with pulling the dev branch. I noticed:
>
> * Ghost person is gone. No more non existent person being detected under water ;).
> * With [...]

hi did u manage to solve this issue of recording id null.. i am facing the same issue in 3.0.0 beta 1

tdurand, 2020-04-21:
@bilx6043 yes, it has been fixed , just pull the "development" branch