opendatacam/opendatacam #184: Release v3.0.0-beta.1, testing needed

tdurand, 2020-03-29:
### Happy to release v3.0.0-beta.1 🎉️🎉️ , featuring:

- Memory leak fixed on jetson's, expected running time should be counted in days now instead of hours ( https://github.com/opendatacam/opendatacam/issues/182 )
- Drag & drop support for file, just drop a file on the UI and it will restart to process it
- Directional counters, see: https://github.com/opendatacam/opendatacam/issues/122#issuecomment-603733234 
- Save / Reload counters config: see https://github.com/opendatacam/opendatacam/issues/122#issuecomment-604952995
- Several smaller bugs fixes, clean up logs etc etc...

I did the build for Jetson nano, ubuntu machines, if you have a TX2 or Xavier, sorry but you should be able to install it [without docker](https://github.com/opendatacam/opendatacam/blob/v3.0.0-beta.1/documentation/USE_WITHOUT_DOCKER.md).  . _🚨️ Make sure to reinstall darknet from the opendatacamv3 branch as specified in the doc_

For the docker installation, still the same procedure _(Make sure to remove your remove your previous config.json, install-opendatacam.sh, run-docker.sh, run-opendatacam.sh before installing new version)_

```
# Download install script
wget -N https://raw.githubusercontent.com/opendatacam/opendatacam/v3.0.0-beta.1/docker/install-opendatacam.sh

# Give exec permission
chmod 777 install-opendatacam.sh

# NB: You will be asked for sudo password when installing the docker container

# Install command for Jetson Nano
./install-opendatacam.sh --platform nano

# Install command for a Nvidia-docker machine (ARCH_BIN=6.1)
# NB: Will run from demo file, you can change this after install, see "5. Customize OpenDataCam"
./install-opendatacam.sh --platform nvidiadocker_cuda_archbin_6_1
```

### Todo list before final 3.0.0 release:

- Fix bugs from your feedback
- Improve docker when nvidia will have fixed their bugs:  https://github.com/opendatacam/opendatacam/issues/178 , should enable us to support raspberry pi cam / CSI camera on jetson nano out of the box
- Update and clean documentation
- Clean up code for drag& drop feature.. maybe add some UI tweaks like a drop indicator and a specific progress message
- Update API docs with new counters attributes
- Improve installation process, maybe ask people if they want to run from camera or run from file at install time.. or maybe add something to change this in the UI
- Document how to run from custom weights

Thanks for your help to test this,  cc @b-g @shams3049  & all users

rantgithub, 2020-03-31:
Hi
sorry for the very basic question, but where is the doc for the installation instructions for this version ?
are we testing only the docker container or non docker install as well.?

rantgithub, 2020-03-31:
disregard, above message, i just found the v3 vesion install

testing now

rantgithub, 2020-03-31:
Hi

I was able to install the  no docker version. No issues.

My findings so far : 

I found an issue with the counting and direction.

the problem is that even when the lines are "alone" not touching any car or object or no object is crossing them, it increases the counter incorrectly.

I have reproduced the problem in other videos

Please using the demo file, put the zones below and you will see that the purple counter is counting incorrectly.

Pic and zones below.

[image]

{"1875aaa5-f133-4557-8b36-3eb2573a70c7":{"color":"yellow","type":"leftright_topbottom","location":{"point1":{"x":289,"y":685},"point2":{"x":545,"y":699},"refResolution":{"w":1853,"h":981}},"computed":{"a":-0.05810546875000017,"b":-245.5762614678899,"xBounds":{"xMin":99.81651376146789,"xMax":188.23529411764707},"lineBearings":[93.3254589676143,273.3254589676143]},"name":"l1"},"8cf2f0b2-fabe-4e53-9fc6-c5804122b92b":{"color":"turquoise","type":"leftright_topbottom","location":{"point1":{"x":546,"y":700},"point2":{"x":806,"y":689},"refResolution":{"w":1853,"h":981}},"computed":{"a":0.04495192307692283,"b":-265.3577981651375,"xBounds":{"xMin":188.58067997841337,"xMax":278.38100377765784},"lineBearings":[87.4261772136469,267.4261772136469]},"name":"l2"},"81b230f2-69ab-4282-862f-ccbd75cc7a7d":{"color":"green","type":"leftright_topbottom","location":{"point1":{"x":810,"y":690},"point2":{"x":1117,"y":689},"refResolution":{"w":1853,"h":981}},"computed":{"a":0.0034609120521172936,"b":-254.17924274571916,"xBounds":{"xMin":279.76254722072315,"xMax":385.7960064759849},"lineBearings":[89.8017051378647,269.8017051378647]},"name":"l3"},"50bf81b7-24fb-4286-80c9-7cbd3899bdfb":{"color":"purple","type":"leftright_topbottom","location":{"point1":{"x":994,"y":556},"point2":{"x":1004,"y":668},"refResolution":{"w":1853,"h":981}},"computed":{"a":-11.90000000000009,"b":3881.394495412875,"xBounds":{"xMin":343.31354560172696,"xMax":346.7674042093902},"lineBearings":[175.19651424093095,355.1965142409309]},"name":"l4"}}

thanks

rantgithub, 2020-03-31:
Some comment that I think was mentioned before

when displaying the FPS, instead of the real value, I think will be better to display the average of , let say last 100 frames, to make the number more stable and not so jumpy that many times is not usable.

bilx6043, 2020-03-31:
well i try to run docker and it run successfully but later i try to run without docker but it give an error... have anyone face anything like that.. i am new to this so dont know so much how to handle this error

output is 
Process YOLO initialized
> Ready on http://localhost:8080
> Ready on http://219.93.102.16:8080
Success init db
(node:16611) [DEP0001] DeprecationWarning: OutgoingMessage.flush is deprecated. Use flushHeaders instead.
Pheew ....! Something unexpected happened. This should be handled more gracefully. I am sorry. The culprit is:  { Error: spawn ./uselib ENOENT
    at Process.ChildProcess._handle.onexit (internal/child_process.js:240:19)
    at onErrorNT (internal/child_process.js:415:16)
    at process._tickCallback (internal/process/next_tick.js:63:19)
  errno: 'ENOENT',
  code: 'ENOENT',
  syscall: 'spawn ./uselib',
  path: './uselib',
  spawnargs:
   [ 'data/coco.names',
     'cfg/yolov3-tiny.cfg',
     'yolov3-tiny.weights',
     'opendatacam_videos/demo.mp4' ] }
Process YOLO started
First request on console stream
Something went wrong: connect ECONNREFUSED 127.0.0.1:8070
Too much retries, YOLO took more than 3 min to start, likely an error

tdurand, 2020-03-31:
@rantgithub many thanks for the report, I've created a separated issue #185 , being able to share the counter configuration proves very handy to reproduce bugs 

regarding FPS, we are computing on the average of the last 3 frames, but I could increase this to smoothen more.

@bilx6043 Seems darknet isn't working properly, can you make sure it is working on this own by testing it with the (Optional) Test darknet part of the documentation .. maybe you forgot to set LIBSO=1 when compiling it : https://github.com/opendatacam/opendatacam/blob/v3.0.0-beta.1/documentation/USE_WITHOUT_DOCKER.md#optional-test-darknet

rantgithub, 2020-03-31:
The error for @bilx6043 is reproducible is the new version of darknet is not use.

Be sure you update darknet to the version v3 requires, otherwise wont work.

rantgithub, 2020-04-01:
Hi

An additional issue, that is more than a display issue than an error

If you have a video and yolo detects a car, and the car moves from right to left (for this example), on the last frame detected the box of the car remains on the screen, until another is detected.

If no object is detected the box remains as an empty box on the screen. This applies to any object on the class list.

This is specially notorious on roads that have low traffic. Please see example below.

Car was detected and the next car shows up 5 min later. The box remains on the screen until next object appears 5min later.

I think once the box is detected on the last frame, the box should be removed, just to keep clean the screen.

[image]

tdurand, 2020-04-01:
@rantgithub , thanks for reporting, I though I fixed something similar already.. Could you please:

- Give me a link to download a short video with which I can reproduce ? 
- Try to open http://localhost:8070 (or IP of the machine opendatacam is running on) when this happens and see if Darknet is still sending this detection over the JSON stream ?

rantgithub, 2020-04-02:
Hi

1. See a sample. You can see here the behavior in several frames.

2. I think problem is coming from darknet. Please try the video and you will see what i mean.

[sample2.zip](https://github.com/opendatacam/opendatacam/files/4419172/sample2.zip)

tdurand, 2020-04-02:
Hi, 

Just tried with "yolov3.weights" , and I don't see a ghost, I just see a big detection "surfboard" (WTF) which is taking almost all the screen for no reason. Is it the same thing you see ? Or which weights are you using ? 

[image]

We experimented problems when building https://www.move-lab.com/project/beatthetraffic/ with those detections taking almost all the screen... Maybe something useful to add would be to have an option in the config.json to filter out the detections that are bigger than a certain thresshold (% of the frame), that is what we were doing for Beat the traffic: https://github.com/opendatacam/node-moving-things-tracker/blob/master/main.js#L170

rantgithub, 2020-04-02:
Hi

This is a very common problem and happens a lot, specially when low resolution cameras. I am not sure why yolo detects like this.

What I see is that was detected as surfboard. If you just define the class as "car", you wont see that  anymore and/or as suggested increasing the threshold on yolo detection helps to reduce this as well.

To duplicate the problem, just assign the class as "car" and you can see the box remains on the screen on those periods where there are no objects on the screen.

tdurand, 2020-04-02:
I restricted only to "car" in the config.json .. But I can't see any ghost like in your screenshot. 

Could you send either a demo screencast of the problem while running opendatacam or explain again how I can reproduce the bug ?

rantgithub, 2020-04-03:
Hi

I am attaching video here of my recording

If you see many boxes are "lost", because there is a gap of empty objects on the screen.

If you see in particular a car with id 25, remains for around 5 seconds no car is displayed, the box remains there until the next object is detected.

doesn't matter what class the object is, is only 1 object on the screen, 
[boxes.zip](https://github.com/opendatacam/opendatacam/files/4424557/boxes.zip)

this is the same video i provided and i can reproduce this issue on any video where traffic is very low or sporadic

The flow is
object1 is detected
object1 is tracked until the end of video
object1 disappear from video, but box remains on the screen
if no object gets into the screen , the box remains there
one new object2 arrives, the current box for object1 is deleted and new box is crated for object2

hope you can reproduce this now. :)

rantgithub, 2020-04-03:
Hi

I think the error on not starting video on object detection still remains.

Same video as before, just longer, if i open it is fine, but if i start to count, this error appears:

==== HTTP Stream closed by darknet, reset UI ====
==== If you are running on a file, it is restarting  because you reached the end ====
==== If you are running on a camera, it might have crashed for some reason and we are trying to restart ====
Stop recording
MJPG_sender: new client 48
Got first message from JSONStream
Error with message send by YOLO, not valid JSON
JSON_sender: new client 51
JSON_sender: kill client 47
Close socket: out = -1, in = -1 
Start recording
Pheew ....! Something unexpected happened. This should be handled more gracefully. I am sorry. The culprit is:  TypeError: Cannot read property 'id' of undefined
    at Object.startRecording (/home/odc/opendatacam/server/Opendatacam.js:518:90)
    at Object.updateWithNewFrame (/home/odc/opendatacam/server/Opendatacam.js:213:14)
    at IncomingMessage.<anonymous> (/home/odc/opendatacam/server/Opendatacam.js:646:18)
    at IncomingMessage.emit (events.js:198:13)
    at addChunk (_stream_readable.js:288:12)
    at readableAddChunk (_stream_readable.js:269:11)
    at IncomingMessage.Readable.push (_stream_readable.js:224:10)
    at HTTPParser.parserOnBody (_http_common.js:124:22)
    at Socket.socketOnData (_http_client.js:451:20)
    at Socket.emit (events.js:198:13)
Something went wrong: Parse Error
Too much retries, YOLO took more than 3 min to start, likely an error
6000

This error appears in a continuous loop

if i restart a couple of times, finally one of those starts

I download the version of 2-3 days ago, just to check if this is fixed or not

rantgithub, 2020-04-03:
Hi

Looking for performance, I noticed that on version 2.x the darknet process was running and now is the userlib. 

Typical darknet process was using around 1.8-2.5 Gig on version 2.x and now on 3.x is around 1.1-1.5

Performing several loads, I noticed GPU usage was reduced but appears the load was transferred to CPU, because CPU remains at 100% for some of the cores on a steady basis. This doesn't happen on 2.x

Just wondering if you are aware of this or able to duplicate. I tested 20 videos and behavior was the same, lower GPU higher CPU.
[image]
[image]
[image]
[image]

tdurand, 2020-04-03:
@rantgithub 
 
1. Ghost boxes: Sorry but I can't reproduce, it seems also you have some difference in display the opendatacam original code (on boxes.mp4), maybe you introduced some custom code that still causes this bug ?? Here is a video I just recorded using yolov3.weights on your sample2.mp4 :  https://youtu.be/vTrIiW3o_KE , you can see that the boxes clears well even if no other boxes are in the frame. Please create a separated issue if you still run into problems and you can give me another test to do so we don't "polute" too much this issue and people can still report other bugs ;-)

2. I fixed this issue after v3.0.0-beta.1 release: https://github.com/opendatacam/opendatacam/commit/f2cb66c6e2782d35fb00b5dd14cd1a737f562b97 , please try to pull the latest changes on the "development" branch and run `npm run build`

3. I have an issue right now with my ubuntu CUDA machine since a little update I did yesterday and can't run anything now... so can't test but will do later...  It might be.. but truth that what mattered to us was memory usage and not CPU / GPU when benchmarking ./uselib , for jetson devices we had a big memory leak that isn't happening with ./uselib so this was the reason for the switch... If it is important to you can I recommend to ask on https://github.com/alexeyab/darknet , he is very responsive and I would be happy to pull the latest changes in a future version of opendatacam if he fixes it. In the future we would like to be able to switch the lower level of neural network framework from darknet to something more "tested" like deepstream or tensorflow / pytorch...

rantgithub, 2020-04-03:
Hi.

1. Let me redownload all and compile, because darknet is something i dont touch. It is clear that on your version this doesn't happen. I will test and will provide feedback.

2. Ok, let me try that with new version. What I found for what I mentioned is that I have to press the refresh button to load. 

3 The issue I am seeing with this CPU at 100% is that now will "kill" the machine with 2-3 instances running. When I was testing,  my machine got frozen for several seconds and I have to kill  the process and restarted.  Initially appears this fixes jetson but breaks Pcs for this high CPU usage. I will test more and provide more feedback.

As a note, in on all tests I have done, on pcs, i havent seen any memory leak, that for what you mentioned is related more to the jetson family. The only thing I saw, is what i reported for the mongodb, but that was fixed couple months ago.

will continue testing

Thank you

:)

rantgithub, 2020-04-04:
Hi

I re download and compile all and now:

1. No more phantom boxes on v3. I need to check my previous versions to see what is causing this. I have the same issue on v2. I will download it again.

2. The issue with the black screen is resolved.

3. I tested modifying the main.js file  parameter:
LARGEST_DETECTION_ALLOWED = 1920 * 25 / 100 for
      detectionsForThisFrame = detectionsForThisFrame.filter((detection) => !isDetectionTooLarge(detection, LARGEST_DETECTION_ALLOWED));

but i didn't see any difference. Big boxes are still displayed on the screen. Regardless of what i modify, like

LARGEST_DETECTION_ALLOWED = 1920 * 12 / 100

Can you please clarify how I should use this ? to prevent this big boxes to be displayed. 

4. High CPU is confirmed and I think is a problem. Specially with more than one instance running, this crashes the machine.
I tested on different PC and same result. 
If you can check on yours and validate to confirm is an issue not only with my environment.
I already sent the question for  https://github.com/alexeyab/darknet and waiting for feedback.

Thank you

rantgithub, 2020-04-04:
Hi

Quick question about ports. (previous question)

on V2, on the yolo.js component, I can change the port under
var endCommand = ['-ext_output','-dont_show','-json_port','9070', '-mjpeg_port', '9090'] 
as well on other files (server,package and opendatacam)

I replaced all ports on these files but nothing is showing up because is saying error for listening on xx70. Looks like something else is needed to change.

My question, if i want to modify the ports, how i can specify port number using the uselib or where else i need to change that.

Thank you

tdurand, 2020-04-04:
1. Ok, on v2 it won't be fixed, it is something I fixed for v3 only, it wasn't related to darknet but to the tracker, see fix: https://github.com/opendatacam/node-moving-things-tracker/commit/a02ebfb34231d708942af291a28cf02907b9458e

2. great! 

3. It was just an example of something we did for another project, but it isn't used right now. I need to implement something like this inside OpenDataCam and put an option in the config.json, I added this to the TODO list.

4. Ok.. I will see next week if I can reproduce, to be very transparent with you also we likely won't invest much time in this as the tradeoff to make Jetson devices more reliable but having desktop CUDA less performant is fine with us... But yes in the best of the world we would like both to work as optimal as it can... But I'm pretty confident that if you put a reproductible bug on the darknet repository the author will find a solution.

5. ./uselib API came with other tradeoff which are the fact that the API is not as complete as the original ./darknet one, see my comment on the darknet repo: https://github.com/AlexeyAB/darknet/issues/4453#issuecomment-593450826 ... I quickly tweaked the C/C++ code to make it work with opendatacam: https://github.com/opendatacam/darknet/pull/5/files , but I'm not very good with C/C++ and couldn't easily make the ports configurable from the command line.  If it matters to you I would be happy to accept a Pull request that enhance mine to gives us `./uselib <yolo_params> <videoinput> -port 8090 -mjpeg 8070`  .  

Also please note that the old darknet API isn't compatible with the new OpenDataCam v3 as the formatting of the detection on the JSONStream is different.... It is something we don't want to put effort into also as it is a pain in the *** to keep track of darknet changes as it is a project without versionning ... 

Many thanks again for all your tests !!

rantgithub, 2020-04-04:
1. I realize that after looking more the code.

3. Ok, I think will be very helpful to have the max_size detection parameter,  because as I mentioned for low resolution videos, these big boxes are a constant "problem". Specially when you have a counting zone, if the box appears on the line it increases the counter in a continuous basis.

4. I clarified to alexyeb about this. The main problem I see now, is I "cant" use v3, because for more than 2 instances, it will kill the machine. V2 using darknet is very stable in terms of CPU and GPU usage.

5. On my case, the ports are really important because I use primarily PCs because the power in comparison to a jetson device. I change the ports as I mentioned for V3 as a test, and now I understand why didnt work. Having ./uselib <yolo_params> <videoinput> -json_stream 8090 -mjpeg 8070 will resolve this limitation. Thank you.

6. I agree on keeping all for v3 as well.

Thank you as well for all responses and follow up, really appreciated :)

rantgithub, 2020-04-04:
Hi

Alexyeb replied back and made a suggestion to modfiy file 

"./uselib is made for low-latency detection: capture->preprocessing->detection->postprocessing->showing->send_json, that's why it uses much more CPU.

So:

for high bandwidth (many streams) - use ./darknet
for low latency (1 stream) - use ./uselib

"

I asked how we can tweak this

Try to change 3 to 30 in this line and recompile:

darknet/src/yolo_console_dll.cpp

Line 251 in a9bae4f

 while(!a_ptr.load()) std::this_thread::sleep_for(std::chrono::milliseconds(3));"

I changed to 30 and it looks like that did the trick

[image]

rantgithub, 2020-04-04:
I played with several configuration and the lowest number we can use is 15, but that will depend on input and machine.

If you can work on the ports request, I can test with more heavy loads,,  :))

This  is the same video with same parameters using darknet just for comparison

[image]

tdurand, 2020-04-06:
@rantgithub , let's continue the discussion here: https://github.com/opendatacam/opendatacam/issues/189 , I've asked you a couple of questions

tdurand, 2020-04-06:
Also please see https://github.com/opendatacam/opendatacam/issues/190 in order to make the darknet ports usage configurable , help would be appreciated

Denizzje, 2020-04-18:
Good day,

Just dropping by here, currently looking at 8,5 day of recording and it still goes strong. Will drop a couple of more feature requests in the "issue" I made earlier ;).

shams3049, 2020-04-21:
Hey, I hope everyone is doing good.

I came across [this](https://datafromsky.com/) solution, could be a great source of inspiration. really fluid design and feature sets.

thimabru1010, 2020-04-21:
> well i try to run docker and it run successfully but later i try to run without docker but it give an error... have anyone face anything like that.. i am new to this so dont know so much how to [...]

I got the same error. Did u solve?

tdurand, 2020-04-22:
@shams3049 Great to hear from you !! Yes I came across this solution a few weeks ago and benchmarked it a bit (downloaded their software and tried on a demo video).. It does lots of things that opendatacam does not do and indeed can be a great source of inspiration..  Thanks for sharing !

@thimabru1010 Seems darknet isn't working properly, can you make sure it is working by testing it with the (Optional) Test darknet part of the documentation .. maybe you forgot to set LIBSO=1 when compiling it : https://github.com/opendatacam/opendatacam/blob/v3.0.0-beta.1/documentation/USE_WITHOUT_DOCKER.md#optional-test-darknet

thimabru1010, 2020-04-22:
Now I solved this error. Maybe was the new version of darknet fork. But I still got an error, on the opendataca 2.0.0 and on the 3.0.0. It seems is  gstreamer library, but I don't know how to solve it.
OBS: I compiled opencv with these flags:
cmake -DCMAKE_BUILD_TYPE=Release \
-DCMAKE_INSTALL_PREFIX=/usr/local \
-D WITH_GSTREAMER=ON \
-D WITH_GSTREAMER_0_10=OFF \
-D WITH_CUDA=OFF \
-D WITH_TBB=ON \
-D WITH_LIBV4L=OFF \
-DWITH_V4L=ON \
-D WITH_FFMPEG=ON \
.. 
'''
(uselib:13682): GStreamer-CRITICAL **: 22:15:59.600: gst_element_make_from_uri: assertion 'gst_uri_is_valid (uri)' failed
GStreamer: Error opening bin: no element "opendatacam_videos"

(uselib:13682): GStreamer-CRITICAL **: 22:15:59.600: gst_element_get_state: assertion 'GST_IS_ELEMENT (element)' failed
'''

I changed from lib4l to v4l, cause I was having another error that I solved also.
The error was something like this:
'''
VIDEOIO ERROR: V4L: device opendatacam_videos/demo.mp4: Unable to query number of channels
'''

Now I am having just gstreamer error.

OBS2: I am using gstream1.0

tdurand, 2020-04-23:
@thimabru1010 Which device are you trying to run opendatacam on ? If you are running on a jetson you don't need to compile OpenCV yourself, it comes already pre-installed with Jetpack..

thimabru1010, 2020-04-23:
> @thimabru1010 Which device are you trying to run opendatacam on ? If you are running on a jetson you don't need to compile OpenCV yourself, it comes already pre-installed with Jetpack..

I am running on my PC. It has a GTX 1060.

tdurand, 2020-04-23:
Honestly not sure how to help you, if your goal is to run on a file , you won't need Gstreamer support for OpenCV.. I installed my OpenCV and gstreamer dependencies on my Ubuntu devices a while ago doing this: https://github.com/opendatacam/opendatacam/blob/master/documentation/USE_WITHOUT_DOCKER.md#optional-how-to-compile-opencv-with-gstreamer-support-on-desktop 

But maybe better you install a newer version of OpenCV and try to compile darknet again...

Denizzje, 2020-04-27:
It seems my Jetson Nano wanted a break from looking at fish this weekend, cant reach it anymore so I assume it crashed and rebooted (dont have ODC running on startup).

I posted 9 days ago, so that makes it anywhere between 15 and 17 days recording time. All those times I regulary watched the stream in the GUI for sometimes hours on end. That used to crash in like 10 minutes on v2 😄 .

Are you going to get v3 out of beta this week by any chance? If so then I shall wait with starting a new recording.

tdurand, 2020-04-27:
@Denizzje Great ! Many thanks for the feedback.. way more stable.. We delayed a bit the work on the project but I aim to get a release candidate v3.0.0-beta.2 by the end of the week.. which likely will contain almost everything we want /can include in v3.0.0

tdurand, 2020-04-28:
> ### Happy to release v3.0.0-beta.1  , featuring:
> * Memory leak fixed on jetson's, expected running time should be counted in days now instead of hours ( #182 )
> * Drag & drop support for file, just drop a file on the UI and it will restart to process it
> * Directional counters, see: [#122 (comment)](https://github.com/opendatacam/opendatacam/issues/122#issuecomment-603733234)
> * Save / Reload counters config: see [#122 (comment)](https://github.com/opendatacam/opendatacam/issues/122#issuecomment-604952995)
> * Several smaller bugs fixes, clean up logs etc etc...
> 
> I did the build for Jetson nano, ubuntu machines, if you have a TX2 or Xavier, sorry but you should be able to install it [without docker](https://github.com/opendatacam/opendatacam/blob/v3.0.0-beta.1/documentation/USE_WITHOUT_DOCKER.md). . _rotating_light Make sure to reinstall darknet from the opendatacamv3 branch as specified in the doc_
> 
> For the docker installation, still the same procedure _(Make sure to remove your remove your previous config.json, install-opendatacam.sh, run-docker.sh, run-opendatacam.sh before installing new version)_
> 
> ```
> # Download install script
> wget -N https://raw.githubusercontent.com/opendatacam/opendatacam/v3.0.0-beta.1/docker/install-opendatacam.sh
> 
> # Give exec permission
> chmod 777 install-opendatacam.sh
> 
> # NB: You will be asked for sudo password when installing the docker container
> 
> # Install command for Jetson Nano
> ./install-opendatacam.sh --platform nano
> 
> # Install command for a Nvidia-docker machine (ARCH_BIN=6.1)
> # NB: Will run from demo file, you can change this after install, see "5. Customize OpenDataCam"
> ./install-opendatacam.sh --platform nvidiadocker_cuda_archbin_6_1
> ```
> 
> ### Todo list before final 3.0.0 release:
> * [x]  Fix counter line edge case bug : #185
> * [ ]  Few more tweaks to darknet: #190 #189
> * [ ]  Improve docker when nvidia will have fixed their bugs:  #178 , should enable us to support raspberry pi cam / CSI camera on jetson nano out of the box
> * [ ]  Update and clean documentation
> * [ ]  Clean up code for drag& drop feature.. maybe add some UI tweaks like a drop indicator and a specific progress message
> * [ ]  Update API docs with new counters attributes
> * [ ]  Improve installation process, maybe ask people if they want to run from camera or run from file at install time.. or maybe add something to change this in the UI
> * [ ]  Document how to run from custom weights / how to change network size
> * [ ]  Document how to handle m3u8 streams with gstreamer
> * [ ]  Add Thresshold  to filter out large detections
> * [ ]  Option to disable tracker and only count things ?
> * [ ]  Add -json_port -mjpeg_port options to the ./uselib api:
> * [ ]  Frame id = 50 only for file, not for streams
> * [ ]  Add favicon
> * [x]  Add delete confirmation modal for recordings
> * [x]  Add version number in side menu
> * [x]  Memory leak fixed on jetson's, expected running time should be counted in days now instead of hours ( #182 )
> * [x]  Drag & drop support for file, just drop a file on the UI and it will restart to process it
> * [x]  Directional counters, see: [#122 (comment)](https://github.com/opendatacam/opendatacam/issues/122#issuecomment-603733234)
> * [x]  Save / Reload counters config: see [#122 (comment)](https://github.com/opendatacam/opendatacam/issues/122#issuecomment-604952995)
> * [x]  Several smaller bugs fixes, clean up logs etc etc...
> 
> Thanks for your help to test this, cc @b-g @shams3049 & all users

tdurand, 2020-05-06:
v3.0.0-beta.2 finally out: https://github.com/opendatacam/opendatacam/issues/205

closed 2020-05-06 by tdurand