opendatacam/opendatacam #185: Counting bug with v3.0.0-beta.1

tdurand, 2020-03-31:
Reported by @rantgithub  

Purple counter is counting stuff happening outside his line

[image]

Counter config to reproduce:

File: demo.mp4

```json
{"1875aaa5-f133-4557-8b36-3eb2573a70c7":{"color":"yellow","type":"leftright_topbottom","location":{"point1":{"x":289,"y":685},"point2":{"x":545,"y":699},"refResolution":{"w":1853,"h":981}},"computed":{"a":-0.05810546875000017,"b":-245.5762614678899,"xBounds":{"xMin":99.81651376146789,"xMax":188.23529411764707},"lineBearings":[93.3254589676143,273.3254589676143]},"name":"l1"},"8cf2f0b2-fabe-4e53-9fc6-c5804122b92b":{"color":"turquoise","type":"leftright_topbottom","location":{"point1":{"x":546,"y":700},"point2":{"x":806,"y":689},"refResolution":{"w":1853,"h":981}},"computed":{"a":0.04495192307692283,"b":-265.3577981651375,"xBounds":{"xMin":188.58067997841337,"xMax":278.38100377765784},"lineBearings":[87.4261772136469,267.4261772136469]},"name":"l2"},"81b230f2-69ab-4282-862f-ccbd75cc7a7d":{"color":"green","type":"leftright_topbottom","location":{"point1":{"x":810,"y":690},"point2":{"x":1117,"y":689},"refResolution":{"w":1853,"h":981}},"computed":{"a":0.0034609120521172936,"b":-254.17924274571916,"xBounds":{"xMin":279.76254722072315,"xMax":385.7960064759849},"lineBearings":[89.8017051378647,269.8017051378647]},"name":"l3"},"50bf81b7-24fb-4286-80c9-7cbd3899bdfb":{"color":"purple","type":"leftright_topbottom","location":{"point1":{"x":994,"y":556},"point2":{"x":1004,"y":668},"refResolution":{"w":1853,"h":981}},"computed":{"a":-11.90000000000009,"b":3881.394495412875,"xBounds":{"xMin":343.31354560172696,"xMax":346.7674042093902},"lineBearings":[175.19651424093095,355.1965142409309]},"name":"l4"}}
```

tdurand, 2020-04-27:
This commit should fix this, currently on "development" branch, but will make it into v3.0.0-beta.2 release this week

https://github.com/opendatacam/opendatacam/commit/fcf8274c40e3aa1a6174d6f9d8eb5ab03bc9758e 

If you can test, would be very welcome !

closed 2020-04-27 by tdurand

rantgithub, 2020-04-27:
hi, ok, i will test this

:)

thank you