opendatacam/opendatacam #188: [3.0.0 beta 1 non docker] Recording to null

Denizzje, 2020-04-01:
Hello there,

The new version seems a great improvement over the previous one. The issue with the video not starting untill a detection is made is solved and it starts very quickly.

My FPS increased from about 17 to ~25, don't know what you did but that is the fastest I have seen Yolov3 tiny run on my Jetson Nano! Also great that you added the directional counters, that is really useful for my case. 👍 

The ghost detections I reported was due to me in my infinite wisdom using the standard .data file instead of my own one, whoops!

Unfortunally the recording to null error strikes again. Same steps and tried as in the issue I made before (sorry I do not know how to reference on Github). Starting the recording via the GUI and API nets the same result. With the API I get a "error 200 internal server error" when trying to start via the API but the null recording does start, to record to nowhere.

I get the following error this time, unsure if it is any different: https://i.imgur.com/mZDsrLL.png

Once again I will leave the recording going on as that "fixed" it once last time without me doing anything, to see if it happens again. 

Keep up the good work!
Denizz

tdurand, 2020-04-01:
Hi @Denizzje , 

Thanks for the test !! I might have spotted the bug you are running into, I just pushed a fix on the development branch: https://github.com/opendatacam/opendatacam/commit/f2cb66c6e2782d35fb00b5dd14cd1a737f562b97

Could you try to pull the latest changes of development branch ?:

```bash
cd YOUROPENDATACAMDIRECTORY
git checkout development
git pull
# You can verify you have latest changes 
# by doing "git log" and seeing if you have the commit try to solve 188
npm run build
```

Regarding performance on the jetson nano, maybe it is the FPS computation which is wrong... or maybe the latest darknet improved perfs with your weights... Also I'm curious on which kind of dataset you trained you weights, are you able to share some info about this ? What are you looking to achieve, it is always interesting to know what users are doing with OpenDataCam.

Let me know

Denizzje, 2020-04-01:
Confirmed fixed! Record button now properly assigns a name and starts recording at once.

My use case:

I have a normal IP camera installed in a dry area of a fish passage. I want to know what times fish travel through it. There is basic motion detection on the camera but I feel it misses out on alot, and I just get pictures which I have to process manually to get something like that.

Experimenting with Yolo made me found out it is very good at detecting forms. With a meagre dataset it already accurately sees most of the smaller fish, and even in bad conditions. It gets confused by big fat fish but that is just a matter of training more.

I needed a lightweigth application which outputs JSON or CSV using the custom model which can be read/sent automaticaly to combine it with other measurements, and I arrived here.

I now use a simple 2 class dataset, with Fish and Plant being the classes. The latter is to determine when alot of junk goes through the passage and to keep the model from being confused by the junk.

tdurand, 2020-04-01:
Yeah !! Great, this fix will make it into the next release.

Very interesting use case ! some follow up questions to satisfy my curiosity 😁️ ( ok if don't want to answer)

- What is the end goal of detecting those fishes ? 
- You say meagre dataset, how many images is this ? (curious because I would like to train for an idea of mine also)
- regarding hardware , what IP cam is it ? is it battery powered? do you have the nano "on-site" or you have the ip cam far away from it ?

Denizzje, 2020-04-01:
The end goal is to have a time line and counting of detected fish, in order to combine that data with other parameters (water quality).

The meagre dataset is now 50 images of fish, and no validation neither. I think Yolo recommends atleast 2000 per class? I have alot of fish on the older non IP cameras but their viewing angle was different, I just use images from this camera. Hopefully that is soon solved as fish migration will start soon ;).

Its a Dahua IP camera, so a standard security CCTV camera, powered over the mains. Chosen for its ease of access remotely and the ability to focus at close distance too. The Nano will be placed with the other equipment so it will be done locally.. However for this proof of concept phase I the Nano here in the office watching the stream over the internet.

It is watching a potato quality 640x480 512 kbit/s stream of the secondary channel from the camera, while I use the main 1080p stream to send pictures of the basic motion detection by the camera itself. I use those for training the model.

tdurand, 2020-04-01:
Thanks !! very nice project !!  For v3.0.0 release we might want to feature some interesting project like yours as we had done for v2.0.0: https://github.com/opendatacam/opendatacam/issues/84, I let you know if it's the case and if you would like to participate cc @b-g 

Regarding the issues with v3.0.0, please keep the feedback coming ! Closing this one for now

closed 2020-04-01 by tdurand