opendatacam/opendatacam #189: CPU usage optimization for darknet ./uselib API

tdurand, 2020-04-06:
Discussion started here with @rantgithub : https://github.com/opendatacam/opendatacam/issues/184#issuecomment-608226738

There is this issue on the darknet repo: https://github.com/AlexeyAB/darknet/issues/5170

Seems we can change a line on the darknet implementation:

in darknet/src/yolo_console_dll.cpp

```c
while(!a_ptr.load()) std::this_thread::sleep_for(std::chrono::milliseconds(3));
```
to 

```c
while(!a_ptr.load()) std::this_thread::sleep_for(std::chrono::milliseconds(30));
```

@rantgithub , could you explain a bit more what would be the drawback of doing this ? I don't really understand what is it doing ? Would it slow down the process for less high end devices ? How is this related to FPS ?

rantgithub, 2020-04-06:
Hi

I can ask on that specific, but I tested several parameters and videos and is directly related to 1000/FPS

when number is lower than 15, CPU spike, because is processing much more.

In order to run it without killing CPU, you have to add basically a delay here. He recommended no more than 30, and the reason that gives you 30 FPS.

I tested several videos and all went well, but testing more with real time because i saw some delays.

Trying to test many scenarios to be sure all is consistent.

tdurand, 2020-04-07:
Ok understood ! Yes actually "throttling" darknet to 30 FPS even if your computer can process things faster is something we wanted to do in order to have a clean output in the UI, when things goes too fast the mjpeg stream can't catch up...

Great ! Will add this to our fork of darknet in the next beta.

rantgithub, 2020-04-08:
Hi

Testing more, I have find issues with delays when using realtime feed coming from m3u8 url (using HLS)

I have tested several live feeds and what is happening there are so many frames missing/skip for ODC that causes incorrect counts.

I compare the direct feed coming from the cameras and the outgoing stream is fine. What appears is happening is that with the overhead of the processing, some frames are lost.

This only happens with realtime feeds. No issue with files.

I am wondering if there is a buffer that can be use on OCV or similar to prevent this frames get lost.

The end result of this, is a incorrect count due to a car that is for counting just disappear from screen because frames are lost.

Any suggestion on why this is happening or can be fixed?

rantgithub, 2020-04-08:
I have a separated question

I am doing a validation now on some videos and the areas are so tight that the counters are almost useless, due to the angle of camera and the direction the objects are moving

I looked the directional counts for v3 and I think this will help a lot with this testing.

I would like to integrate that on v2, until v3 is fully tested.

I saw the logic under the opendatacam.js file but would like to check with you, if I want to add this to v2, what files are need to change.

Thank you

tdurand, 2020-04-09:
1. I think the frames lost are lost by Darknet, because OpenDataCam should be able to process anything that comes from darknet up to 100 FPS or more.. the display can be buggy but the actual detections frame should be processed. Or didn't I understand the problem ?

2. I don't understand why is preventing you to use v3 for this ? There is no easy diff available for the implementation of the directional counter, I touched many parts of the code, and there is lots of differents interdepend changes...

rantgithub, 2020-04-09:
Hi

1. I am not sure why this is happening, but is a "big" problem, specially with more than 1 instance. Only solution i have found so far, is to buffer the stream using a 3rd party, because with the direct url  m3u8, is just to fragmented. I will ask Alexyeb and see if he has any input

2. It is because i have touched many parts of the code on v2, in this way looking is how to put all together. I am testing this frames issue with v3 as well. I think we need a kind of internal buffer

rantgithub, 2020-04-10:
Hi

I received an answer from Alexyeb and there is not really realtime and lost of frames is expected, 

What he suggested

"
Ok try to compile without Optical flow tracker. I.e. commandt these lines
//#define TRACK_OPTFLOW
//#define GPU"

And add this line
cap.set(CV_CAP_PROP_BUFFERSIZE, 10);
between these 2 lines:

darknet/src/yolo_console_dll.cpp

Lines 360 to 361 in 3d9aa2a

 cv::VideoCapture cap; 
 if (filename == "web_camera") { 

I added and I think improved a little bit.

But for realtime, there is a need of 3rd party buffer, otherwise you will lost frames or freeze video a lot.

All is fine working with videos but not with direct feeds.

this is an example of with and without buffer, using as example VLC

[vlc.zip](https://github.com/opendatacam/opendatacam/files/4461362/vlc.zip)

rantgithub, 2020-04-10:
Hi

I have spent many hours testing this live stream and trying to figuring out what is this happening in odc, but i am experiencing a very big delay with processing any real time streams.

this is a link you can use for testing with "remote_cam" on the config file.

https://strmr5.sha.maryland.gov/rtplive/8301f15800f700d700437a45351f0214/chunklist_w1189770051.m3u8

I have tried to modify darknet as previously described, but no different results. I think this is an issue with Opencv or something else is causing this.

please let me know if you can reproduce this issue, because i dont see v3 usable as is, because the delay is very big.

thanks

tdurand, 2020-04-14:
let's discuss the potential improvements to the stream there : #194  and keep this for the CPU usage optim

tdurand, 2020-04-28:
@rantgithub looking into this today.. so changing to milliseconds(30) is still something we need ?

rantgithub, 2020-04-28:
I think no, difference was too small
30 put really really slow the live feed

tdurand, 2020-04-28:
ok, so closing this

closed 2020-04-28 by tdurand