opendatacam/opendatacam #190: [Darknet] Add Configurable ports option to ./uselib API

tdurand, 2020-04-06:
Right now we did a minimal fork of darknet ./uselib API : https://github.com/opendatacam/darknet/pull/5 to be able to integrate it with OpenDataCam , but we lost in the process the ability to make the JSON Stream and MJPEG stream port configurable.

This is handy in case you want to run multiple instances of OpenDataCam on one machine...

What we have to do is to edit the yolo_console_dll file to get the port info if they are specified and pass this info to both the json stream and mjpeg stream function. 

It shoudn't be very hard to do, should take as example how to parse arguments from the detector.c file: https://github.com/AlexeyAB/darknet/blob/master/src/detector.c#L1836

Goal would be to have:

`./uselib <yolo_params> <videoinput> -json_stream 8090 -mjpeg_stream 8070`

instead of 

`./uselib <yolo_params> <videoinput>`

cc @rantgithub if you have time to work on this

rantgithub, 2020-04-07:
hi

I am looking into this

tdurand, 2020-04-08:
Ok, make sure to start your work from this PR I did : https://github.com/opendatacam/darknet/pull/5 , that already makes changes to darknet ./uselib (adding Mjpeg stream and json stream) which aren't in the "default" ./uselib from ./darknet .

tdurand, 2020-04-28:
I just did the work to make ./uselib support those flags: https://github.com/opendatacam/darknet/pull/5/commits/51deaf6bdfb3da297b53da39788f0e9ff77eadcf

But I've also seen that now seems that the "old" API ./darknet demo problems with memory leaks on ARM devices is fixed https://github.com/AlexeyAB/darknet/issues/5214  , so I will double check this and if it is true just trash all the work done with ./uselib and go back to the ./darknet demo API as it is the one that support more flags etc etc.. and would enable us to have almost no difference between our darknet fork and the main one from alexeyAB..

rantgithub, 2020-04-28:
I saw this as well.

based on the comments from him when i asked about this cpu and streams,  darknet appears to be more robust than the uselib.

if the memoery leak was the main issue, this will address both issues.

tdurand, 2020-04-28:
yes, a bit of a pity I spent several days in total polishing the ./uselib API to trash everything... 😁️ But that's for the better ! I'll keep you updated tomorrow on this

rantgithub, 2020-04-28:
a quick question, i saw the  commits on demo.c and httpstream.c to test y4  with v2. 

i saw the changes with demo.c but i didnt see the httpstream.c file is listed

can you please confirm is only 1 file or 2 files?

i am testing the yv4 with v2  but some things are not working, like the auto load the video doesnt work at the end of the stream. I have compare the files and all looks fine to me.

trying to see what i am misisng

if the memory leak works i will move all the v3 because all fixes but for now i keep testing this with v2 due to the real feeds issue

tdurand, 2020-04-28:
@rantgithub the only fix to apply on darknet should be to disable the bbox into the image of the mjpeg stream: https://github.com/opendatacam/darknet/pull/1/files .. If nothing else changed since our fork it should do.. But I also need to revert some changes I did on OpenDataCamv3 to be compatible with a different format of JSON stream of the detections.

I would be you I'll just wait tomorrow / after tomorrow. I will work on this

rantgithub, 2020-04-28:
Correc,t  what i understood from the code is that will be straight forward, but that is not what is doing, reason i am asking.

i didn't see the other file on the commit, let me  compare and update all and i will follow up as well.

And yes, lets see if the memory leak is fixed.

rantgithub, 2020-04-28:
i checked all these 3 files and all is as yolov3

i think there is something else because 2 things are happening

1. on video, what the streams ends the odc freeezes
2. on a long count, also freezes because appears the stream goes away

anyway, just to provide feedback of testing for you to check this when you are making the changes.

:))

tdurand, 2020-04-29:
@rantgithub 

great news, I ran the for 8 hours the new darknet version with the "classic" API ./darknet and no memory leaks. 

So I went ahead to update our darknet fork and opendatacam v3 to go back to use the main darknet API ;-) and it works great.. here is how you can try it:

-  Reinstall darknet using the branch "odc" of our darknet fork https://github.com/opendatacam/darknet/tree/odc  .. The diff from the main alexeyAB darknet is just this: https://github.com/opendatacam/darknet/pull/6

- Pull the latest "develop" branch of opendatacam , don't forget to run `npm install` because I also updated lots of javascript dependencies.

I aim to release a beta.2 with all this and more later this week

tdurand, 2020-04-29:
PS: also I will make json_port and mjpeg_port configurable in the config.json

rantgithub, 2020-04-29:
:))

ahhhh very good to hear.

i will download an start to test :)

tdurand, 2020-04-29:
It is the development branch of opendatacam you should pull, not "develop", it doesn't exist: https://github.com/opendatacam/opendatacam/tree/development

rantgithub, 2020-04-30:
Hi

I downloaded and build. Testing all, initially all appears fine. But there is an issue I have been able to repro several times, but no really identify what is causing this error.

I leave odc running, just that on the screen, no counting or anything, and after several hours the screen is "white", meaning no streaming anymore.

When i check the console, the stream just stops : 

"known client: 56, sent = 30172, must be sent outlen = 30172
 MJPEG-stream sent. 
known client: 56, sent = 30179, must be sent outlen = 30179
 MJPEG-stream sent. 
known client: 56, sent = 30200, must be sent outlen = 30200
 MJPEG-stream sent. 
known client: 56, sent = 30218, must be sent outlen = 30218
 MJPEG-stream sent. 
known client: 56, sent = 30259, must be sent outlen = 30259:

and if i checked port 70,90 nothing is coming.

I see the darknet process running "normal", no cpu or memory spikes.

If i restart odb, now all comes back. (killing the task with ctrl+c, and restart odc)

any suggestion what to check  to trace why stops?

tdurand, 2020-04-30:
Thanks for the test.. hum weird.. This is likely a problem with darknet.. what we need is that the JSON stream doesn't stop, if the MJPEG stream hangs this is not a big problem as the counting would continue.. How long did it take to hang ? 

I just pushed a some more commits on the darknet fork (https://github.com/opendatacam/darknet/tree/odc)  branch "odc" to clean up the repeated logs of darknet.

I'm busy right now trying to get a v3.0.0-beta.2 out .. but next week will try to repro.. If the problem is just with the JSON / MJPEG stream, I'm not worried we will find a way..

tdurand, 2020-04-30:
Closing this one, BTW I made the ports configurable directly in config.json: https://github.com/opendatacam/opendatacam/blob/development/documentation/CONFIG.md#ports

closed 2020-04-30 by tdurand

rantgithub, 2020-04-30:
hi

i will continue testing......and yes i saw the ports as well

rantgithub, 2020-05-04:
Hi 

I am doing a verification for the total counts and I am finding a 10-30% discrepancy between how  the model counts on V2 against the new one on V3 when the slope of the line is stepper.
The higher the slope the less accurate the line counts.

I put side to side v2 with v3, same areas, feed, network and v2 is counting always more than v3.
(Now is counting right, meaning no other objects on the area like I initially reported)

If the line is horizontal, the counts are very similar, but the higher the slope is, the bigger is the difference on the count.

This is a sample i took for 4h.

[image]
[image]

I have tried couple of feeds and I am able to repro this.

Just to check if you can repro as well and see if is something on the math causing this behavior

rantgithub, 2020-05-04:
[image]
[image]

another example, same everything (video, network, lines)

tdurand, 2020-05-04:
Hi, 

Could you put this in a new issue ? Makes it easier to track and discuss. 

Thanks

commit 55b5d32a (2020-04-30): Evo: fix #190 make PORTS configurable from config.json