opendatacam/opendatacam #192: Can We run Opendatacam using Google colab notebook ?

ABMARK43, 2020-04-09:
Hey , i want to run opendatacam for demo purpose. Currently i do not own a Gpu device , so i wanted to know , how can i run it google colab notebook.

tdurand, 2020-04-09:
I don't not google colab notebook. On the cloud you should be able to install on any server that has SSH access to a Ubuntu and has a CUDA GPU available. We successfully deployed it on a Paperspace instance.

We do not have a specific doc for cloud installation, but if you follow this: https://github.com/opendatacam/opendatacam/blob/master/documentation/USE_WITHOUT_DOCKER.md , it should do

tdurand, 2020-04-09:
After having a quick look I think it is not possible with a  google colab notebook

ABMARK43, 2020-04-09:
Thank you for your time .

ABMARK43, 2020-04-09:
Though i will try at least once to run it on Colab notebook . My only
concern is , Instead of sending the output video to server can we save it
to a video file .

tdurand, 2020-04-09:
Hi, we don't save any output video, we just save the metadata for privacy reason

closed 2020-04-09 by tdurand