opendatacam/opendatacam #193: [Feature requests ] Various recording (GUI) requests

Denizzje, 2020-04-09:
Hello there,

I have been using the application "in the field" for a bit now, with great success. Longest run time I had was 27 hours, and that one got stopped because the mobile ISP decided that I needed a new IP in the middle of the night. Running another 24h+ session now.

 I would like to ask for the following features:

**1. Delete recordings confirmation prompt.**
It unfortunally has happened that me or someone else deleted a recording accidently, mostly after there was some delay at pressing the trash can button and the delete happening of the actual recording which was meant to delete (due to remote connection). Unfortunally a day of counter data now has been lost due to this. I would like to see some protection like an (optional) confirmation prompt to prevent accidental deletes.

**2. The ability to name recordings**
Right now, the programme seems to generate a random recording name. I would like to see it possible to name the recording. Even better would be if I can make a standardised recording name, and the programme adds a datetime to it to give an example.

**3. The ability to resume recordings**
Due to various reasons I had to stop a recording and start with a new one (crashes, IP change, accidently pulling the plug from my Jetson 😄 ). It would be great if I can just resume the previous recording instead of a new one.

Hopefully I am not asking too much ;).

tdurand, 2020-04-09:
Hi !!

Many thanks for this, love to see real life usage feedback

1. I can understand the need for this 😉 , it should be pretty simple to add, I guess we can do it for v3

2. Right now it generate a random id and append a date to it, but when you save it you are free to rename the file no ? 

3. This is a good idea, but won't be possible for v3 as it requires substancial work to implement properly.... but we will keep in around for the future.

Very interested in your long sessions usage feedback, and also would love to feature your project for v3 release if you like. (v3 release will be delay and happen likely in 1-2 months once the craziness of coronavirus will be a bit hopefully a bit less)

Denizzje, 2020-04-09:
No problem! Currently just passed the 2000 minutes mark , so 33.33 hours and the Jetson Nano still shows no signs of giving up (on v2 I could tell it was going to crash soon as the UI became unresponsive). I unfortunally have to end this recording tomorrow because  I have a newly trained model ready.

1. Great, sorry for that mess of a sentence there haha.

2. Indeed, when saving the CSV I can name it what I want, though Id like to automatise it as much as possible so I can either send data in a structured manner, or have the API "read out" for further processing. Now I first have to get the randomly generated name for the recording to find out what the name is somewhere. It would be great if I can for example do something like location_date. 

3. Awesome. I already figured that that isnt something you add with two lines of code. Hope it can be arranged sometime 👍 .

For the featured project, I will get back to that later 😄 .

tdurand, 2020-04-10:
> No problem! Currently just passed the 2000 minutes mark , so 33.33 hours and the Jetson Nano still shows no signs of giving up (on v2 I could tell it was going to crash soon as the UI became [...]

Nice !! 

1. no problem!

2. I understand, problem is that for the database it is much more convenient to have an unique id. How would you like to automatise the process in the best of the world ? because right now what you can do is:

request this API endpoint ( https://opendatacam.github.io/opendatacam/apidoc/#api-Recordings-List_recordings ) , that return list of recordings. (you can pass limit = 1 to have only the latest)

```
GET /recordings
```

And there you get the summary of the recording along with the id of it is you want to request more details,.

I think we don't want to add another feature with UI that give you the ability to name the recording, because it is not reliable, for example people name it "myrecording", and then name another one the same name.. and then you can't request it uniquely.. You could append the date to it, but then it is as complex as an ID to request.. (would need date + time)... Does it make sense ?

3. Yes, will add it on the backlog

> For the featured project, I will get back to that later .

Yes, with all the covid related events, we have decided to delay v3 release until a few weeks, so would be nice to get in touch in around in 2-3 weeks so maybe you can gives us more insight of you project and we can feature it as one of the use case of OpenDataCam. (Could you drop me an email so I have your contact ? my email is on my github profile)

tdurand, 2020-04-28:
1. Done with https://github.com/opendatacam/opendatacam/commit/627756860b0843bb05f7fb19eb5cfea4de18bcd5

[image]

tdurand, 2020-04-28:
Closing this one and tracking the remaining feature request here: https://github.com/opendatacam/opendatacam/issues/200

closed 2020-04-28 by tdurand