opendatacam/opendatacam #194: Performance issue - Long latency processing live feeds

rantgithub, 2020-04-10:
Hi.

From all tests performed, there is a big delay I am experiencing on V3 when playing live stream.

Please use this public link  to reproduce the issue.

https://strmr5.sha.maryland.gov/rtplive/8301f15800f700d700437a45351f0214/chunklist_w1189770051.m3u8

and using the config file, with "remote_cam" and this url, you can see how slow is the performance.

I tested this in/out of my network and the result was the same

As comparison if you open VLC and press CTRL+N, paste this link, you can see the difference between the live feed and what odc display.

appears there is a big bottleneck on opencv or similar, that is causing this big delay. 

Please let me know if you can reproduce this.

tdurand, 2020-04-14:
Hi !

Right now it is night on the stream so can't see much, will try later as it seems to be in the USA. But one question, is the performance diferent between ./uselib and ./darknet API ?

rantgithub, 2020-04-14:
Hi

Yes....using uselib, makes these stream extremely slow, the whole thing freezes....

With darknet we can work on this, freezes and moves, with uselib runs extremely slow

i have spent on this a lot of time and i think problem is coming from darknet

because even when i play it directly, darknet cant handle the m3u8. Can handle other feeds, but not that one, not sure why.

I asked alexyab and I tried his suggestions but no improvement. 

I think based on all this, restream is the only way to release darknet from it.

ODC is getting a freezing stream and is just displaying that.

If you do a video all works fine, this only happens with live feeds and not all of them, but m3u8, is very segmented. Once you can see the link, you will be able to duplicate the issue.

i will provide some feedback from restreaming

tdurand, 2020-04-14:
Both ./uselib and ./darknet uses OpenCV VideoCapture() API to read streams so you should get the same performance... ?

Also we don't want to complexify OpenDataCam with stuff that should be handle outside it. That said If you want to document how to setup a stream "proxy" I would be happy to include it in the documentation

Good luck in investigating this !

rantgithub, 2020-04-14:
to clarify, if you use darknet, no odc, you will see what i mean.

with darknet detectorcommand

i agreed this is not odc issue, but because there is an option with live feed for odc, and opencv can handle m3u8, if someone uses will causes odc to freeze/crash.

i took me a while to isolate the parts, and try to see which encoder was doing what and why.

i asked alexyab about a previous post about this and i am waiting for his response.

tdurand, 2020-04-14:
Ok great ! I'll let this open so you can update if you have any more info

rantgithub, 2020-04-15:
Hi

I have done many (many many :) ) tests and combinations to follow the behavior of darknet /yolo and odc and there is a significant difference between how a video is processed  against a real time feed .

1. a real-time stream, like a m3u8, causes a delay on the whole system. This doesnt happen is you stream this in a video and play the video after.

2. to isolated the issue, i tested darknet alone, with the link i previously  provide.

./darknet detector demo cfg/coco.data cfg/yolov3.cfg yolov3.weights https://strmr5.sha.maryland.gov/rtplive/8301f15800f700d700437a45351f0214/chunklist_w1189770051.m3u8 

the result is recorded on the video attached.

Even with some delays, the output is still good and there is a consistency with the play

3. I repeated the same, but now using ODC, v2 and v3.

with V2, i got delays of around 1-2 seconds in intervals of 5 seconds and the video still moves.
But V3, completed freezes the video for several seconds and performance is really bad and cpu goes 100% steady.

4. i have been able to duplicate this with any lived feed and in different pcs and from that is clear that there is a problem that is causing a big delay once odc receives the stream from yolo for this kind of feed.

I am using standard config file, with standard yolo weights, in this way you can reproduce the issue.

[config.zip](https://github.com/opendatacam/opendatacam/files/4484008/config.zip)

[darknetodcv3output.zip](https://github.com/opendatacam/opendatacam/files/4483972/darknetodcv3output.zip)

[odcoutputv3.zip](https://github.com/opendatacam/opendatacam/files/4484004/odcoutputv3.zip)

I think this is a big problem, because will only allow odc to work on recorded files but not feeds.

I have to try give it a bigger buffer coming from for instance VLc, or a ffmpeg server, with the result is the same.

Please test it and let me know if you can reproduce this issue on your end. 

I think there is a bottleneck at the time of waiting for frames and process, maybe with the  interaction of opencv. (If is a guess based on how many tests i have performed)

hope this summarizes all the tests i have done and can point to where the problem is coming.

Thanks as usual :)

rantgithub, 2020-04-15:
Something to add, i have tested in  different cards.

this test i did in 2 titan rtx with 48Gig of Ram and only running this, just to discard that is a lack of resources.

even with different configurations of hardware and environment same results are observed

hope all this helps :)

i hope you can reproduce this on your side

tdurand, 2020-04-16:
Great thanks, will try to repro next week when I will be working on the next beta version.

Could you try to compare directly ./darknet with ./uselib without using ODC ? My guess is that it is a problem coming from ./uselib

tdurand, 2020-04-16:
Also did you try with and without the fix done it : https://github.com/opendatacam/opendatacam/issues/189 ? maybe this introduce the bug ?

rantgithub, 2020-04-16:
Hi

Actually i though about that and i left everything "default" for v3.

Changing the parameter on fix #189  has not show any difference. I asked alexyab and try all suggestions and no difference for improvement. A bigger number like 30 make it crash completely. I left 3.

Although i agreed this is likely that use lib make it worst, this is a behavior that is also duplicated on version v2.

tdurand, 2020-04-28:
@rantgithub 

Ok I investigated this one, and I can reproduce the lag when you input the m3u8 stream directly into OpenCV ... it unusable you are right

Turns out that using gstreamer to decode the stream does work, here is an example pipeline you can put in config.json

```json
"souphttpsrc location=http://devimages.apple.com/iphone/samples/bipbop/gear4/prog_index.m3u8 ! hlsdemux ! decodebin ! videoconvert ! videoscale ! appsink"
```

If you want to try just without darknet / opendatacam

```bash
gst-launch-1.0 souphttpsrc location=http://devimages.apple.com/iphone/samples/bipbop/gear4/prog_index.m3u8 ! hlsdemux ! decodebin ! videoconvert ! videoscale ! autovideosink
```

But it seems there are issues with https stream for TLS certificate with the example one of maryland,  for now I couldn't play HTTPS stream with gstreamer, only with VLC. 

I'm trying to figure out this, but I guess this will be the fix, to rely on Gstreamer instead of OpenCV to decode streams

tdurand, 2020-04-28:
Seems unfortunately that there is a bug with HTTPS streams that is still unresolved: https://gitlab.freedesktop.org/gstreamer/gst-plugins-bad/issues/385    .... So I guess you will need to proxy the HTTPS stream into HTTP to be able to process it with Darknet / OpenDataCam

rantgithub, 2020-04-28:
Hi

Glad you were able to repro this.

The question that comes, why this gets really slow/crash using uselib. 

Without using uselib on v2/v3,  you get delays and freezes here and there but not like this.

From all this, restreaming from https m3u8 to http and to odc/darknet appears will be the solution.

I will create a re-streaming server and run some tests to evaluate the performance.

rantgithub, 2020-04-28:
question for the decoder

any major difference using opencv or gstreamer with odc? because even for darknet all is compiled using opencv

tdurand, 2020-04-28:
From what I understood (i'm really not an expert and I'm just doing trial & error for these things) , Gstreamer should be more efficient than OpenCV for this kind of stream handling, but OpenCV is still usefull because it takes the preprocessed output of Gstreamer and feed it to darknet.... 

I don't know why this is slower with uselib, for me it makes no sense as both ./uselib and ./darknet use VideoCapture() from OpenCV...

Anyways, unfortunately I can't spend much more time investigating this.. I guess it is good enough for me that we can support m3u8 stream by using a Gstreamer pipeline for HTTP.. and if it's HTTPS then you'll need to proxy it in some way... Actually would be nice if you can contribute some documentation if you succeed to do it...

rantgithub, 2020-04-28:
Ok

Yes, i am not an expert on this either, all these findings has been performing hundreds of tests.

and yes, if there is an option for m3u8 that will work as well.

i am testing this now and let me see what i can find

tdurand, 2020-05-05:
Closing this as we can't improve this much more.

closed 2020-05-05 by tdurand