opendatacam/opendatacam #195: New Paper: FairMOT

b-g, 2020-04-11:
Might be of interest ...

FairMOT – A simple baseline for one-shot multi-object tracking 

https://github.com/ifzhang/FairMOT

[image]

tdurand, 2020-04-14:
very interesting as it seems to outperform all current MOT challenge ranked algorithm, but also seems more heavy to run than our very simple  https://github.com/opendatacam/node-moving-things-tracker 😁️  . 

To keep around for future improvements

rantgithub, 2020-04-14:
I have a question

I am doing a project where the object id is more important than the count itself.  (for tracking purposes)

when i track a car, for instance, and it has object id 55, and when it moves in a street and let say there is a tree, when the car passes the tree the object id changes to 89. Because it lost track of it and identifies a different car before and after the tree.

Wondering is there is something can be tweaked to the tracker  have more "uiniques" objects id on the screen.

tdurand, 2020-04-15:
@rantgithub right now with our current tracker implementation : https://github.com/opendatacam/node-moving-things-tracker there is not much you can do to improve this.. If YOLO looses the object for more than 5 frames our tracker will consider it gone. You could try to increase this threshold: https://github.com/opendatacam/node-moving-things-tracker/blob/master/tracker.js#L10  , but then you are at risk of maintaning lots of "ghost" that can actually do more arm than good. 

A future plan could be to use more advanced ways doing Multiple object tracking like the one presented in this paper to be smarter about this,  but the thing is that we need first to have a way to properly benchmark the performances of OpenDataCam : https://github.com/opendatacam/opendatacam/issues/87 to be able to assess if it would improves / worsen the results.. Which is very tricky to do.. you can read the issue if interested, because it depends mostly on which scenario / what kind of scene you are looking to track..