opendatacam/opendatacam #196: [Feature Request] Any plans to make configuration file (conf.json) updated via Settings Page?

isra60, 2020-04-14:
Hi. Is there any plans to make a settings page to modify the JSON configuration file and make the app more user friendly? I suppose the react component should be capable of reset the darknet command or other process.

tdurand, 2020-04-14:
Hi

Yes we want to do that in a future version, but it is not that straightforward to implement and we have limited time to dedicate to it unfortunately.

bilx6043, 2020-04-20:
i have tried to update config file via server and it successfully updated.. but need to restart the ODC again to take effect..so i add a restart command in server to make it more user friendly..

tdurand, 2020-04-20:
Nice, if you are able to share you work in a Pull request.. (I'm not sure how this could work if you install opendatacam as a docker container)

bilx6043, 2020-04-21:
> Nice, if you are able to share you work in a Pull request.. (I'm not sure how this could work if you install opendatacam as a docker container)

i havent tested this on docker container.. i run without docker and change the server file.. sure i will share it

zeljkomarinkovic, 2021-02-26:
@bilx6043 How is it going with this? Have you managed to do something? Can you share code so I can contribute too?