opendatacam/opendatacam #197: issue in recording ID

bilx6043, 2020-04-16:
version 3.0.0-Beta.1

i download and try without docker.. everything seems to be perfect and normal.. when i click the recording button it starts recording. but when i see status through localhost:8080/status there is no recording id.. when i start recording through localhost:8080/recording/start it gives an error.. but the recording is started without recording id.. 

is there any bug or am i dong something wrong

bilx6043, 2020-04-16:
if i see the log it states that 

start recording
TypeError: Cannot read property 'id' of undefined
   at Object.startRecording(/home/ppk/AIM/opendatacam/server/opendatacam.js:518:90)

sometimes when i hit localhost:8080/recording/start it give OK but most of the time it gives error.. that is a bit weird.. may be i have done something wrong

tdurand, 2020-04-16:
Hi, 

Sorry, yes this bug has been fixed since we released beta-1 .

Please try to pull the "development" branch: https://github.com/opendatacam/opendatacam/tree/development

And recompile ODC:

```
npm i
npm run build
```

let me know.

bilx6043, 2020-04-16:
> Hi,
> 
> Sorry, yes this bug has been fixed since we released beta-1 .
> 
> Please try to pull the "development" branch: https://github.com/opendatacam/opendatacam/tree/development
> 
> And recompile ODC:
> 
> ```
> npm i
> npm run build
> ```
> 
> let me know.

ok i did the same n it seems that the issue is resolved/.. thanks

tdurand, 2020-04-16:
Ok ! Gonna release a v3.0.0-beta.2 next week with this fix integrated

closed 2020-04-16 by tdurand

bilx6043, 2020-04-21:
> Ok ! Gonna release a v3.0.0-beta.2 next week with this fix integrated

ok .. awaiting that .. i have tried the same thing new module i am still facing the recording id issue.. waiting for beta version .. thanks

tdurand, 2020-04-21:
It should come this week, but until then you can just pull from the development branch as I told you , didn't it fix the bug ?

bilx6043, 2020-04-22:
oh sorry my bad yes i pull the development branch its fixed already