opendatacam/opendatacam #199: Yolov4

tdurand, 2020-04-25:
Yolov4 came out with what it seems to be great improvements in accuracy for the big weight file: https://arxiv.org/abs/2004.10934

Doesn't seem to have a tiny-yolov4 yet: https://github.com/AlexeyAB/darknet/issues/5310

Good news is that should be straitforward to make opendatacam support yolov4 as one of the contributor of this paper is AlexeyAB, the author of the fork of darknet we use.

Will need to spend time to figure out which weight is best to "default" ship with OpenDataCam v3 for jetson nano, tx2, xavier

shams3049, 2020-04-25:
I am not able to find the weights for V4, can you point me to them. I can try them out in my setup, I have all three version of Jetsons.

tdurand, 2020-04-25:
AlexeyAB updated darknet yesterday with them: https://github.com/AlexeyAB/darknet#pre-trained-models

But I guess our darknet fork used with opendatacam might not be compatible right now with them.. I need first to update : https://github.com/opendatacam/darknet/tree/opendatacamv3

rantgithub, 2020-04-26:
Hi

quick question

for version 3, we are using uselib and we cant use it with v2. What  what files will be needed to update to use odc version2 with yolov4 ?

I am already testing yolov4  and would like to use it on the current v2.

Thank you

tdurand, 2020-04-27:
@rantgithub Hi, 

You would need to try to apply those patches (those tagged with v2): https://github.com/opendatacam/darknet/pulls  to the an to date version of darknet.

BTW this week I finally was able to free some time to make a hopefully close to final push to add the missing things to the v3... so will be in touch on the other issues

tdurand, 2020-04-27:
Also there is plan to have yolov4-tiny: https://github.com/AlexeyAB/darknet/issues/5346#issuecomment-619628732

rantgithub, 2020-04-27:
hi

ok

i actually tested v4 and the response is very good and the amount of memory has reduced. I tested  during training and during inference.

but still the issue of the live feed remains.

good, just let me know

tdurand, 2020-05-05:
Closing this, v3.0.0-beta.2 will support YOLOv4

closed 2020-05-05 by tdurand