opendatacam/opendatacam #203: DepthAI Support - Spatial AI Support w/ OpenVINO

Luxonis-Brandon, 2020-05-01:
Hi everyone!

We stumbled upon your website today and love this project.  So we make an open-source 'spatial AI' camera called DepthAI.  We'd really like to integrate it with opendatacam so that opendatacam can get spatial results from the realworld, real-time.

[DepthAI](https://www.crowdsupply.com/luxonis/depthai) combines disparity depth and neural inference (object detection) to product structured data of what objects are and where they are in physical space.  Below is an example:
[image]

So the idea is to be able to build things that real-time measure the world in physical coordinates, without having to do camera calibration to the environment.  Just plop it and the device works.  And for real-time, here's and example:
https://photos.app.goo.gl/3HaGBT7FW4XoQAL66
(P.S., ignore the random z errors in that one, it's using an uncalibrated unit while I was testing latency.)

So we'd love to get this integrated in opendatacam!  Since we discovered your system today, we don't yet know how to start.  Since all our stuff is open source though, it shouldn't be that bad:

Python: [https://github.com/luxonis/depthai](https://github.com/luxonis/depthai)
C++ API: [https://github.com/luxonis/depthai-api](https://github.com/luxonis/depthai-api) 
Hardware: [https://github.com/luxonis/depthai-hardware](https://github.com/luxonis/depthai-hardware)

The platform is OpenVINO-based, and we are making open-source training flows for it as well.  

Example here: [https://docs.luxonis.com/tutorials/object_det_mnssv2_training/](https://docs.luxonis.com/tutorials/object_det_mnssv2_training/)  (Things like this will eventually have their own [Github](https://docs.luxonis.com/tutorials/object_det_mnssv2_training/) as well.)

Thoughts?

Thanks,
Brandon

Luxonis-Brandon, 2020-05-01:
Just to follow up, we compiled this on the Tx2 and it works great!  Just followed the build instructions for OS X ([here](https://docs.luxonis.com/api/#build-depthai-and-test-for-mac-os-x) actually, and they worked on the Tx2 w/ no error at all, which is cool.

So here's the visualization of the metadate coming off DepthAI on the Tx2:
[image]

[image]

A note... w/ DepthAI, since all the AI processing is done on DepthAI, you actually don't need a powerful host then...  anything would work (even a microcontroller).  

So then opendatacam could install on things which are way to slow to process neural inference/etc.

Thoughts?

Thanks,
Brandon

tdurand, 2020-05-01:
Hi, 

Thanks for you message. Yes this is alright to contact us through the GitHub 😉️.... 

Firstly very nice project you are doing, I think it is great to see people doing things outside the NVIDIA kingdom !

I found this diagram on your website which helped me understand what is DepthAI doing:

[image]

Here is the OpenDataCam architecture:

[image]

From what I see integrating with Depth AI would mean delegating all the lowerpart (neural network process + video input) to you and only use OpenDataCam for data processing and the nice UI.

The benefits would be 

- to have the 3D coordinates of the objects as a new metadata
- support other hardware alternative.. seems yours is priced around 350 USD ? (your processing unit + stereo camera) , but it does run something less accurate than YOLO, is it MobileSSD, does it do multiple object detections ?  

On the cost of doing this I see:

- Make an easy install of OpenDataCam "lite" on raspberry pi (not sure if docker is supported) because it makes no sense to install it on a Jetson as it won't be using any processing power
- Write some custom "glue code" to hook to your detection stream and your video stream and start / stop it
- Maintain documentation + support changes from your end
- Generally add lots of complexity to the project

Also right now the most pragmatic path to support 3D tracking in OpenDataCam would be test and document using a ZED camera: https://github.com/opendatacam/opendatacam/issues/81  as it is already supported on the darknet framework we use.. I admit less power efficient and more costly, but pretty straightforward.

I will be honest here, our project aims to be as simple and as minimal as possible.. We don't make money out of it so this is way too big for us.. Aside supporting this on the long term... it is already lots of work to catch up with one platform (jetson's) and one neural network architecture (YOLO), so can't imagine beeing able to add more. We are even thinking of [integrating deeper with Nvidia Ecosystem and use Deepstream SDK](https://github.com/opendatacam/opendatacam/issues/112) to be able to have less work on maintaining & testing the darknet framework

That said if you want to do it the other way around and offer an OpenDataCam fork to install on the raspberry pi cam.. We can speak of doing consultancy for you, or you can also do it yourself as our work is licensed MIT 🎉️

Happy to continue the discussion though, and let me know if I overlooked something regarding a potential integration of the projects .

Have a nice day 

Thibault

Luxonis-Brandon, 2020-05-05:
Hi @tdurand ,

Thanks for the thorough answer!

> support other hardware alternative.. seems yours is priced around 350 USD ? (your processing unit + stereo camera)?

Yes, the 3-camera (2 for depth, 1 for 12MP stills and 4K video) is [$299](https://shop.luxonis.com/collections/all/products/bw10980bc).  And single-camera (12MP/4K video) is [$199](https://shop.luxonis.com/collections/all/products/bw1093).

> But it does run something less accurate than YOLO, is it MobileSSD, does it do multiple object detections?

So it can run YOLOv3 as well (although we haven't personally tried it yet, see [here](https://github.com/AlexeyAB/darknet/issues/5079#issue-585403577) for approximate framerates.

We're mostly interested in running YOLOv4, which will actually have a variant, YOLOv4-VPU for YOLOv4 at 30 FPS on DepthAI (called VPU below, Intel's name for the Myriad X):
https://github.com/AlexeyAB/darknet/issues/5079#issuecomment-620741750

So this is then the highest accuracy currently-available real-time object detector out there, which also will run at 30FPS on DepthAI.  

>Make an easy install of OpenDataCam "lite" on raspberry pi (not sure if docker is supported) because it makes no sense to install it on a Jetson as it won't be using any processing power

So a popular use-case we've seen is actually to use DepthAI with a more powerful host such that the first-stage neural inference is performed by DepthAI, and the powerful host then has resources to run a second stage (and/or 3rd, 4th, 5th stage) without running out of resources or becoming too slow.  

So for example if you wanted to do object detection on the DepthAI to find all the people and vehicles, and then run neural networks on the detected people for their emotion, their gaze estimation, pose estimation, the vehicle type, the license plate detection and OCR, etc.  Or even things like combining LSTMs to do action 

When DepthAI offloads the first-stage then the host is free to do these additional stages (and many of them) as its GPU (or neural processor) isn't consumed by doing the object detection.

And then all of these can be run real-time (with xyz location of the detected objects).  So you can have a log of the trajectory of all the people who look angry, etc.  Heh.

>Write some custom "glue code" to hook to your detection stream and your video stream and start / stop it

FWIW, here's the format for object detectors:
https://github.com/luxonis/depthai/blob/master/resources/nn/mobilenet-ssd/mobilenet-ssd_depth.json

>Maintain documentation + support changes from your end

Definitely true.  And we do change things fairly quickly these days since we're relatively early-on.  So a valid concern.

> Generally add lots of complexity to the project

Fortunately on say a Jetson Nano effectively everything is already included.  So just enabling the camera and parsing the results above is all that's required additionally.  

>Also right now the most pragmatic path to support 3D tracking in OpenDataCam would be test and document using a ZED camera: #81 as it is already supported on the darknet framework we use.. I admit less power efficient and more costly, but pretty straightforward.

So we can get YOLOv3 (darknet) running on DepthAI I don't think with much problem.  We did MobileNetSSD because it's faster and our core focus has been on real-time application.  And we're wanting to put more work into YOLOv4 (we already have it training, see below):
[image]

Once we get a decent flow, we'll post another one of these tutorial series on how to train it, like [here](https://docs.luxonis.com/tutorials/object_det_mnssv2_training/).  And yes I had actually noticed #81 .  ZED looks to make great products. And the D435 mentioned there is the `2-generations-ago` version of DepthAI (it uses the Intel D4, the successor of which is the Myriad 2, and the successor of that is the Myriad X, what's in DepthAI).

> I will be honest here, our project aims to be as simple and as minimal as possible.. We don't make money out of it so this is way too big for us.. Aside supporting this on the long term... it is [...]

Yes, makes a ton of sensoe to me.  And Deepstream seems really cool.  One marketing slide I saw on it showed 8x 1920x1080 person+vehicle+bicycle object detections running at 30FPS.  Which seemed unbelievable.  They've since taken that down (and I didn't think to screenshot it), so I'm not sure if it was a typo.

>That said if you want to do it the other way around and offer an OpenDataCam fork to install on the raspberry pi hooked to Myriad X.. We can speak of doing consultancy for you, or you can also do it yourself as our work is licensed MIT 🎉️

Yes, I think a fork (or PR, if you're open to it, it seems like it would be easy for us) would make a ton of sense.  We're also MIT license FTW.  🎉️

>Happy to continue the discussion though, and let me know if I overlooked something regarding a potential integration of the projects .

I think you're spot-on in your understanding and analysis.  

The easiest way to propose integration could be for us to just do a PR.  And make it purely as a low- (and ideally, no-) encumberance option.

tdurand, 2020-05-05:
Hi !

> Yes, the 3-camera (2 for depth, 1 for 12MP stills and 4K video) is $299. And single-camera (12MP/4K video) is $199.

Nice, but this needs also the processing board Myriam X ?? Or is it included ?

> We're mostly interested in running YOLOv4, which will actually have a variant, YOLOv4-VPU for YOLOv4 at 30 FPS on DepthAI (called VPU below, Intel's name for the Myriad X):
> AlexeyAB/darknet#5079 (comment)
> 
> So this is then the highest accuracy currently-available real-time object detector out there, which also will run at 30FPS on DepthAI.

Impressive indeed !

> So a popular use-case we've seen is actually to use DepthAI with a more powerful host such that the first-stage neural inference is performed by DepthAI, and the powerful host then has resources to [...]

Okay nice, but yes this must increase deployment & setup complexity as you need to install software on two distinct hardware ? 

> [image]

If I may, about this use case, with OpenDataCam we are trying to avoid promoting running any "face" related examples, and in general things that are related to people (like people counting leaving supermarkets or social distancing...)... We are aware that our tool can enable to do this.. but at least we try not to promote it.. We think some of these use cases might be a bit too much of "tech solutionism" and the best way to help with them might not be to have a Camera staring at people... Also this is very much a constant debate and other contributors of this projects might have other opinions on this... but you should know that we are opinionated about this.. _( recently also the guy behind the YOLO architecture [stopped doing computer vision](https://twitter.com/pjreddie/status/1230524770350817280) cause of ethics concern )_ 

> And yes I had actually noticed #81 . ZED looks to make great products. And the D435 mentioned there is the 2-generations-ago version of DepthAI (it uses the Intel D4, the successor of which is the Myriad 2, and the successor of that is the Myriad X, what's in DepthAI).

Ok, I have 0 knowledge of the 3D stereo camera + processing current "market" status .. good to know

> Yes, makes a ton of sensoe to me. And Deepstream seems really cool. One marketing slide I saw on it showed 8x 1920x1080 person+vehicle+bicycle object detections running at 30FPS. Which seemed [...]

Yes, but their developer experience + documentation is very bad right now if you don't have C/C++ background.... Integrating with it isn't easy at all.. But I guess with the next version it will be easier.

> The easiest way to propose integration could be for us to just do a PR. And make it purely as a low- (and ideally, no-) encumberance option.

Yes I think this would be the way to go. Actually we can exchange when you start looking into this.. 

In the best of the worlds we would like to see OpenDataCam as "neural network" agnostic and have for example 

- a plugin to hook to darknet
- a plugin to hook to deepstream
- a plugin to hook to depthAI
- a plugin to hook to iOS / Android

And only support one "official" plugin and easy setup (on Jetson for now) .. but leave it open for people to contribute documentation for third party integration. 

It is actually not very hard to do, I think there is only 3 pieces of code that would need to be abstracted into more generic functions that could be then rewrited for other platforms.

- YOLO process orchestration 
- JSON stream hooking 
- MJPEG stream hooking

Luxonis-Brandon, 2020-05-05:
Will respond in more detail soon.  But wanted to answer this question real-quick:

> Nice, but this needs also the processing board Myriam X ?? Or is it included ?

This is included.  So it runs the neural inference and produces metadata back out over USB (so theoretically you don't even need video output, just structured data).  

And it can in parallel do 1080p h.264 or h.265 (HEVC) encoding so the host doesn't have to do this.  Which can be used with `python3 test.py -v [path/to/video.h264]` and then hooked to gstreamer or what have you.

[https://docs.luxonis.com/faq/#how-do-i-record-video-with-depthai](https://docs.luxonis.com/faq/#how-do-i-record-video-with-depthai

tdurand, 2020-05-06:
> This is included. So it runs the neural inference and produces metadata back out over USB (so theoretically you don't even need video output, just structured data).

Very nice, so this makes it even more accessible 😉️

Luxonis-Brandon, 2020-05-06:
OK, sorry about the delay.

> Very nice, so this makes it even more accessible 😉️

Thanks!

> Okay nice, but yes this must increase deployment & setup complexity as you need to install software on two distinct hardware ?

So DepthAI actually USB-boots.  So you don't have to install any software on it.  So when you run `python3 test.py` it's actually loading everything from the host (takes about a second on USB2) to run.  So you never have to worry about loading software to it, upgrading it, etc.  It just runs what it's told from the host... very similar to a USB camera, but a `smart` USB camera which can run neural networks, object tracking, etc.

> If I may, about this use case, with OpenDataCam we are trying to avoid promoting running any "face" related examples, and in general things that are related to people (like people counting leaving [...]

Thanks for the heads up!  

Not sure if you've seen his resume.  When I naively brought up YOLO to very-experienced AI/ML guys several years ago after belatedly learning about YOLO, they (1) laughed at me and then (2) said he was `simultaneously the most and least employable person in machine learning` and showed me his resume [here](https://pjreddie.com/static/Redmon%20Resume.pdf), and then I understood what they were meaning.

> Yes, but their developer experience + documentation is very bad right now if you don't have C/C++ background.... Integrating with it isn't easy at all.. But I guess with the next version it will be easier.

Good to know.  So we've done full TRT optimization and it was really painful but we were able to get our models to improve from 1FPS (TensorFlow on Tx2), to 5FPS w/ TF-TRT, to ~35-45 FPS w/ full TRT on Tx2.  It seems like NVIDIA w/ DeepStream is trying to make this less painful and easier to integrate... so it's funny to hear about the pain points given my impression of their goal here.  :-)

Re: the plugin approach.  Yes, sounds great!  

Thanks again!

tdurand, 2020-05-06:
> So DepthAI actually USB-boots. So you don't have to install any software on it. So when you run python3 test.py it's actually loading everything from the host (takes about a second on USB2) to run. [...]

Great, this sounds very easy to integrate, plug & play, I like this ;-) 

> Not sure if you've seen his resume. When I naively brought up YOLO to very-experienced AI/ML guys several years ago after belatedly learning about YOLO, they (1) laughed at me and then (2) said he [...]

I actually like the resume ;-), I see many benefits in "unorthodox" ways of thinking. And his concerns regarding the societal impacts of computer vision / AI are very relevant.

> Good to know. So we've done full TRT optimization and it was really painful but we were able to get our models to improve from 1FPS (TensorFlow on Tx2), to 5FPS w/ TF-TRT, to ~35-45 FPS w/ full TRT [...]

True, we started OpenDataCam in the first place because we felt the "ease of use" of neural network that very hard for "beginners" .. We don't have a big C/C++ or AI/ML background, we just wanted to enable people to solve meaningful uses case in an user friendly way.

I guess the Deepstream API felt very cryptic for us but it might no be for everyone... Also I think it is mostly caused by a very bad documentation lacking concrete example... But this will improve with Deepstream v5.0.

> Re: the plugin approach. Yes, sounds great!

👌️

Luxonis-Brandon, 2020-05-06:
>I actually like the resume ;-), I see many benefits in "unorthodox" ways of thinking. And his concerns regarding the societal impacts of computer vision / AI are very relevant.

It also reflects a bravery that I don't have.  :-)

>True, we started OpenDataCam in the first place because we felt the "ease of use" of neural network that very hard for "beginners" .. We don't have a big C/C++ or AI/ML background, we just wanted to enable people to solve meaningful uses case in an user friendly way.

Yes, totally agree.  We have a shared mission there in that the whole goal of DepthAI is to allow spatial AI to be used by anyone, and embedded into products and prototypes.  

We started it because my friends kept getting run over by distracted drivers when riding their bikes and there was no platform that allowed mitigating this distracted-driver-killing-people-who-ride-bikes problem using computer vision.  And such a platform had to be embedded to enable such a smart tail light.

So we either had to give up, or make the platform.  We chose to make the platform.

See here for more details:
[https://discuss.luxonis.com/d/8-it-works-working-prototype-of-commute-guardian](https://discuss.luxonis.com/d/8-it-works-working-prototype-of-commute-guardian)

We're getting very close on DepthAI to being able to productize to make a smart bike light like described there.  We need to do hard-sync between depth and AI results to allow quite-fast-moving objects to still be tracked in 3D space accurately, particularly when there is extreme lateral motion (i.e. side impact instead of from behind).  But we're quite close.  

Cheers,
Brandon

tdurand, 2020-05-06:
> We started it because my friends kept getting run over by distracted drivers when riding their bikes and there was no platform that allowed mitigating this [...]

Nice use case to solve !!

AlexeyAB, 2020-05-09:
@tdurand @Luxonis-Brandon Hi,

> JSON stream hooking
> MJPEG stream hooking

Yes, I think the best way to combine DepthAI with OpenDataCam - implement TCP-streams JSON/MJPEG in DepthAI.
So if DepthAI will implement it:
1. then no changes to OpenDataCam are needed (except for reading extra parameter Depth or Z-coordinate)
2. DepthAI and OpenDataCam can be hosted on different devices.
3. One OpenDataCam instance can work with several DepthAI instances/cameras.
4. You can restart one soft (DepthAI) without restarting another soft (OpenDataCam)

OpenDataCam already uses this approach and reads Darknet's JSON/MJPEG streams.
For example, in the Darknet you can use:
`./darknet detector demo cfg/coco.data cfg/yolov3.cfg yolov3.weights test.mp4 -json_port 8070  -mjpeg_port 8090`
or
`./darknet detector demo cfg/coco.data cfg/yolov3.cfg yolov3.weights test.mp4 -json_port 8070  -mjpeg_port 8090 -dontdraw_bbox`

Then if you open http://127.0.0.1:8070/ in the Chrome/Firefox, you will see such Json:
```
{
 "frame_id":905, 
 "objects": [ 
  {"class_id":14, "name":"bird", "relative_coordinates":{"center_x":0.488930, "center_y":0.294208, "width":0.039871, "height":0.044511}, "confidence":0.653603}, 
  {"class_id":14, "name":"bird", "relative_coordinates":{"center_x":0.668744, "center_y":0.946393, "width":0.047600, "height":0.063851}, "confidence":0.304215}
 ] 
}, 
```

----

> If I may, about this use case, with OpenDataCam we are trying to avoid promoting running any "face" related examples, and in general things that are related to people (like people counting leaving supermarkets or social distancing...)... Also this is very much a constant debate and other contributors of this projects might have other opinions on this... but you should know that we are opinionated about this.. ( recently also the guy behind the YOLO architecture [stopped doing computer vision cause of ethics concern](https://twitter.com/pjreddie/status/1230524770350817280) )

1. Only do what you like, technology non-disclosure arrangements, ethics concern, ... there can be many reasons, but the most positive is worth voicing. Jan 2020: Apple reportedly acquired Xnor.ai, a Seattle-based startup for $200 million https://www.forbes.com/sites/janakirammsv/2020/01/19/apple-acquires-xnorai-to-bolster-ai-at-the-edge/#5d6871be3975 Feb 2020: Joseph stopped doing CV research.

2. In this case **Edge-computing (smart-cameras)**  are more suited than Cloud, because all videos are not transferred anywhere and are not collected, because it is processed on the device (inplace). So this partially solves the problem of data collection (ethics concern).

3. **Edge-computing** (detection/recognition on smart cameras) due to low accuracy/speed are ideally suited for **advisory tasks** - when the device gives hints, but does not control something. If it gave the right clues - well, if it gave the wrong ones, **then it did no harm.** F.e.: search for offenses, crimes, shooting, fires, dangerous driving, traffic accidents, safety precautions, social distance (covid19), if someone fell or someone became ill ...

----

In general, csdarknet53-opt (YOLOv4 without Mish-activation) 256x256 (**async=3**) 
* 11 FPS / (1 watt Myriad X)  = 11 FPS per watt
* ~100 FPS / (180 watt RTX2070) = 0.55 FPS per watt (if TRT is used)
So Neurochip (MyriadX) can be 20x times more efficient than GPU (RTX).

So I think neurochips can be very promising tech and smart-camera with neurochips can be a very promising product.

tdurand, 2020-05-10:
@AlexeyAB thanks for jumping in !

> Yes, I think the best way to combine DepthAI with OpenDataCam - implement TCP-streams JSON/MJPEG in DepthAI.

yes, this would be the easiest way 🙌️..  but also open to have "adaptors" in case other technical solution is chose to stream detections / camera feed

----

> Only do what you like, technology non-disclosure arrangements, ethics concern, ... there can be many reasons, but the most positive is worth voicing. Jan 2020: Apple reportedly acquired Xnor.ai, a Seattle-based startup for $200 million https://www.forbes.com/sites/janakirammsv/2020/01/19/apple-acquires-xnorai-to-bolster-ai-at-the-edge/#5d6871be3975 Feb 2020: Joseph stopped doing CV research.
> 
> In this case Edge-computing (smart-cameras) are more suited than Cloud, because all videos are not transferred anywhere and are not collected, because it is processed on the device (inplace). So this partially solves the problem of data collection (ethics concern).
> 
> Edge-computing (detection/recognition on smart cameras) due to low accuracy/speed are ideally suited for advisory tasks - when the device gives hints, but does not control something. If it gave the right clues - well, if it gave the wrong ones, then it did no harm. F.e.: search for offenses, crimes, shooting, fires, dangerous driving, traffic accidents, safety precautions, social distance (covid19), if someone fell or someone became ill ..

Thanks for jumping in ! It is very nice to see those discussions happen... My views are a bit different for some of theses things as for example I believe that solving social distancing (for covid) isn't a technological problem: https://algorithmwatch.org/en/our-position-on-adms-and-the-fight-against-covid19/ , and introducing half broken solutions can do more harm than good.

But I also believe that they are plenty of use cases that can benefit of easy, affordable and accessible computer vision..  I like a lot the views from the people at fast.ai that say that "we need everyone involved with AI" in order to build solutions that take in account as much edge cases and diversity as possible : https://www.youtube.com/watch?v=LqjP7O9SxOM

----

> In general, csdarknet53-opt (YOLOv4 without Mish-activation) 256x256 (async=3)
> 11 FPS / (1 watt Myriad X) = 11 FPS per watt
> ~100 FPS / (180 watt RTX2070) = 0.55 FPS per watt (if TRT is used)
> So [...]

This benchmark is very impressive... would mean that with a 200 USD depthAi module ( 1 W) + a  Raspberry Pi or other low consumption board ( 1 W) running OpenDataCam you can have same performance as a Jetson Xavier that uses 20-30W of power for a third of the price 🤯️. AND that is also 10-12x more power efficient 🔋️🔋️.

Luxonis-Brandon, 2020-05-12:
So as an update we will be jumping on our initial implementation for this starting at the end of this month (when our intern starts, who is working remote of course).  He's already perused this all and is excited to get going on it!

Best,
Brandon

Luxonis-Brandon, 2020-05-31:
So as a quick update we have the MJPEG streaming from DepthAI POC working:
https://github.com/luxonis/depthai-experiments/blob/master/mjpeg-streaming/README.md

So we ran into an issue with the Jetson Nano we have (it’s not booting) otherwise I think we’d already have an example running with OpenDataCam. 

We also are working on OpenDataCam on macOS but hit a snag on npm installation that we haven’t debugged yet.

Luxonis-Steven, 2020-06-01:
Hi all, another quick update. Right now, we have JSON stream to port 8070 and MJPEG stream to port 8090 while DepthAI is running on a Raspberry Pi 4 with Ubuntu 18.04. We also have OpenDataCam installed on Raspberry Pi 4. 

Next, we need to find where to modify on OpenDataCam (for example, the parameter related to darknet is no longer applied since DepthAI did that part job).

tdurand, 2020-06-01:
Hi !

Sounds great !

> We also are working on OpenDataCam on macOS but hit a snag on npm installation that we haven’t debugged yet.

Never tried on macOS as there was no nvidia graphic card... let me know

> Hi all, another quick update. Right now, we have JSON stream to port 8070 and MJPEG stream to port 8090 while DepthAI is running on a Raspberry Pi 4 with Ubuntu 18.04. We also have OpenDataCam installed on Raspberry Pi 4.

🔥️  good job

> Next, we need to find where to modify on OpenDataCam (for example, the parameter related to darknet is no longer applied since DepthAI did that part job).

You'll find some info here: https://github.com/opendatacam/opendatacam/issues/212#issuecomment-631549967 .. 

Maybe we could add a new flag in config.json

```json
NEURAL_NETWORK_BASE: "darknet" || "depthai"
```

And rename 

`PATH_TO_YOLO_DARKNET` to `PATH_TO_NEURAL_NETWORK` 

And then you could add the params needed in `NEURAL_NETWORK_PARAMS` under new entries

Then in the YOLO.js file: https://github.com/opendatacam/opendatacam/blob/master/server/processes/YOLO.js (we could rename it) , we would have some logic like

```
if(config.NEURAL_NETWORK_BASE === "depthai") {

} 

if(config.NEURAL_NETWORK_BASE === "darknet") {

} 
```

This is just some ideas I have right now but I'm not sure how you launch depthai

Luxonis-Steven, 2020-06-01:
Awesome! Thanks a lot! I've checked the #212 and #112 - they are super useful. From what I see right now, three main files (config.json, server.js, and YOLO.js) need to be updated to hook with depthai. Let me know if anything missing. 

> Never tried on macOS as there was no Nvidia graphic card... let me know

Yes, correct - but since depthai does the NN process, we don't need Nvidia graphic card anymore when OpenDataCam runs. Please correct me if I am missing anything. But macOS has an issue with installing buffertools (node module).

> This is just some ideas I have right now but I'm not sure how you launch depthai

Thanks again. To run depthai, we use `python3 test.py` in terminal. I tested run OpenDataCam and depthai separately. OpenDataCam can get results from JSONStream and stuck on the initializing page for now. Will keep you posted.

tdurand, 2020-06-01:
> Awesome! Thanks a lot! I've checked the #212 and #112 - they are super useful. From what I see right now, three main files (config.json, server.js, and YOLO.js) need to be updated to hook with depthai. Let me know if anything missing.

I think so, but just try and let me know.. feel free to ask if you run into troubles.. 

> Yes, correct - but since depthai does the NN process, we don't need Nvidia graphic card anymore when OpenDataCam runs. Please correct me if I am missing anything. But macOS has an issue with installing buffertools (node module).

Yes of course, I was actually sucessfully running it from MacOS in simulation mode a while ago.. So it should work.. Sometimes node is a bit messy.. Make sure you have node.js >= 10.x , remove node_modules , remove package-lock.json and try to npm install again.. If it doesn't work please open an issue with the logs will try to help you debug

> Thanks again. To run depthai, we use python3 test.py in terminal. I tested run OpenDataCam and depthai separately. OpenDataCam can get results from JSONStream and stuck on the initializing page for now. Will keep you posted.

okay.. yes for now the integration with darknet is a bit hacky so you might have to change some stuff to integrate with depthai, let me know

tdurand, 2020-06-02:
FYI i'm running into the same issue with buffertools when npm install on ubuntu.. Not sure why.. I think this is an issue with node v12+ with an old dependency

tdurand, 2020-06-02:
Ok, I've removed an old dependency that caused problem with node v12.x , if you pull the development branch it should work now . commit : https://github.com/opendatacam/opendatacam/commit/2be0907cd9c1cec48c84e300a4dc54a03f841148

Luxonis-Steven, 2020-06-02:
Sweet! Yes I had it on ubuntu when my node v12+ (so I changed back to v10). I just tested and successfully installed OpenDataCam on macOS. Thanks!

Luxonis-Steven, 2020-06-03:
Hi, an approach I made: I tried to disconnect YOLO.js with server.js and OpenDataCam.js and then ran `npm run start` and depthai separately (keep them running at the same time). I can see: 
- JSONStream by using `console.log(msgchunk)` in OpenDataCam.js 
- MJPEGStream on `http://IPaddress:8080/webcam/stream`

But it stays on the initializing page forever - never get into the actual UI after filled the entire progress bar. Also, in the terminal, it doesn't show any error as well - only notice - 
`setvideoresolution `
`> Ready on http://localhost:8080`
`> Ready on http://IP:8080`
`Success init db`
`Got first message from JSONStream`

If there is anything you may aware of, please feel free to let me know. Thanks again.

tdurand, 2020-06-03:
Sounds good progress !

Hum.. and if you put some console.log in there: https://github.com/opendatacam/opendatacam/blob/master/server/Opendatacam.js#L211 ? 

It needs to call `updateWithNewFrame` function here: https://github.com/opendatacam/opendatacam/blob/master/server/Opendatacam.js#L647 to then set `isListeningToYOLO` to true to remove the init screen.

Also you might have to force this to true: https://github.com/opendatacam/opendatacam/blob/master/server/processes/YOLO.js#L83 , but I think this doesn't change a thing.. it is just there for the API and not used by the UI

Let me know.. It is very tightly coupled to darknet/YOLO as you see ;-)

Luxonis-Steven, 2020-06-03:
Quick update - I got it work (the UI turns on)!! Was my mistake to format the JSON string by adding an extra comma at the end - so OpenDataCam can't identify JSON but can output the JSON with console.log. 

Next, I will test the features and see how it works with DepthAI.

tdurand, 2020-06-03:
Very nice congrats !! 

For mongodb on mac you should look at this doc: https://docs.mongodb.com/manual/tutorial/install-mongodb-on-os-x/

Luxonis-Brandon, 2020-06-03:
So @tdurand @themarpe just had the idea of actually doing the MJPEG video encoding directly on DepthAI/megaAI which would then make it effectively no load to the host as everything would be done on DepthAI/megaAI and the host would effectively be a relay.

See: https://github.com/luxonis/depthai/issues/127

So we'll be adding that not too long from now.  :-)  Should make integration even simpler.

tdurand, 2020-06-03:
I guess it is a good feature for you to have for other integrations ;-)

Luxonis-Brandon, 2020-06-03:
Yes in the short term I'm going to use it to show video on my iPhone to my handlebars on my bike with the [BW1097](https://shop.luxonis.com/products/depthai-rpi-compute-module-edition) mounted to the back of the bike like shown [here](https://discuss.luxonis.com/d/8-it-works-working-prototype-of-commute-guardian/10).

Cheers,
Brandon

Luxonis-Steven, 2020-06-05:
FYI OpenDataCam with DepthAI also works on macOS as well.

tdurand, 2020-06-05:
nice ! yes should work with any platform supporting DepthAI and supporting node.js + mongodb

b-g, 2020-06-08:
Hi @Luxonis-Brandon! Very nice work! Could you put together a DepthAI shopping list for me so that I can give it a go as well? (Is there any Edu discount? Would buy it for [my university](https://www.hfg-gmuend.de/) and our IoT lab there :)

Luxonis-Brandon, 2020-06-09:
Hi @b-g ! 

Awesome, thanks!  (And sorry about the delay!) 

Yes definitely can do (listed below).  And yes we can do an EDU discount as well.  So email brandon at luxonis dot com and I'll give you a code that will allow you to purchase units from shop.luxonis.com with the EDU discount.  

Shopping list:
 - DepthAI: Onboard Cameras Edition ([here](https://shop.luxonis.com/products/bw10980bc) - works with any host and gives Spatial AI information.  

So this is a 3-onboard-camera solution w/ onboard integration of stereo depth and neural results to give spatial results (all done on the Myriad X built into the product).  

For example, XYZ location of people, like [here](https://github.com/luxonis/depthai-experiments/tree/master/social-distancing).

 - megaAI: Monocular Camera [here](https://shop.luxonis.com/collections/all/products/bw1093)  This is for applications that don't need spatial information.  It has the same 12MP color camera as `DepthAI: Onboard Cameras Edition`, but no stereo pair, so no spatial information. 

 - DepthAI: Modular Cameras Edition - use this if you need spatial information -and- you want a custom stereo baseline.  So for example if you're doing things especially close-in (like strawberry picking  or of Strawberries, like [here](https://github.com/luxonis/depthai-ml-training/tree/master/colab-notebooks#easy-object-detector-training-)), having the flexibility to change the stereo baseline can be super handy.  And same goes for things that are especially far away (the included FFC support up to ~25cm stereo baseline).  (For the minimum depth calculation see [here](https://docs.luxonis.com/faq/#what-are-the-minimum-and-maximum-depth-visible-by-depthai).)

Luxonis-Brandon, 2020-06-10:
Hi @b-g (my initial twin!, I too am b-g (Brandon Gilles))

Super!  Vielen Dank und bis bald!  (Ich habe zum Spaß probiert, Deutsch zu lernen!)

shams3049, 2020-07-02:
Was just going through my feed and found this. As some point this techniques can replace the need for Depth camera altogether. 

Source link: https://roxanneluo.github.io/Consistent-Video-Depth-Estimation/
Video Explanation: https://www.youtube.com/watch?v=QSVrKK_uHoU

Luxonis-Brandon, 2020-07-02:
Thanks @shams3049 , yes I had actually seen that (actually I posted about it [here](https://discuss.luxonis.com/d/41-using-temporal-information).

So the trade with those sorts of monocular depth estimators vs. binocular depth is processing/compute needs.

So it's like squeezing a balloon (as most trades are described).  So here's the summary:

Monocular Depth:  
Advantage: 1 camera
Disadvantages: 
- Significant compute load (most I've seen require ~100 TOPS or so for real-time depth of binocular quality/accuracy).
- Requires a lot of training, can get confused by scenes.  So putting a toy car in front of a monocular depth estimator trained on real-life size cars will give very-wrong results.

Binocular (Stereo) Depth:
Advantage: Low compute (for example it runs in parallel to neural inference on DepthAI)
Disadvantage: 
- 2 cameras, which in some cases is annoying/prohibitive in terms of industrial design integration (e.g. on a mirror on a Tesla)
- Semi-Global matching does not work great in some situations, so it can have holes/etc.

So actually longer-term on DepthAI (and perhaps even megaAI) we're going to investigate using neural networks for depth work.  We don't know if we're have enough neural compute to do it though, and if so, how hard it will be.  Example things we're going to investigate:

 - Can we do monocular depth (say off of [megaAI](https://www.crowdsupply.com/luxonis/megaai))?
 - Can we improve binocular (semi-global matching) depth results using a neural network.

So then the trade on which to use is based on the engineer constraints.  So on Teslas they have tremendously-large neural compute available, and the industrial design is super important (it's a lot of the reason the cars are sold), so on many models they do monocular cameras combined with radar.  But even in their case, it looks like from leaked photos newer Tesla models will include stereo.

So on small/embedded applications where huge compute isn't available (i.e. you don't have a RTX 2080Ti level of neural compute), binocular/stereo depth is highly advantageous as it's such lower power... and leaves what neural compute you may have for things like object detection, pose estimation, etc. that you may need the hardware for.

And all that said, binocular/stereo depth combined with neural inference for improving the depth will always be more accurate than monocular depth, all other things held constant, as it has so many more degrees of freedom to leverage to learn what the actual depth is in the scene. 

You can actually see the same things in humans.  Folks who are blind in one eye do still have depth perception.  The occipital lobe adapts and actually uses more resources for this when one eye is lost.  But, they never have the same level of depth perception as with two eyes, simply because there is less information available (specifically, the disparity information is extremely valuable for depth information, especially when fed into a neural network).

I hope that all makes sense.

Thoughts?

Thanks,
Brandon

shams3049, 2020-07-02:
@Luxonis-Brandon Thanks for the rundown. Traditionally I can agree with you on all the aspects of a dedicated stereo camera. 

But for the sake of **innovation**, I see a workflow here, specifically, Deepstream architecture allows for a pipeline with multiple models. Such a solution appears superior to me in the long run. One simple reason that I can come up with is that a neural network based solutions can be radially adopted once key architecture is layed down, where as a stereo camera tech might be slow in comparison.

AlexeyAB, 2020-07-02:
@Luxonis-Brandon Hi,
It seems for each range these are apporaches which give better accuracy:
* for the lowest distance - depth estimation based on brightness
* for low-medium disatance - depth estimation based on stereo cameras
* for high disatance - depth estimation based on monocular camera (since disparity of stereo-camera is ~1px)
    * based on 1 frame: https://arxiv.org/abs/1907.01341
    * based on several frames: https://arxiv.org/abs/2004.15021

For each approach we can use either hand-written algorithm or neural network.
I think in the future, there will be a single network for all 3 approaches.

Maybe soon I will do acceleration of monocular depth estimation.

Luxonis-Brandon, 2020-07-02:
> @Luxonis-Brandon Thanks for the rundown. Traditionally I can agree with you on all the aspects of a dedicated stereo camera.
>
> But for the sake of **innovation**, I see a workflow here, specifically, [...]

Thanks.  Yes, that's right.  As you mention for research/innovation purposes (where often high-compute is a tacit requirement) having the flexibility affording by neural-only is a big advantage.

Luxonis-Brandon, 2020-07-02:
> @Luxonis-Brandon Hi,
> It seems for each range these are approaches which give better accuracy:
> 
> * for the lowest distance - depth estimation based on brightness
> * for low-medium distance - depth estimation based on stereo cameras
> * for high distance - depth estimation based on monocular camera (since disparity of stereo-camera is ~1px)
>   
>   * based on 1 frame: https://arxiv.org/abs/1907.01341
>   * based on several frames: https://arxiv.org/abs/2004.15021
> 
> For each approach we can use either hand-written algorithm or neural network.
> I think in the future, there will be a single network for all 3 approaches.
> 
> Maybe soon I will do acceleration of monocular depth estimation.

Thanks!  Great summary.  And yes for long distance where the disparity becomes 1 pixel, monocular neural techniques make a ton of sense.  And in fact above a certain distance in humans, stereo depth is no longer used, it's purely monocular.  I don't remember what distance that is, though.

And would love to see some acceleration of monocular approaches, particularly if you could get something that could say run on a Myriad X.  That would be really neat.

>I think in the future, there will be a single network for all 3 approaches.

Yes, that would be super neat.

So speaking of which, is it's super-likely that neural techniques applied to stereo images likely can glean out information quite well when there is less than 1 pixel separation.  As the network can likely learn slight difference is brightness/etc. to the point where I would guess sub-pixel disparity down to perhaps 1/20th of a pixel?  Algorithmic now can do 1/5th of a pixel, so I would be shocked if such neural approaches couldn't do better.  (We can do 1/5th sub-pixel on DepthAI but we haven't released it yet.)

Thanks again,
Brandon

Luxonis-Brandon, 2020-07-30:
Oh I should have posted here sooner!  So I know there are a few folks on here who were interested in buying some units.  

So we have a KickStarter now with 50% off the MSRP going for the next 13 days:
https://www.kickstarter.com/projects/opencv/opencv-ai-kit/description

Thanks,
Brandon

b-g, 2020-08-03:
Hi @Luxonis-Brandon, many thanks for the hint! Great! Just backed a "OAK-D Campaign Edition" ... much looking forward to Dec 2020! :) Best of luck with the Kickstarter!

tdurand, 2020-12-24:
Some nice insights on the OAK-D  : https://blog.roboflow.com/luxonis-oak-d-custom-model/

Luxonis-Brandon, 2020-12-28:
Thanks for the post.  We are huge fans of what Roboflow is doing.  Many of our customers are using Roboflow for their model development.

bharath5673, 2021-01-20:
hello @Luxonis-Brandon ....

is there any ways to run OAK just with .xml and .bin files ???

Luxonis-Brandon, 2021-01-20:
So the .xml and .bin need to be converted to a .blob that DepthAI uses.  OpenVINO actually does this behind the scenes for NCS2, for example.  We have an online converter to do this ([here](http://luxonis.com:8080/)) and then a guide for doing it locally, [here](https://docs.luxonis.com/en/latest/pages/tutorials/local_convert_openvino/).  And the following community-member (from Geax) example shows how to take some OpenVINO .bin and .xml and convert them to run on OAK:
https://github.com/geaxgx/oakd_palm_detection

Thoughts?

Thanks,
Brandon

bharath5673, 2021-01-20:
> So the .xml and .bin need to be converted to a .blob that DepthAI uses. OpenVINO actually does this behind the scenes for NCS2, for example. We have an online converter to do this ([here](http://luxonis.com:8080/)) and then a guide for doing it locally, [here](https://docs.luxonis.com/en/latest/pages/tutorials/local_convert_openvino/). And the following community-member (from Geax) example shows how to take some OpenVINO .bin and .xml and convert them to run on OAK:
> https://github.com/geaxgx/oakd_palm_detection
> 
> Thoughts?
> 
> (Oh and here is that model running: https://photos.app.goo.gl/e5utk4uKFNtuNgtn7)
> 
> Thanks,
> Brandon

Yes I got u, I can do all that openvino conversions locally, but wat I'm worried abt is. Blob 😬 , n how it works, and how to do it on my local machines..

Coz I tried some custom nn, openvino converted them successfully and tried to blob them,, but it throws error :(((

And really tysm @Luxonis-Brandon  for this wonderful device ❤️❤️❤️❤️❤️❤️❤️💕

Luxonis-Brandon, 2021-01-20:
Thanks for the kind words!  So some networks need additional normalization/conversion settings.  See [here](https://docs.luxonis.com/en/latest/pages/faq/#my-model-requires-pre-processing-normalization-for-example-how-do-i-do-that-in-depthai) for example.  And please feel free to join our Discord to discuss more real-ish-time and also some community members can probably advise as well for your specific model: [https://discord.gg/EPsZHkg9Nx](https://discord.gg/EPsZHkg9Nx)

Thanks again!

Luxonis-Brandon, 2021-08-16:
Hi everyone,

So we just had an idea internally on OpenDataCam that we wanted to share.  Now that we have PoE models, it's likely possible to run OpenDataCam with literally just an OAK-D-PoE or OAK-1-PoE.  

If that's of interest we could make a version that literally has this pre-flashed onto an OAK-D-POE (or 1-PoE) where all that's needed is to plug the device into Ethernet and it shows up as a fully-configured OpenDataCam (including with depth-sensing capability).

Would this be of interest?

Thanks,
Brandon

tdurand, 2021-08-19:
Hi,

I guess any additional hardware that can run OpenDataCam is of interest 😉 ,  I'm sorry I didn't follow closely the OAK latest developments, but please let us know what kind of inputs you would need from us to do this ? 

We are working on releasing a v4 version : https://github.com/opendatacam/opendatacam/issues/402 , that will be mostly the same but with every dependency up to date.. which is running late but we could then add a section featuring ODC on the OAK..

Best, 

Thibault

themarpe, 2021-08-19:
Hi @tdurand 
I think we actually have everything ready - mostly just exposing the idea of this capability, which I think could be pair very nicely with the platform. 
The main idea is that the host in this case could be much less performant without requirements for a strong GPU, etc.. as it would only deal with streaming and business logic of the analyzing and using the detection information coming from the device.

Below is a picture with a coarse explanation of this. 
PoE models, like OAK-1-PoE have capability to run basically any kind of networking server on them. That means that creating a DepthAI App that creates the exact same server, but flashed to the device running completely standalone would be very simple (we now officially have on device Python scripting support, which makes spinning this kind of things up very easy). The App would serve both JSON stream of detections and an MJPEG camera stream, directly from the camera.

I think this could be a great fit for your platform as it could really remove the need for the host complexity and make it such that even less performant hosts would be able to easily deal with multiple devices at once.

The main difference between an USB device and PoE in this case is that USB devices still needs some support in the codebase, to expose the incoming data to a server for JSON and MJPEG streams. While PoE devices would be fully decoupled of that and opendatacam would just have to get the IP and connect to it.

What do you think?

[image]

tdurand, 2021-08-20:
sounds good, I guess for a first try the best way to go would be for you to fork and do a proof of concept of this we can point people to ? and then we see if we can have this integrated in the codebase.. I guess we would just need to remove the part where we launch the darknet process.. 

PS: you can do a POC from the `development` branch as plenty has changed there

b-g, 2021-08-20:
Yes would be mega to have with this feature! An even more of the box setup would be quite a step forward! IMO highly appreciated be a lot of people!

> The main difference between an USB device and PoE in this case is that USB devices still needs some support in the codebase, to expose the incoming data to a server for JSON and MJPEG streams. While [...]

Does this mean that non PoE OAK devices are out of luck in the near future? My impression from the Kickstarter is that the minority of the backers purchased the PoE options (or?) ... and as the ODC project aims for accessibility, it would be highly preferable to support a large(r) part of the OAK devices.

themarpe, 2021-08-20:
@b-g USB devices aren't out of scope, just that they must be a tad more coupled and require `depthai` dependency. @Luxonis-Steven did an initial PoC for USB devices a while ago over at https://github.com/luxonis/opendatacam `development` branch AFAIK. 

If we revisit, will most likely cover all 3 options:
 - Standalone PoE (flashed app that exposes JSON & MJPEG streams  on its own)
 - regular PoE usecase (starts the app on the device on "spawn" event and creates the JSON & MJPEG streams)
 - USB (same as regular PoE) 

Standalone PoE is IMO architecturally very nice as you'd be able to basically create a product (eg. IP Camera of sorts, ...) which could be "opendatacam compatible". Along its own streams/services it would also expose the required MJPEG & JSON streams

Luxonis-Brandon, 2021-08-20:
Yes, and the POE versions are already IP67 sealed so good for outdoor use.  Agreed that we should just cover all options though.  

[image]

b-g, 2021-08-23:
@themarpe @Luxonis-Brandon 2x wow! :)

vsaw, 2022-10-18:
I think this has been idle for years. Sorry :-/

DonChef660, 2022-10-19:
Hej Luxonis Team :) @Luxonis-Brandon, @themarpe & @Luxonis-Steven 
Is/was there any progress from your side?
If not, any plans to do so?
Would even greatly appreciate, if you could point me to the right direction, how to setup the oak-d lite with OpenDataCam!

@vsaw The OpenDataCam Project is still alive right? Any Roadmap for V4?

Thanks guys your work is much appreciated!

vsaw, 2022-10-19:
@DonChef660 See [State of the OpenDataCam Project - Q3 2022](https://github.com/opendatacam/opendatacam/discussions/565)

themarpe, 2022-10-25:
Hi @DonChef660 

Things are a tad out of date now, but the above post https://github.com/opendatacam/opendatacam/issues/203#issuecomment-902765713 should guide you into a general direction.

I doubt you'll be able to run things without modifications, but if you rely on latest v2 DepthAI library, and refactor how Detection results are passed onwards to OpenDataCam you should have a solution in hand to support OAK-D Lite devices :)