opendatacam/opendatacam #204: Line slope affects total count

rantgithub, 2020-05-04:
Hi

I am doing a verification for the total counts and I am finding a 10-30% discrepancy between how the model counts on V2 against the new one on V3 when the slope of the line is stepper.
The higher the slope the less accurate the line counts.

I put side to side v2 with v3, same areas, feed, network and v2 is counting always more than v3.
(Now is counting right, meaning no other objects on the area like I initially reported)

If the line is horizontal, the counts are very similar, but the higher the slope is, the bigger is the difference on the count.

This is a sample i took for 4h.

[image]
[image]
[image]
[image]

tdurand, 2020-05-04:
Hi, 

thanks for the report. might be possible that it is less sensitive for higher slope if the object isn't super well tracked as a piece of the code is reasoning on the Y delta.... But I'm not sure..

- Could you do this experiment but having the line with different slopes exactly in the same spot ? Because for example on the first example, objects might be better tracked for the 113 line area and a bit less well tracked for the 103 zone.. does it make sense? Do something like the green line: 

[image]

- Another thing you can try is to toggle ON the "tracker accuracy" in the menu and see if there are more id re-assigment there .

- Also if you can send me the example footage after doing those test so I can try to reproduce

Thanks !

rantgithub, 2020-05-04:
Hi

I have done several tests with live feeds.

here are another results, which have been very consistent.  V3 always counts less than V2.

[image]
[image]

you can try this public link, (hopefully works for you)

https://strmr5.sha.maryland.gov/rtplive/5e0089a0025c0075004d823633235daa/chunklist_w230336176.m3u8

and just put the 2 lines with a slope like on the pic and see if you can repro my results.

just let it run for 1-2 h and see if you get a difference like i have.

Just to clarify, i am using the same settings in both, including the network size and model. 

:)

tdurand, 2020-05-04:
thanks, unfortunately this week I won't have time to spend on this (and generally I think if I don't have a quick way to repro it with video footage it is hard for me to dedicate several hours into this).. 

I'm not sure I understand what is really the problem you are looking to fix:

- Is it that the algorithm for line counting in v2 was more accurate than v3 ( it was right to count more things ?)

- Is it the line slope difference ? in this case please can you do the experiment I mentionned here in my previous comment : https://github.com/opendatacam/opendatacam/issues/204#issuecomment-623270912  ?

rantgithub, 2020-05-04:
As summary, the algorithm for V2 it looks like more accurate than the one for V3 for steeper slopes. v2 always counts more objects than v3 does. (10-30% difference). For lines with very low slope (horizontal lines) the count is very similar.

This comparison is based in taking the same video and loading the same lines and count over the same amount of time using the same network and tracker settings.

Based on my tests you can repro this with any video that you have. 

Let me work on having a video for you of 1-2h and you can test there.

I think there is something on the calculation that is making the count less accurate.

rantgithub, 2020-05-14:
Hi

Just to follow up on this.

I have tested 1-4h of videos side-side (v2-v3)  and it is clear the slope makes a good difference in the total. For just experimentation, appears thta a slopes more than 50% (to the eye) make the counter line to ignore a good percentage (20-30%) on my tests.

With slopes under 45^, the count is very similar.

rantgithub, 2020-05-14:
I have a quesiton, before open a new issue

when you drop a video on the screen, the expected behavior is to start to count like a real time feed?

because i upload the video and when i press the record button starts to count but when the video end doesnt clear the counter.

I put a short 5min video, and at the end of the video the count continues again.

Meaning in a normal video, you count from beginning to end and counter is reset at the end.

On this, the video ends and if you have total of 50, when the video restarts the next object goest to 51.

Checking if this is expected behavior or a bug

tdurand, 2020-05-14:
> Hi
>
> Just to follow up on this.
>
> I have tested 1-4h of videos side-side (v2-v3) and it is clear the slope makes a good difference in the total. For just experimentation, appears thta a slopes more [...]

could you send me some test footage ? like something of 10-15 min.. with the counter configurations you are using ?  So I can reproduce

> I have a quesiton, before open a new issue
>
> when you drop a video on the screen, the expected behavior is to start to count like a real time feed?
>
> because i upload the video and when i press the [...]

please open a new issue, seems to be a bug. Expected behaviour when you start recording on any file is OpenDataCam should restart, record and then at the end of the file stop t the recording / counting.

rantgithub, 2020-05-14:
Hi

Ok, i was able to create a video for you

here is the video and the areas as test.

What i have observed is that even the recognition and tracker are fine, the doesn't
[odcv3.zip](https://github.com/opendatacam/opendatacam/files/4631447/odcv3.zip)

 get counted when crosses the line. If you do a manual count you can see what I mean. Over 1-2h period, this makes a big difference in the total number.

As I mentioned this is reproducible in may videos with a slope like this

rantgithub, 2020-05-14:
and this is one of the things is hard to explain, 

I did a line and see how it shows after restart

there is not way to delete it

[image]

same video as I sent it.

is detected but is completely  outside of boundaries.

rantgithub, 2020-05-15:
[odc.tar.parta.zip](https://github.com/opendatacam/opendatacam/files/4632188/odc.tar.parta.zip)

rantgithub, 2020-05-15:
[odc.tar.partb.zip](https://github.com/opendatacam/opendatacam/files/4632189/odc.tar.partb.zip)

rantgithub, 2020-05-15:
[odc.tar.partc.zip](https://github.com/opendatacam/opendatacam/files/4632192/odc.tar.partc.zip)

rantgithub, 2020-05-15:
[odc.tar.partd.zip](https://github.com/opendatacam/opendatacam/files/4632195/odc.tar.partd.zip)

tdurand, 2020-05-15:
Hi @rantgithub !

Thanks again for all your test ! I will spend time investigating this next week !

rantgithub, 2020-05-15:
Hi

Quick question, you will see on this video that we get those "cars"/"objects" that takes 80% of the screen.

I remember you mentioned this a couple weeks ago about the max size. Just to check if this was implemented because i tried the limit sizes, but no suscess.

any suggestion what we need to change to prevent this big boxes to show up.

Thanks

most of the videos i have, shows these big objects detected that makes the screen look awful and some times, trigger counters on the lines depending of the position.

tdurand, 2020-05-15:
not done yet, but on the todo list for next version: https://github.com/opendatacam/opendatacam/issues/205 : " Add Thresshold to filter out large detections" 😉

tdurand, 2020-06-02:
@rantgithub could you send me the test footage with [wetransfer](https://wetransfer.com/) or [firefox send](https://send.firefox.com/) or anything else ;-) .. when I extract your zip file the video aren't working..

I'm gonna look into improving the counting algorithm today

tdurand, 2020-06-02:
Hi @rantgithub 

Could you try to checkout the branch `fix-counting` (Do a `npm install` again as I updated some dependencies)

I've simplified a bit the counting logic ([see diff](https://github.com/opendatacam/opendatacam/pull/216)), maybe this will improve things.. I didn't try with your footage because I can't extract the video

Let me know

rantgithub, 2020-06-02:
Hi @tdurand 

Sorry i just saw the video request......

Let me download and test.

It is good timing because actually I am doing several validations now and I can compare.

I will follow up

rantgithub, 2020-06-03:
Hi

I am comparing and I have a quick question to make the same count with all the same.

on the original calculation the deltaY was

  let deltaY = - trackedItem.y - countingAreaProps.a * trackedItem.x - countingAreaProps.b;

and to adjust the position due to, for instances close/big objects, the H was added/subtracted like

let deltaY = - (trackedItem.y - trackedItem.h)  - countingAreaProps.a * trackedItem.x - countingAreaProps.b;

because this was removed on the new logic, where this can be adjusted to have the same result as when using deltaY.

Thanks

tdurand, 2020-06-03:
Hi 

You should play with

```
let intersection = checkLineIntersection(
              countingAreaProps.point1.x,
              countingAreaProps.point1.y,
              countingAreaProps.point2.x,
              countingAreaProps.point2.y,
              trackerItemLastFrame.x,
              - trackerItemLastFrame.y,
              trackedItem.x,
              - trackedItem.y)
```

which is using the center of bbox to something like this which is this case will be computing the potential crossing of the line with the trajectory of the bottom of the bbox with the counting line (I didn't test be this should work)

```
let intersection = checkLineIntersection(
              countingAreaProps.point1.x,
              countingAreaProps.point1.y,
              countingAreaProps.point2.x,
              countingAreaProps.point2.y,
              trackerItemLastFrame.x,
              - (trackerItemLastFrame.y - trackerItemLastFrame.h / 2),
              trackedItem.x,
              - (trackedItem.y - trackedItem.h / 2)
```

Thanks

tdurand, 2020-06-03:
PS: I think that what was causing the difference between v2 and v3 was that you tweaked `deltaY` but not `checkLineIntersection` .

By getting rid of deltaY you should have consistent results hopefully. Just let me know..

rantgithub, 2020-06-03:
I am doing several tests and i am trying to understand how this is counting/behaving

I understand what you said and the logic and I will expect the same result but, 

the horizontal lines are very accurate, but when the box of an object crosses a "vertical" line, is not triggering  the counter

just to give you an idea, on one example is suppose to have +100 and the new logic counted only 5, even i saw the crosses

i am setting a full lab fopr this, and if i can repro consistently will send videos and zones.

there is something that is not doing what is supposed to do and I am trying to confirm what is the problem

tdurand, 2020-06-03:
Ok.. Please send me one video footage + counter config so I can reproduce.

rantgithub, 2020-06-03:
Hi

I think I found the problem after many tests.

Please see the video attached

Based on the logic, same id, should not be counted twice. For the video you can see maybe we lost the car 1, meaning we should counted 2, but what is happening is been counted it 15 times. (from 3-18 on the video)

This only happens when the object is close by the line, in a typical stop sign. When is straight you dont see this behavior, but once you are in a traffic/stop/go condition like on the video, the counter goes "crazy" and over long period of time, the difference is huge.

I think this is similar of what was happening when I saw the counter was detecting other cars.

That is why my straight lines are  accurate and lines on stop are very bad.
[odc_counting_double.zip](https://github.com/opendatacam/opendatacam/files/4726411/odc_counting_double.zip)

rantgithub, 2020-06-04:
Hi again

After more testing to give more details, what is happening the id is changing due to a poor network recognition and because the box is on the edges of the counting line, triggers multiples counts at the same time. the main issue with this behavior, in an unattended count after 4 hours, we got +400 difference, which off course is not good.

But what we have seen is that appears that the line on this position is still counting objects that are not crossing the line, meaning similar issue when i reported this behavior.

just thinking here, on this condition one solution will be to determine if the position of the boundary box is the same, but the issue with that will come it is possible another object is on the same position and not be counted.

We have to readjust the lines to more "traffic" flow to minimize this. We are testing to confirm that in fact the lines still counts under certain slopes another objects even when the object doesn't cross the line. I will keep you post it.

tdurand, 2020-06-04:
> That is why my straight lines are accurate and lines on stop are very bad.
> odc_counting_double.zip

Yes this is definitely a problem of ID switches... Unfortunately we can do nothing about it with the counting algorithm... This would need to be solve at the tracker level: https://github.com/opendatacam/node-moving-things-tracker/issues/10

> Based on the logic, same id, should not be counted twice. For the video you can see maybe we lost the car 1, meaning we should counted 2, but what is happening is been counted it 15 times. (from 3-18 on the video)

Yes this "guard" is already built in in the code : https://github.com/opendatacam/opendatacam/blob/fix-counting/server/Opendatacam.js#L384

> just thinking here, on this condition one solution will be to determine if the position of the boundary box is the same, but the issue with that will come it is possible another object is on the same position and not be counted.

I see what you are saying.. and yes this is very hard to do because you can optimize for this case of cars stopped at the traffic sign.. but then this will cause problem for other tracking situation... Back in the days I spent LOTs of time playing with this.. and it is hard ;-) . 

What I can do it that there is two parameters of the tracker that I will try to expose in the config.json so you can play around with this... Here having DEFAULT_UNMATCHEDFRAMES_TOLERANCE higher might help.

```javascript
// DEFAULT_UNMATCHEDFRAMES_TOLERANCE 
// This the number of frame we wait when an object isn't matched before considering it gone and assign a  new id to it
var DEFAULT_UNMATCHEDFRAMES_TOLERANCE = 5;
// IOU_LIMIT, exclude things from beeing matched if their IOU (intersection over union) is lower than this
// 1 means total overlap whereas 0 means no overlap
var IOU_LIMIT = 0.05
```

All that said, I would like to merge this simplification of the counting algorithm ... Before I do maybe you can tell me if you see big differences (in a bad way) ? In my opinion it shouldn't change much the results from previous versions.

tdurand, 2020-06-04:
FYI: Added new params to the config.json for fine tuning tracker settings:

https://github.com/opendatacam/opendatacam/blob/development/config.json#L13

Documentation: https://github.com/opendatacam/opendatacam/blob/development/documentation/CONFIG.md#tracker-settings

I also merged the `fix-counting` changes into development . Even if it does not change the result it simplifies the codebase.

rantgithub, 2020-06-04:
Hi

Ok, very good, this will help to play with settings under some of these similar situations

I agreed the overall will be the same result.

What we saw yesterday a couple of times, is the counter increased without the object crossing the line. I should say happened like 3-5 times. Cars where travelling parallel to the line and the counter increased.

I am testing to verify this is not happening anymore, because that should not happen with the current logic.

I will provide feedback after more testing

tdurand, 2020-06-04:
Ok thanks ! will close this for now, just reopen in the case you have more to add to the discussion !

closed 2020-06-04 by tdurand

rantgithub, 2020-06-04:
just to clarify, which branch/version has the latest ? to download all again

tdurand, 2020-06-04:
`development` , will delete the other branchs