opendatacam/opendatacam #205: New version: v3.0.0-beta.2 , need help with testing

tdurand, 2020-05-06:
### Releasing v3.0.0-beta.2 🎉️ , featuring:
 
- Support for YOLOv4 (set `"NEURAL_NETWORK": "yolov4"` in config.json)
- Upgraded docker setup and build workflow, see https://github.com/opendatacam/opendatacam/issues/178 , now you can run the docker container from jetson without needing to hack the support for CUDA.. It also simplified the Dockerfile file and made the build of OpenCV happening 
- Result of previous point is support for RaspberryPi cam v2 (or any CSI cam compatible with nvarguscamerasrc) when running from the docker container for Jetson nano. Just set "VIDEO_INPUT": "raspberrycam"
- Improved performance by 6 FPS for Linux CUDA desktop machine by using cuDNN for the docker build  
- Made ports configurable from config.json
- Add a configurable confidence threshold in config.json to filtrer out detections that are below this threshold. (default keep > 20% confidence)
- Add favicon
- Add "confirmation" modal when deleting recordings
- Add version number in side menu
- Updated all javascript dependencies: Next.js , Redux, React etc etc...
- Revert back to using the main ./darknet API as memory leak has been also fixed

### How to install

I did the build for Jetson nano, ubuntu machines, if you have a TX2 or Xavier, sorry but you should be able to install it [without docker](https://github.com/opendatacam/opendatacam/blob/v3.0.0-beta.2/documentation/USE_WITHOUT_DOCKER.md).  . _🚨️ Make sure to reinstall darknet from the odc branch as specified in the doc_

For the docker installation, still the same procedure _(Make sure to remove your remove your previous config.json, install-opendatacam.sh, run-docker.sh, run-opendatacam.sh before installing new version)_

```
# Download install script
wget -N https://raw.githubusercontent.com/opendatacam/opendatacam/v3.0.0-beta.2/docker/install-opendatacam.sh

# Give exec permission
chmod 777 install-opendatacam.sh

# NB: You will be asked for sudo password when installing the docker container

# Install command for Jetson Nano
./install-opendatacam.sh --platform nano

# Install command for a Nvidia-docker machine
# NB: Will run from demo file, you can change this after install, see "5. Customize OpenDataCam"
./install-opendatacam.sh --platform nvidiadocker
```

### Next up

We aim to release v3.0.0 in ~ 1 month or before as we are waiting that Jetpack 4.4 final lands along with the Jetson Xavier NX.. And maybe we will also be able to integrate tiny-yolov4 if it is released by then.

- [ ] Clean up code for drag& drop feature.. maybe add some UI tweaks like a drop indicator and a specific progress message
- [ ] Update API docs with new counters attributes
- [ ] Improve installation process, maybe ask people if they want to run from camera / CSI cam or run from file at install time.. or maybe add something to change this in the UI
- [ ] Update documentation for Jetson nano with Raspberrypi  CSI cam
- [ ] Document how to run from custom weights + maybe add a way to use custom weights with docker build
- [ ] Document gstreamer use and pipelines
- [ ] Documentation for yolov3-tiny-prn and yolov4
- [ ] Document tweaking width height size in yolo cfg
- [ ] Add Thresshold  to filter out large detections
- [ ] Option to disable tracker and only count things ?

### Complete changelog from v2.x to v3.x

- Memory leak fixed on jetson's, expected running time is weeks now instead of hours ( https://github.com/opendatacam/opendatacam/issues/182 )
- Drag & drop support for file, just drop a file on the UI and it will restart to process it
- Directional counters, see: https://github.com/opendatacam/opendatacam/issues/122#issuecomment-603733234 
- Save / Reload counters config: see https://github.com/opendatacam/opendatacam/issues/122#issuecomment-604952995
- Support raspberrypi cam (CSI cam) for Jetson nano
- Support for YOLOv4
- Made ports configurable from config.json 
- Add favicon
- Add delete confirmation modal for recordings
- Add version number in side menu
- Support raspberrypi cam (CSI cam) for Jetson nano
- More reliable counting
- Add a configurable confidence threshold in config.json to filtrer out detections that are below this threshold. (default keep > 20% confidence)
- Updated all javascript dependencies: Next.js , Redux, React etc etc...
- Several smaller bugs fixes, clean up logs etc etc...

Thanks for your help testing : @shams3049 @rantgithub @Denizzje @b-g

b-g, 2020-05-06:
🙏 Mega! Will give it a go in the next days!

rantgithub, 2020-05-06:
will help with the tests here :)

shams3049, 2020-05-06:
Christmas is here. Will try them out on all jetsons version. and give you an update roughly around Next week Tuesday.

Denizzje, 2020-05-06:
Very nice! Hopefully no more accidently wiped recordings 😆 . Hope they can make a Yolov4-tiny soon for my tiny Jetson. Or perhaps Yolov4 standard might just work with all the improvements it has?

Hope to be able to update soon, will report back then.

tdurand, 2020-05-07:
> Very nice! Hopefully no more accidently wiped recordings laughing . Hope they can make a Yolov4-tiny soon for my tiny Jetson. Or perhaps Yolov4 standard might just work with all the improvements it has?

Yes I hope too, and no Yolov4 won't work on Jetson nano, would be too slow

> From the "Next up" list, I'd like to +1 the maybe add a way to use custom weights with docker build. Would be great if I could use docker :).

Yes will try to make mounting of external weights on the docker container possible

tdurand, 2020-05-07:
BTW @Denizzje If you'd still like that we feature your project when we release v3.0.0 could you drop me an email at tibo.durand [at] gmail.com ? we would be very glad to do so !

Same for @rantgithub , the work  you are doing on public camera on US highways sounds very interesting.

tdurand, 2020-05-08:
FYI AlexeyAB gently included a fix into darknet for us that we were maintaining on our branch: https://github.com/AlexeyAB/darknet/commit/da050367a8b174f751a2c9720195d51a369c4a4a  , will include in next version OpenDataCam

jonnymccullagh, 2020-05-10:
Running opendatacam:v3.0.0-beta.2-nano in docker on Jetson Nano I'm getting errors:
```
./darknet: error while loading shared libraries: libcudart.so.10.0: cannot open shared object file: No such file or directory
Restart YOLO
```
I tried making darknet with GPU=1 but that was on the host system rather than in any container so probably had no effect. Is this the same issue mentioned above ?

tdurand, 2020-05-10:
> I tried making darknet with GPU=1 but that was on the host system rather than in any container so probably had no effect. Is this the same issue mentioned above ?

Yes this has no effect if you run from a docker container

Could you tell me which version of jetpack your are running ? https://github.com/opendatacam/opendatacam/blob/master/documentation/jetson/FLASH_JETSON.md#how-to-find-out-my-jetpack-version

jonnymccullagh, 2020-05-10:
> Could you tell me which version of jetpack your are running ? https://github.com/opendatacam/opendatacam/blob/master/documentation/jetson/FLASH_JETSON.md#how-to-find-out-my-jetpack-version

` - NVIDIA Jetson Nano (Developer Kit Version)
   * Jetpack 4.4 DP [L4T 32.4.2]
   * NV Power Mode: MAXN - Type: 0
   * jetson_clocks service: inactive
 - Libraries:
   * CUDA: 10.2.89
   * cuDNN: 8.0.0.145
   * TensorRT: 7.1.0.16
   * Visionworks: 1.6.0.501
   * OpenCV: 4.1.1 compiled CUDA: NO
   * VPI: 0.2.0
   * Vulkan: 1.2.70
`

tdurand, 2020-05-11:
Right now seems the docker container isn't compatible with Jetpack 4.4 .. there is a perf issue with cuDNN 8.0 : https://forums.developer.nvidia.com/t/darknet-slower-using-jetpack-4-4-cudnn-8-0-0-cuda-10-2-than-jetpack-4-3-cudnn-7-6-3-cuda-10-0/121579/8 , but also we need to precompile darknet with CUDA 10.0 in the docker container:  https://forums.developer.nvidia.com/t/compiling-darknet-in-l4t-base-r32-3-1-docker-image-get-error-cicc-not-found/113154/10?u=tibo.durand and seems this causes problems with CUDA 10.2 .

 Your only option if to reflash to Jetpack 4.3 for now. Sorry , but we hope nvidia will fix this soon

b-g, 2020-05-22:
@tdurand tiny tiny detail ... please update the OpenMojis `npm i openmoji`, there is now a simple [API](https://github.com/hfg-gmuend/openmoji/blob/master/API.md) which might be handy :)

- [ ] lease update the OpenMojis `npm i openmoji`

tdurand, 2020-05-22:
@b-g thanks for the heads up, need to change how this works here https://github.com/opendatacam/opendatacam/blob/9e3d86e70ef8e2e1fa57d0c0eb1dffb85b925929/components/shared/OpenMoji.js#L11

to something like this

```javascript
openmoji.openmojis.find((openmoji) => openmoji.hexcode === myhexcode);
```

and then remove the icons from the static/ directory

jonnymccullagh, 2020-06-01:
I'm still having problems after buying a logitech usb camera. Using  - NVIDIA Jetson Nano (Developer Kit Version)
   * Jetpack 4.3 [L4T 32.3.1]
   * NV Power Mode: 5W - Type: 1
   * jetson_clocks service: inactive
 - Libraries:
   * CUDA: 10.0.326
   * cuDNN: 7.6.3.28
   * TensorRT: 6.0.1.10
   * Visionworks: 1.6.0.500n
   * OpenCV: 4.1.1 compiled CUDA: NO
   * VPI: 0.1.0
   * Vulkan: 1.1.70 
Used: opendatacam/v3.0.0-beta.2

Is there a specific log for Yolo I could check?

Container logs attached.
[container-logs.txt](https://github.com/opendatacam/opendatacam/files/4711575/container-logs.txt)

` seen 64, trained: 32013 K-images (500 Kilo-batches_64) 
video file: v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink
[ WARN:0] global /var/local/git/opencv/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Internal data stream error.
[ WARN:0] global /var/local/git/opencv/modules/videoio/src/cap_gstreamer.cpp (886) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /var/local/git/opencv/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
`

tdurand, 2020-06-01:
@jonnymccullagh 

From the logs it seems it is not a problem with the camera but rather that it never starts Darknet .. you should have some logs like this when YOLO starts

``
opendatacam_1  | Process YOLO started
opendatacam_1  | (node:86) [DEP0001] DeprecationWarning: OutgoingMessage.flush is deprecated. Use flushHeaders instead.
opendatacam_1  |  CUDA-version: 10000 (10000), cuDNN: 7.6.3, GPU count: 1  
opendatacam_1  |  OpenCV version: 4.1.1
opendatacam_1  |  compute_capability = 530, cudnn_half = 0 
opendatacam_1  |    layer   filters  size/strd(dil)      input                output
opendatacam_1  |    0 conv     16       3 x 3/ 1    416 x 416 x   3 ->  416 x 416 x  16 0.150 BF
opendatacam_1  |    1 max                2x 2/ 2    416 x 416 x  16 ->  208 x 208 x  16 0.003 BF
opendatacam_1  |    2 conv     32       3 x 3/ 1    208 x 208 x  16 ->  208 x 208 x  32 0.399 BF
opendatacam_1  |    3 max                2x 2/ 2    208 x 208 x  32 ->  104 x 104 x  32 0.001 BF
opendatacam_1  |    4 conv     64       3 x 3/ 1    104 x 104 x  32 ->  104 x 104 x  64 0.399 BF
opendatacam_1  |    5 max                2x 2/ 2    104 x 104 x  64 ->   52 x  52 x  64 0.001 BF
opendatacam_1  |    6 conv    128       3 x 3/ 1     52 x  52 x  64 ->   52 x  52 x 128 0.399 BF
opendatacam_1  |    7 max                2x 2/ 2     52 x  52 x 128 ->   26 x  26 x 128 0.000 BF
opendatacam_1  |    8 conv    256       3 x 3/ 1     26 x  26 x 128 ->   26 x  26 x 256 0.399 BF
opendatacam_1  |    9 max                2x 2/ 2     26 x  26 x 256 ->   13 x  13 x 256 0.000 BF
opendatacam_1  |   10 conv    512       3 x 3/ 1     13 x  13 x 256 ->   13 x  13 x 512 0.399 BF
opendatacam_1  |   11 max                2x 2/ 1     13 x  13 x 512 ->   13 x  13 x 512 0.000 BF
opendatacam_1  |   12 conv   1024       3 x 3/ 1     13 x  13 x 512 ->   13 x  13 x1024 1.595 BF
opendatacam_1  |   13 conv    256       1 x 1/ 1     13 x  13 x1024 ->   13 x  13 x 256 0.089 BF
opendatacam_1  |   14 conv    512       3 x 3/ 1     13 x  13 x 256 ->   13 x  13 x 512 0.399 BF
opendatacam_1  |   15 conv    255       1 x 1/ 1     13 x  13 x 512 ->   13 x  13 x 255 0.044 BF
```

But I'm not sure why, as you have the right Jetpack version

tdurand, 2020-06-04:
@b-g quick note about openmojis, I just updated all the icons of OpenMojis to 12.3.0  but I didn't use the npm package (sorry !) .. as it just adds unecessary complexity from my opinion.. I would need to add some script that copy the svgs from the node_modules to the public folder of the next.js app... (or maybe I didn't understand how to use the package ;-) ) . see https://github.com/opendatacam/opendatacam/commit/cfb5cfb47b438421f0ace130e962be1fd505e5c1

willytrueba, 2020-06-15:
Hi,

I have a Jetson Nano with Raspberry 2.0 camera, and I have this error when try to execute.

> OpenCV(4.1.1) /usr/src/app/opencv-4.1.1/modules/videoio/src/cap_images.cpp:253: error: (-5:Bad argument) CAP_IMAGES: can't find starting number (in the name of file): nvarguscamerasrc ! video/x-raw(memory:NVMM),width=1280, height=720, framerate=30/1, format=NV12 ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=360 ! videoconvert ! video/x-raw, format=BGR ! appsink in function 'icvExtractPattern'

How can I solved it?