opendatacam/opendatacam #206: counting data not been recorded on DB

rantgithub, 2020-05-06:
Hi

Performing tests, the counting data on the mongodb is only been saved for the first 1min.

 I tested several videos and even the counter is still running, the data on the DB stops around 1min.

The video completes, the total counter is displayed on the screen and the data on the database only shows like if recorded for seconds.

rantgithub, 2020-05-06:
This is very similar to previous error reported.

Database not showing any data when the button is pressed

checking logs, but no errors. Counter on the screen increases but database not loading and/or not recording data

tdurand, 2020-05-06:
I just tried briefly on a long file (3 min) and it worked... Do you have the bug for streams also ?

tdurand, 2020-05-06:
are you testing with or without docker ?

rantgithub, 2020-05-06:
hi, i was able to repro several times

yes, without  docker

This is/was not happening on beta1....

Error below:

21749:0x3f85790] 29928937 ms: Scavenge 1353.1 (1424.2) -> 1352.3 (1425.2) MB, 7.4 / 0.0 ms  (average mu = 0.062, current mu = 0.000) allocation failure 
[21749:0x3f85790] 29928964 ms: Scavenge 1353.1 (1425.2) -> 1352.3 (1425.7) MB, 3.5 / 0.0 ms  (average mu = 0.062, current mu = 0.000) allocation failure 
[21749:0x3f85790] 29929022 ms: Scavenge 1353.2 (1425.7) -> 1352.4 (1426.2) MB, 3.6 / 0.0 ms  (average mu = 0.062, current mu = 0.000) allocation failure 

<--- JS stacktrace --->

==== JS stack trace =========================================

    0: ExitFrame [pc: 0x3a8b9205be1d]
    1: StubFrame [pc: 0x3a8b9205d3a6]
Security context: 0x2bc94ef1e6c1 <JSObject>
    2: insert [0x1499dbc49e81] [/home/odc/odcb2/opendatacam/node_modules/mongodb/lib/core/sdam/server.js:~371] [pc=0x3a8b938dd81b](this=0x10282668b981 <EventEmitter map = 0x3518c1e9da59>,ns=0x3237d6c95b21 <String[19]: opendatacam.tracker>,ops=0x3237d6c959a9 <JSArray[1]>,options=0x3237d6c95959 <Object map = 0x2cd0bce698...

FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory
 1: 0x8fb090 node::Abort() [node]
 2: 0x8fb0dc  [node]
 3: 0xb031ce v8::Utils::ReportOOMFailure(v8::internal::Isolate*, char const*, bool) [node]
 4: 0xb03404 v8::internal::V8::FatalProcessOutOfMemory(v8::internal::Isolate*, char const*, bool) [node]
 5: 0xef7462  [node]
 6: 0xef7568 v8::internal::Heap::CheckIneffectiveMarkCompact(unsigned long, double) [node]
 7: 0xf03642 v8::internal::Heap::PerformGarbageCollection(v8::internal::GarbageCollector, v8::GCCallbackFlags) [node]
 8: 0xf03f74 v8::internal::Heap::CollectGarbage(v8::internal::AllocationSpace, v8::internal::GarbageCollectionReason, v8::GCCallbackFlags) [node]
 9: 0xf06be1 v8::internal::Heap::AllocateRawWithRetryOrFail(int, v8::internal::AllocationSpace, v8::internal::AllocationAlignment) [node]
10: 0xed0064 v8::internal::Factory::NewFillerObject(int, bool, v8::internal::AllocationSpace) [node]
11: 0x11701ee v8::internal::Runtime_AllocateInNewSpace(int, v8::internal::Object**, v8::internal::Isolate*) [node]
12: 0x3a8b9205be1d 
Aborted (core dumped)
npm ERR! code ELIFECYCLE
npm ERR! errno 134
npm ERR! OpenDataCam@3.0.0-beta.2 start: `PORT=8080 NODE_ENV=production node server.js`
npm ERR! Exit status 134
npm ERR! 
npm ERR! Failed at the OpenDataCam@3.0.0-beta.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

tdurand, 2020-05-06:
I guess this is an issue with node.js / mongodb package

Could you try to 

- remove the node_modules folder in opendatacam
- run `npm install` again

Also which version of node do you have ?  you can know by typing `node --version` ... Maybe you can try upgrading or installing again... not sure..

tdurand, 2020-05-06:
Since beta.1 the only changes was to upgrade the package version of mongodb from 3.3.x to 3.5.x .

rantgithub, 2020-05-07:
Hi

I reinstall npm install and same error

i have version 10.20.1

If i click on the data , i get unexpected error has occurred or doesn't show anything at all.

any suggestions ?

tdurand, 2020-05-07:
not sure, could you try this:

- Install opendatacam with docker to see if you have the same issue, if it is the case this would mean that there is a bug with this mongodb version... 

- If you don't have the bug, try resinstalling mongodb on your machine: https://github.com/opendatacam/opendatacam/blob/master/documentation/USE_WITHOUT_DOCKER.md#mongodb-for-generic-ubuntu-machine-with-cuda-gpu

rantgithub, 2020-05-14:
Hi.

It looks like this version doesn't work. Beta 1 with 3.3.x works and with this Beta 2  3.5.x, doesnt.

This happens on non-docker version.,

rantgithub, 2020-05-14:
Hi

after checking the errors, I have to run 

npm update

After I did this, the recording of data started to work fine.

tdurand, 2020-05-14:
Yes, I think `npm install` should have done too .. you had some problem that didn't install the latest mongodb

closed 2020-05-14 by tdurand