opendatacam/opendatacam #207: File option pretty slow

fabiwlf, 2020-05-07:
Testing the new version with an 1280x720 video and yolov3-tiny. Getting just 6-7 FPS with my jetson nano (just running odc, nothing else). Is there an option to stream the file with gstreamer / doing downscaling for analysis?

Thanks in advance!

tdurand, 2020-05-07:
Hi, 

Not normal, you should get 16 FPS with jetson nano.

The file input resolution doesn't matter, it is then downscaled by YOLO at 416x416.

Could you tell me this:

- Are you using the docker container ? 
- Which version of Jetpack are you using, because with the latest 4.4 developer preview and cuDNN 8.0 there is a bug and you loose 10 FPS : https://forums.developer.nvidia.com/t/darknet-slower-using-jetpack-4-4-cudnn-8-0-0-cuda-10-2-than-jetpack-4-3-cudnn-7-6-3-cuda-10-0/121579

fabiwlf, 2020-05-07:
Thanks, I didn't know about the 4.4 bug. Let me try to go back to 4.3 and see if it's fixed.

[image]

tdurand, 2020-05-07:
Yes it should fix it, I had the same bug a week ago.

I was wondering if you were running OpenDataCam with the docker container ? Because maybe it only effects the installs without docker

fabiwlf, 2020-05-07:
I compiled it without docker, and I'm running it since a few month already with yolov3 (ca 3-4 fps, which was fine for my purpose) but changing to the tiny model didn't result as expected.

tdurand, 2020-05-07:
Okay ! I think if you run the docker container with Jetpack 4.4 it should work as the docker image has the correct version of cuDNN.. I was curious as I didn't test myself before flashing back to 4.3. Anyways, let me know and feel free to close the issue if it is fixed.

Also if you have a nice use case you are using OpenDataCam for and you are willing to share, we are looking to feature some usage of it for v3.0.0 release.

fabiwlf, 2020-05-07:
I'm using odc to monitor CCTV cameras. Have it running on a Google Cloud VM and on the Jetson Nano. I'll try to get it working fine with Jetpack 4.4.

closed 2020-05-07 by fabiwlf

fabiwlf, 2020-05-09:
Didn't have success with Jetpack 4.4. It's 10 FPS slower than with Jetpack 4.3. Hope NVIDIA will fix this.

What about the Deepstream SDK? The confidence problem should be solved in the most recent version. Is it possible to run ODC on Deepstream 5.0? I know about [https://github.com/opendatacam/opendatacam-deepstream](url) but can't get it working.

tdurand, 2020-05-10:
> Didn't have success with Jetpack 4.4. It's 10 FPS slower than with Jetpack 4.3. Hope NVIDIA will fix this.

Yes, but they successfully reproduced it and are looking into it: https://forums.developer.nvidia.com/t/darknet-slower-using-jetpack-4-4-cudnn-8-0-0-cuda-10-2-than-jetpack-4-3-cudnn-7-6-3-cuda-10-0/121579/8  .. So they will fix it at some point ;-)

BTW, did you try to run OpenDataCam as a docker container while on Jetpack 4.4 ? Because this should work as the docker container is pulling the Jetpack 4.3 dependencies

> What about the Deepstream SDK? The confidence problem should be solved in the most recent version. Is it possible to run ODC on Deepstream 5.0? I know about https://github.com/opendatacam/opendatacam-deepstream but can't get it working.

We didn't investigate Deepstream 5.0 as is is just releasing now.. not sure if even the documentation is out yet... but we keep deepstream in our radar and you can follow progress here: https://github.com/opendatacam/opendatacam/issues/112 .. And also contribute ;-)