opendatacam/opendatacam #208: Error connecting to IP Stream

andtheWings, 2020-05-10:
Hi, I am using v3.0.0-beta.2-nvidiadocker on my laptop. I successfully tested it out using a video file. Thought I would try it out with [a random IP Cam in my city](https://www.insecam.org/en/view/468911/) I got on Insecam. I set `"VIDEO_INPUT": "remote_cam"` and `"remote_cam": "http://69.108.44.90:83/"`. When starting up, I got the following error that runs indefinitely:

[image]

tdurand, 2020-05-10:
You need to get the actual image stream link and not the general interface of the webcam: maybe this one will work: http://69.108.44.90:83/GetData.cgi?CH=1

andtheWings, 2020-05-11:
I tried the new link and got running messages like this: [image]

tdurand, 2020-05-12:
hum.. this feed does not seem to be compatible out of the box with OpenCV .

This gstreamer pipeline to decode a MJPEG stream should work:

```souphttpsrc location=http://69.108.44.90:83/GetData.cgi?CH=1 ! multipartdemux ! image/jpeg,width=640,height=480 ! appsink```

But maybe the camera isn't fast enough.. Can't help much more, try to find cameras with low resolution /  fast FPS and direct access to MJPEG stream.

closed 2020-05-12 by tdurand

andtheWings, 2020-05-12:
Ok, thanks for the help!