opendatacam/opendatacam #209: error: unsatisfied condition: cuda>=10.2

expressosub, 2020-05-14:
**After proceeding with the installation of the nvidia-toolkit my result of this command shows:**
______________________________________________________
 docker run --gpus all nvidia / cuda: 9.0-base nvidia-smi

+ ------------------------------------------------- ---------------------------- +
| NVIDIA-SMI 418.87.01 Driver Version: 418.87.01 CUDA Version: 10.1

_____________________________________________________________

**but when proceeding with the commands:**

Install commands:

wget -N https://raw.githubusercontent.com/opendatacam/opendatacam/v3.0.0-beta.2/docker/install-opendatacam.sh

chmod 777 install-opendatacam.sh

./install-opendatacam.sh --platform nvidiadocker

_____________________________________________________________________
**when finished downloading give me:**

Error response from daemon: Cannot restart container opendatacam: OCI runtime create failed: container_linux.go: 349: starting container process caused "process_linux.go: 449: container init caused \" process_linux.go: 432: running prestart hook 0 caused \\ \ "error running hook: exit status 1, stdout:, stderr: nvidia-container-cli: requirement error: unsatisfied condition: cuda> = 10.2, please update your driver to a newer version, or use an earlier cuda container \\\ \ n \\\ "\" ": unknown

____________________________________________________________

the image is installed but I can't start the container correctly

**- the result of command nvidia-smi on my computer are:**

~$ nvidia-smi
Wed May 13 23:57:36 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 418.87.01    Driver Version: 418.87.01    CUDA Version: 10.1     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GT 750M     On   | 00000000:08:00.0 N/A |                  N/A |
| N/A   54C    P8    N/A /  N/A |    279MiB /  2002MiB |     N/A      Default |
+-------------------------------+----------------------+----------------------+

With Cuda 9 and GeForce GT 750M compatibility 3.0 is it possible to install opendatacam?

tdurand, 2020-05-14:
> With Cuda 9 and GeForce GT 750M compatibility 3.0 is it possible to install opendatacam?

I honestly don't know. Could you try to install latest CUDA 10.2 ? 

I had put some guide back in the days that might help you: 

https://developer.nvidia.com/cuda-downloads

https://github.com/opendatacam/opendatacam/blob/master/documentation/nvidia-docker/INSTALL_NVIDIADOCKER.md#22-install--update-nvidia-driver

expressosub, 2020-05-17:
Hi tdurand, I gave up on installing using a docker and installed it directly on the computer. It is likely that I made a mistake in installing the nvidia docker or my nvidia driver is not compatible with the latest nvidia toolkit. After installing directly on my computer, I was able to install opendatacam and now I am using it correctly. Congratulations on creating this tool.

tdurand, 2020-05-18:
great ! thanks

closed 2020-05-18 by tdurand