opendatacam/opendatacam #211: Move docker build process to the cloud

florianporada, 2020-05-20:
I was wondering if it would be possible to build the entire docker image for the ODC on a cloud platform / not on the device itself.

Generally docker offers the feature to build ARM based images on a x86 machine. (https://www.docker.com/blog/multi-arch-images/)
StereoLabs is already doing something similar (https://www.stereolabs.com/docs/docker/building-arm-container-on-x86/).

Maybe we can use  GitHub Actions to automate this, which would be a super nice feature when it comes to patching stuff etc.

What do you think @tdurand ?

tdurand, 2020-05-20:
I mean, yes it would be very nice... I'm not sure if it is possible / what would be the cost of this service (is it free for opensource)

If you are willing to investigate this.. feel free 😉 .. maybe wait for final v3.0.0 release as I'm gonna make more changes to the docker setup: https://github.com/opendatacam/opendatacam/issues/178

vsaw, 2020-12-27:
I've been looking into this and the GitHub action to build ARM socket images is easy to set up. See this branch which contains a sample workflow to build the docker image: https://github.com/vsaw/opendatacam/tree/211-build-docker-github-actin 

For this to be useful we would have to do two more things though 

1. Remove hard tag v3.0.1 from Dockerfile and allow to build from source/development branch
2. store docker images in some registry (maybe GitHubs own Package Manager?) so that People can have easy access to development containers

I'll try to add those if I have Time.

florianporada, 2020-12-28:
Hey Valentin, 

thanks for looking into this issue and the work you put in. 🙏 🙂

1. I guess you are referring to this line? https://github.com/opendatacam/opendatacam/blob/ae22eb8927aab5e8c116904587ae5ee728bb1bb0/docker/build/xavier/Dockerfile#L28 I think we could use a build-argument and parse the version number from the `package.json` file. 

2. Github could be used as registry - yes. Do you have any thoughts about not using dockers official registry?

cc @tdurand - any thoughts as well?

vsaw, 2020-12-28:
@florianporada You're welcome. Opendatacam has provided me with a lot of value, so I'm happy to contribute.

Regarding the docker build:

1. We could pass the tag/branch through build variables. But this would mean we can not validate pull requests but only changes that have been committed to git. As a solution to this limitation I suggest to use the checkout action (which does a "local" merge to check the whole pull request) and copy the merge result to the docker image on build.
2. No reason against docker hub, we can go there if you prefer :-)

tdurand, 2020-12-29:
The thing is I'm not sure that will work is that when we [build darknet](https://github.com/opendatacam/opendatacam/blob/master/docker/build/nano/Dockerfile#L13) when creating the Docker image from what I tried and tested back in the days was we needed to do it in the target platform. (Building on jetson nano for jetson nano etc etc) ... as we set a special ARCH flag to optimize speed: https://github.com/opendatacam/opendatacam/blob/master/docker/build/nano/Makefile#L42 . But truth is I never tested to build somewhere else so it might work ?

And docker hub, github, whatever is simpler works.. I guess if Github offers container hosting let's go full GitHub ..

Regarding the general issue, currently the thing that prevents OpenDataCam to have more releases is that it takes around 1 day to build / test and upload all the container when tagging a new version as there is lots of manual things to do.. so if this gets automated it would be a huge win. (see my release checklist ;-) https://github.com/opendatacam/opendatacam/blob/master/documentation/DEVELOPMENT_NOTES.md#release-checklist )

vsaw, 2020-12-29:
@tdurand 

> The thing is I'm not sure that will work is that when we [build darknet](https://github.com/opendatacam/opendatacam/blob/master/docker/build/nano/Dockerfile#L13) when creating the Docker image from [...]

If i understand the build correctly docker builds the image inside a arm64 QEMU instance. Therefore the resulting binary should work out of the box on Jetson devices. However I still need to verify this. See [Multi-Platform Images](https://github.com/marketplace/actions/build-and-push-docker-images#multi-platform-image)
 
> And docker hub, github, whatever is simpler works.. I guess if Github offers container hosting let's go full GitHub ..

Both are straight forward to setup with only minimal differences

- Dockerhub only requires one additional step to be done by you (generate API key for the workflow).
- GitHub I should be able to set up without your help. But this would potentially require users to import a new registry into their configurations

> Regarding the general issue, currently the thing that prevents OpenDataCam to have more releases is that it takes around 1 day to build / test and upload all the container when tagging a new version [...]

For the time being I would be happy to have regular "development previews". We can keep the major releases manual for now.

tdurand, 2020-12-29:
> GitHub I should be able to set up without your help. But this would potentially require users to import a new registry into their configurations

Okay, I guess maybe keep Dockerhub to avoid any additional friction in the user side, but also we can do both ..

> For the time being I would be happy to have regular "development previews". We can keep the major releases manual for now.

Yes, good idea, if you can automate development preview let's start with that .. Let me know if you need some dockerhub api key etc etc.. Before end of january won't be able to dedicate time to this on my end

vsaw, 2021-01-03:
Development branches builds are now automatically done on every change to the `development` branch and uploaded to Dockerhub using the  `development-<PLATFORM>` tag.

See https://hub.docker.com/r/opendatacam/opendatacam/tags?page=1&ordering=last_updated&name=development for a list of current available images.

Closed by https://github.com/opendatacam/opendatacam/pull/354

# See also

https://github.com/opendatacam/opendatacam/issues/356

PR #354 (2021-01-03): Build Docker Images using GitHub Actions
# Goal

This is a work in progress pull request with the goal to 

- add our docker build setup to our Continuous Integration setup
- publish Opendatacam development previews to dockerhub

# Status

The following things are implemented in this pull request (or still to be implemented)

- [x] Build Opendatacam docker image from local source
- [x] Adds docker build to Opendatacam's CI/CD
- [x] Push development build to dockerhub
    - [x] Generate Dockerhub API key to use in the action
    - [x] Update the action
- [x] Verify the development preview is wo [... cut]
closed 2021-01-03 by vsaw