opendatacam/opendatacam #212: JSON Specification for using custom Neural Network

florianporada, 2020-05-20:
As we spoke about maybe splitting the Tracking/Counting part from the neural network, I had the Idea of writing down the specification the neural network part would have to have to be able to communicate with the ODC component.

From my understanding the JSON stream would need something to sync itself with the video stream (e.g. framecount) the dimensions and the tracking information. 

Is there already something like this?

tdurand, 2020-05-20:
Tracking and Counting are already separated from the neural network process (they are done in node.js, file opendatacam.js), see architecture diagram:

[image]

> I had the Idea of writing down the specification the neural network part would have to have to be able to communicate with the ODC component.

this is a good idea, but I think we should just aim to document it right now as the current way of hooking is very specific to darknet and isn't necessarly the way we want to hook to others potential neural networks framework.. 

> From my understanding the JSON stream would need something to sync itself with the video stream (e.g. framecount) the dimensions and the tracking information.

No, with the current architecture video stream and detections stream (JSON) are never "synced" , the source of truth if the JSON detections stream and the video stream is just there for UI purposes, but might have sometimes a slight delay with the JSON stream. I think this is fine as it simplifies much more the problematic.

To hook to another neural network framework it is not just a matter of the JSON "format" of the detections also. Very roughly right now to hook to OpenDataCam you should have a program that:

- Send detections as a JSON stream on the port specified, specification of the "frame" is : 

```
{
   "frame_id":16,
   "objects":[
      {
         "class_id":6,
         "name":"car",
         "relative_coordinates":{
            "center_x":0.485693,
            "center_y":0.570632,
            "width":0.091478,
            "height":0.156059
         },
         "confidence":0.892377
      }
   ]
}
```

- Send video stream as MJPEG stream on the port specified

- Print the video size on the stdout like "Video stream: 640 x 480" when starting (this is very hacky) https://github.com/opendatacam/opendatacam/blob/master/server.js#L72 , so OpenDataCam can parse it 

- You also need to change the YOLO.js class to orchestrate your program  https://github.com/opendatacam/opendatacam/blob/master/server/processes/YOLO.js#L47

So you see right now this is very tightly coupled with darknet way of working.. But this could be changed to be made as "connectors" that you can overwrite depending on what you are integrating too. For example integrating to deepstream https://github.com/opendatacam/opendatacam/issues/112 or integrating to depth AI https://github.com/opendatacam/opendatacam/issues/203 

I think the best way to go it by experimenting as we go... If someone forks OpenDataCam and makes a good integration with another neural network framework and share it back... then we can see to adapt it into the main repo and makes a config flag to choose between darknet and this other thing. The effort is very big to do it ourself...

The other burden will also be to update documentation, build new Docker easy install container for each neural network framework etc etc... and maintain along the way.

I'm not sure if this answers the question 😉.. we could put part of what I'm documenting in this issue somewhere in the doc along with the Architecture diagram.

florianporada, 2020-05-25:
It does answer the question, thanks!
I see that it's quite a challenge to untangle those two components. 
In general either the Machine Learning component needs to provide those exact requirements or there needs to be an adapter-type thing which does the 'translation'.

I'll keep this in my backlog tho :)

vsaw, 2022-02-13:
I decided that this will happen after ODC v4.0.0. Will keep this issue open as a reminder, but remove from the milestone.