opendatacam/opendatacam #213: Install script stops all running docker containers

divbell, 2020-05-23:
I suppose I should have looked that script over better before running on my jetson nano k8s node. :)

Is this really necessary?

`$ ./install-opendatacam.sh --platform nano
Installing opendatacam docker image
index : 0
Stop any running docker container...
383089bcad8a
57b1795421c2
ed84ddd88063
2a53cfbfd309
0b12a6fedd5c
99ae7f283146
75eb78d6d582
acf00c0a4f22
b62303187aaf
ef3992abd377
1b5065b2b406
8553a0ab62ac
3cff1a5b8620
f1635ca17b3d
5599d57455af
9f2b7fbae51e
f3ac1a1f743f
abd8d8c360c4
ba9634571797
f05180e376cd
6db609c3a541
f958ae388326
1ba8773572fe
1221364e6441
76663cae53da
1273e9496be0
582ee83e270c
0aeb6c2cb88b
87a6552837af
2ad78e1e68f0
20c83a2b57f3
562ccf320407
6fcfeec7d06c
6e6eb6075aae
1e7fa0361971
1cb4d81f2bf7
14496aee7fe0
afe58dde0a93
616ecce853f2
289731ff67a0
5081b2695801
4bc319bfa7c7
181961e4d050
f8f9966f1809
51ea87386811
7cb50202dadf
cb7630426cf0
422e74c3b37f
f412ddd65358
5d4a99e9c367
417b145adf32
ce95be5bf270
4d511c373518
c8b9a0fa611e
f7ec935c5a21
0aedd6c211f2
1198cd7fc743
037f71ec270d
Installing opendatacam v3.0.0-beta.2 for platform: nano ...
`

tdurand, 2020-05-23:
Sorry for this, probably not .. It was a quick fix to avoid people starting twice the container but we can improve that to stop other opendatacam container only.

Wondering what is your setup, several jetson nano working together in a K8s cluster ? Does this improves performance ??

divbell, 2020-05-23:
Nothing terribly bad seems to have happened by killing those containers, fwiw.  

As for the k8s cluster, I created a 3-node Jetson Nano k8s cluster because I wanted a container model for Jetson projects and the k8s orchestration to deploy to any node in the system and abstract that all.  I'm primarily using IP cams and consider the Jetsons kind of brittle so if one Jetson dies, the idea is the pod gets rescheduled on a different node and I minimize an outage.  It's all pretty experimental but I have deepstream-app stuff running doing basic object detection and tracking.  Just diving into opendatacam.

tdurand, 2020-05-25:
Okay ! Are you using a cloud service that offers jetson hosting in the cloud or running you own infrastructure ?