opendatacam/opendatacam #214: ./darknet: error while loading shared libraries: libcudart.so.10.0: cannot open shared object file: No such file or directory

hamelsveld, 2020-05-31:
Hi,

Im getting the error:
./darknet: error while loading shared libraries: libcudart.so.10.0: cannot open shared object file: No such file or directory

 nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2019 NVIDIA Corporation
Built on Wed_Oct_23_21:14:42_PDT_2019
Cuda compilation tools, release 10.2, V10.2.89

- NVIDIA Jetson Nano (Developer Kit Version)
   * Jetpack 4.4 DP [L4T 32.4.2]
   * NV Power Mode: MAXN - Type: 0
   * jetson_clocks service: inactive
 - Libraries:
   * CUDA: 10.2.89
   * cuDNN: 8.0.0.145
   * TensorRT: 7.1.0.16
   * Visionworks: 1.6.0.501
   * OpenCV: 4.1.1 compiled CUDA: NO
   * VPI: 0.2.0
   * Vulkan: 1.2.70

is there a way to fix this error?

hamelsveld, 2020-05-31:
downgrading solves it

closed 2020-05-31 by hamelsveld

tdurand, 2020-06-01:
great you found the solution, I just added a note in the readme about Jetpack 4..4 DP : https://github.com/opendatacam/opendatacam/commit/db7b6f61503777afc6135397d92498a1394025fc

QingxinWx, 2020-07-23:
Hi how doyou solved it？