opendatacam/opendatacam #217: Tracker showing negatives numbers on data

rantgithub, 2020-06-05:
Hi

Notice that the number of objects are getting negative when realtime data is observed

Please see attached picture
[image]

Behavior observed is number of objects increases but jumps to a negative number and comes back

tdurand, 2020-06-05:
Thanks. Will have a look next week. Can you reproduce it all the time ?

rantgithub, 2020-06-05:
hi, yes,
i am not sure if is an error, because you see the number show up for like 1sec and after the normal count remains

rantgithub, 2020-06-05:
Asking  if you have seen this before on your test, but from all the dev i have download, i see these 2 lines on the bottom right corner completely outside of the boundaries.

I didn't create them neither cant delete them

I post this on a previous bug and I thought was for that video, but it looks like is on the database

[image]

those black lines are on bottom right corner and the center is below the line of the video , meaning there is no way to delete them

tdurand, 2020-06-05:
ahhh, maybe you can try this

Download the "counter config" . 

Remove the lines from the json and upload again...

Those might be "ghost" from a previous config you did

rantgithub, 2020-06-05:
no, because this is odcdev database, it is "alone" and only use it for this purpose

i went inside of the database and i found those black lines

I just deleted them directly,

check if you have those on this dev version

tdurand, 2020-06-05:
hum... I don't have them.. weird

tdurand, 2020-06-05:
there is not database "init data" in the code so I guess this was some old stuff laying around you local mongodb ;-)

rantgithub, 2020-06-05:
ok, good :)

alanb128, 2020-06-18:
On 3.0.0 beta 3 I'm seeing zero for the Tracker counter even though it downloads data.
[image]

tdurand, 2020-06-18:
Thanks, I wasn't able to reproduce the issue.. are you able to reproduce it every time ?

alanb128, 2020-06-18:
I did two separate recordings in that session and the tracker displayed zero for both. I just tested again now, and noticed during "ongoing" recording that the tracker was jumping between negative and positive values. Once the recording was stopped, it showed 235 tracker objects and 19 counter objects. (19 being correct)

tdurand, 2020-06-18:
Ok, looks like @rantgithub issue.. will try again to reproduce.. seems just a bug with the display !

rantgithub, 2020-06-18:
Yes

this is exactly what I experienced

Denizzje, 2020-06-25:
My Jetson Nano with beta 1 does exactly the same. I didn't do anything with the Mongo database when moving from v2 to 3. Never reported it because I don't use the tracker. But yes it does this thing with jumping between zero and random negative numbers.

vsaw, 2020-10-21:
I believe I found the cause for this: Looking at the following lines:

https://github.com/opendatacam/opendatacam/blob/master/server/Opendatacam.js#L302-L305

`biggestTrackedItemIdThisFrame` will be `0 ` if there no tracked items in this frame. This will then calculate to a negative number if the tracking did not start ab `0`.

The fix should be simple. Something along those lines should work:

```js
// Compute nbItemsTrackedSinceRecordingStarted based on ids (assume that id increment is one)
if(trackerDataForThisFrame.length > 0) {
    const biggestTrackedItemIdThisFrame = trackerDataForThisFrame[trackerDataForThisFrame.length - 1].id;
    const nbItemsTrackedSinceRecordingStarted = biggestTrackedItemIdThisFrame - Opendatacam._refTrackedItemIdWhenRecordingStarted;
    Opendatacam.totalItemsTracked = nbItemsTrackedSinceRecordingStarted;

}
```

vsaw, 2020-11-17:
@tdurand this was fixed in https://github.com/opendatacam/opendatacam/pull/299 and can be closed now

closed 2020-11-17 by tdurand