opendatacam/opendatacam #221: YOLOv5

tdurand, 2020-06-11:
Some craziness in the YOLO world right now 😎️ . 

The people behind the port of Yolov3 to PyTorch and iOS released : https://github.com/ultralytics/yolov5

Some article about it: https://blog.roboflow.ai/yolov5-is-here/ ( but this article seems to be very much "marketing" , and **does not draw a fair comparison betweek Yolov4 and yolov5.. it is not written by the actual authors of YOLOv5** )

From what I understood (could be wrong)

- YOLOv5 is not actually much more performant than YOLOv4 for the full weights but it is improving the small models (like tiny-yolo) that aren't released yet by YOLOv4 darknet
- YOLOv5 seems a bad naming .. 
- YOLOv5 isn't compatible with darknet it is based on PyTorch
- YOLO architecture is more relevant than ever, with lots of research going on 🔥️🔥️

Regarding OpenDataCam .. this means that "yet another neural network framework" could be hooked to it (Pytorch) .. but for v3 release we will obviously stick with darknet

Some more details for those interested:

From @alexeyAB : https://github.com/AlexeyAB/darknet/issues/5920#issuecomment-642268465

> @glenn-jocher did a lot for the development and improvements of Yolo and showed a lot of ideas, he created at least 2 very good repositories on Pytorch. Thus, he gave Yolo a long life outside of Darknet. All this hype around the Yolov5 was not raised by him.

From @glenn-jocher ( of ultranalytics) : https://github.com/ultralytics/yolov5/issues/4#issuecomment-642422169

> which builds on years of hard work (by myself and many community contributors) porting and perfecting YOLOv3 to PyTorch from Darknet, which pjreddie amazingly began and @AlexeyAB so impressively continued and advanced. Most importantly in following the spirit of YOLO, we are targeting results that we hope may be in reach of every individual who would express an interest in the field, not merely available to those with unlimited resources at their disposal. Our smallest model, to cite an example, trains on COCO in only 3 days on one 2080Ti, and runs inference faster and more accurately than EfficientDet D0, which was trained on 32 TPUv3 cores by the Google Brain team. By extension we aim to comparably exceed D1, D2 etc. with the rest of the YOLOv5 family.

AlexeyAB, 2020-06-11:
About roboflow article: https://github.com/AlexeyAB/darknet/issues/5920#issuecomment-642812152

> Invalid comparison results in the roboflow.ai blog: https://blog.roboflow.ai/yolov5-is-here/
> 
> Actually if both networks YOLOv4s and ultralytics-YOLOv5l are trained and tested on the same framework with the same batch on a commond dataset Microsoft COCO: https://github.com/WongKinYiu/CrossStagePartialNetworks/issues/32#issuecomment-638064640
> * weights size: YOLOv4s `245 MB` vs YOLOv5l `192 MB` vs YOLOv5x `366 MB`
> 
> * test-dev accuracy on MSCOCO: YOLOv4s-608 `45% AP` vs YOLOv5l-736 `44.2% AP` (**YOLOv4 is more accurate**)
> 
> * speed with batch=16: YOLOv4s-608 `10.3ms` vs YOLOv5l-736 `13.5ms` (**YOLOv4 is faster**)
> 
> * roboflow.ai shared the Latency-Accuracy chart withultralytics-YOLOv5 which are measured with batch=32 and then divided by 32, while latency must be measured with batch=1, because the higher batch - the higher latency, latency of 1 sample can't be less than latency of the whole batch, so **real latency of YOLOv5 can be up to `~1 second`** with high batch-size=32-64
> 
> * they stated 140 FPS for YOLOv5 (s/m/l/x ???) (what batch-size ???) while **YOLOv4 achieves ~400 FPS** just with batch=4 by using OpenCV-dnn or TensorRT on GPU RTX 2080ti (table above)
>

danielbarry, 2020-06-12:
@tdurand See RoboFlow's updated blog: https://blog.roboflow.ai/yolov4-versus-yolov5/

tdurand, 2020-06-12:
thanks.. interesting read.. Overall it is very good to see so much effort and research around YOLO...just too bad we are so bad at naming things... The semantics of v4, v5 implies that one is better than another...

For OpenDataCam, we will release very soon the OpenDataCam v3 which we will still be using Darknet as our neural network framework.. featuring the new improvement of Yolov4 . 

In the future a big project would be to be "neural network" agnostic and think of OpenDataCam as a nice wrapper around any Object Detections framework (Yolov4, Yolov5, deepstream, depth ai etc etc .....) providing a clean UI, ease of use, tracking and counting  but this would require a significant effort...

closed 2021-05-22 by tdurand